wmbusmeters/wmbusmeters #607: Timeout

thorsten, 2022-09-09:
Hi,

I'm using a Nano CUL 868 on a Raspberry Pi 4 with the Home Assistant Add-on.  But I'm getting a timeout then reading the data:

> s6-rc: info: service s6rc-oneshot-runner: starting
> s6-rc: info: service s6rc-oneshot-runner successfully started
> s6-rc: info: service fix-attrs: starting
> s6-rc: info: service fix-attrs successfully started
> s6-rc: info: service legacy-cont-init: starting
> s6-rc: info: service legacy-cont-init successfully started
> s6-rc: info: service legacy-services: starting
> s6-rc: info: service legacy-services successfully started
> Syncing wmbusmeters configuration ...
> Registering meters ...
> Generating MQTT configuration ... 
> Broker core-mosquitto will be used.
> MQTT Discovery cleanup...
> Running wmbusmeters ...
> (config) "device" "/dev/ttyUSB0:cul:t1"
> (wmbusmeters) version: 1.9.0-1-gaa2f602
> (config) using device: /dev/ttyUSB0:cul:t1 
> (config) using device: MAIN=/dev/ttyUSB0:mbus:2400 
> (config) number of meters: 0
> (lookup) with file/hex "/dev/ttyUSB0" cul
> (serial) check if /dev/ttyUSB0 can be accessed
> (serial) tty /dev/ttyUSB0 can be accessed
> (serial) EVENT thread interrupted
> (serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,144) modem(DTR|RTS)
> (serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,224,18,140,197,127,248) modem(DTR|RTS)
> (serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
> (serial /dev/ttyUSB0) sent "560D0A"
> (serial) EVENT thread interrupted
> (serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
> (cul) probe response "V 1.67 nanoCUL868<0D><0A>"
> (serial) EVENT thread interrupted
> (serialtty) closed /dev/ttyUSB0 (detect cul)
> (cul) are you there? yes
> (main) opening /dev/ttyUSB0:cul:t1
> Started config cul on /dev/ttyUSB0 listening on t1
> (cul) on /dev/ttyUSB0
> (serial) EVENT thread interrupted
> (serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,144) modem(DTR|RTS)
> (serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,16,16,140,197,127,248) modem(DTR|RTS)
> (serialtty) opened /dev/ttyUSB0 fd 3 (cul)
> (main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
> (cul) set link mode t
> (serial /dev/ttyUSB0) sent "6272740D0A"
> (serial) EVENT thread interrupted
> (serial) received binary "3F202862727420697320756E6B6E6F776E2920557365206F6E65206F6620"
> (cul) checkCULFrame "? (brt is unknown) Use one of "
> (cul) no eol found yet, partial frame
> (serial) received binary "41204220432045206520462066204720682069204B206B206C204D206D205220542074205520562057205820782059205A207A0D0A"
> (cul) checkCULFrame "? (brt is unknown) Use one of A B C E e F f G h i K k l M m R T t U V W X x Y Z z<0D><0A>"
> (cul) no leading 'b' so it is text and no frame
> (wmbus device) timeout request id 1
> (cul) received ""(cul) setting link mode(s) t1 is not supported for this cul device!
> (serial) stopping manager
> (config) "device" "MAIN=/dev/ttyUSB0:mbus:2400"
> (config) "donotprobe" "/dev/ttyAMA0"
> (config) "logtelegrams" "false"
> (config) "format" "json"
> (config) "logfile" "/dev/stdout"
> (config) "shell" "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
> (config) "" ""
> (wmbusmeters) using log file /dev/stdout
> s6-rc: info: service legacy-services: stopping
> s6-rc: info: service legacy-services successfully stopped
> s6-rc: info: service legacy-cont-init: stopping
> s6-rc: info: service legacy-cont-init successfully stopped
> s6-rc: info: service fix-attrs: stopping
> s6-rc: info: service fix-attrs successfully stopped
> s6-rc: info: service s6rc-oneshot-runner: stopping
> s6-rc: info: service s6rc-oneshot-runner successfully stopped

My config:

>   data_path: /config/wmbusmeters
>     enable_mqtt_discovery: false
>     conf: |-
>       loglevel=debug
>       device=/dev/ttyUSB0:cul:t1
>       device=MAIN=/dev/ttyUSB0:mbus:2400
>       donotprobe=/dev/ttyAMA0
>       logtelegrams=false
>       format=json
>       logfile=/dev/stdout
>       shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
>     meters: []
>     mqtt: {}

Any hint how to fix it?

thorsten, 2022-09-24:
@weetmuts do you have any idea?

weetmuts, 2022-09-24:
You are setting t1 mode but it says: "(cul) setting link mode(s) t1 is not supported for this cul device!"
You have to carefully compile your cul software so that it contains t1 and c1 wmbus.
Here is a such a firmware: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html

thorsten, 2022-09-24:
That solved my issue, thanks for the hint!

closed 2022-09-24 by thorsten