wmbusmeters/wmbusmeters #608: Qundis driver: unknown

magcode, 2022-09-11:
Hello and thank you for this great work.
I'm using a CUL to listen to a "Qundis module 5.5 heat"

According to their [documentation](https://qundis.de/wp-content/uploads/2017/10/Onepager_Qmodule5.5heat.pdf) there should be two different messages: One message every 112 seconds, another message every 7.5 minutes

I think I see both, but only one of them can be decoded:

```
(wmbusmeters) HUP received, restarting and reloading config files.
(wmbusmeters) logging started 2022-09-11 09:25:48 using 1.9.0-13-gc15f959
[2022-09-11_09:25:49] Started config cul on /dev/usb-ec3k listening on c1
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 37169538
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x23
                device: cul
                  rssi: -55 dBm
                driver: unknown!
telegram=|_5344934438951637233778077983115067934423040DFF5F3500824C00000E0007C005FFFF00000000BF2C00000000DF28000000000000000000000000000000000000000000000000000000002F046D1B07CB29|+73
Received telegram from: 37169538
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x23
      Concerning meter: 67501183
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat meter (0x04)
                   ver: 0x23
                device: cul
                  rssi: -56 dBm
                driver: qheat
telegram=|_3C4493443895163723377283115067934423044D0000200C05000000004C0500000000426CBF2CCC080500000000C2086CDF28326CFFFF046D1B07CB29|+88
Received telegram from: 37169538
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x23
                device: cul
                  rssi: -56 dBm
                driver: unknown!
telegram=|_5344934438951637233778077983115067934423040DFF5F3500824D00005F0107C005FFFF00000000BF2C00000000DF28000000000000000000000000000000000000000000000000000000002F046D1C07CB29|+184
Received telegram from: 37169538
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x23
                device: cul
                  rssi: -55 dBm
                driver: unknown!
telegram=|_5344934438951637233778077983115067934423040DFF5F3500824D0000F10007C005FFFF00000000BF2C00000000DF28000000000000000000000000000000000000000000000000000000002F046D1E07CB29|+293
```

Would there be any chance to read the "unknown" telegram as well?

weetmuts, 2022-09-11:
You can take a look at the contents of the unknown telegram like this:
https://wmbusmeters.org/analyze/5344934438951637233778077983115067934423040DFF5F3500824C00000E0007C005FFFF00000000BF2C00000000DF28000000000000000000000000000000000000000000000000000000002F046D1B07CB29

It contains a identifier of the combined meter+radio adapter and a date time.
Also some properietary data. Historical readings perhaps? 

As you can see there is not that much of interest to us currently in the unknown telegram.
But we could extract the enhanced id and add it to the qheat json output if you like.

weetmuts, 2022-09-11:
Here is the content of the other telegrams, that we can understand:
https://wmbusmeters.org/analyze/3C4493443895163723377283115067934423044D0000200C05000000004C0500000000426CBF2CCC080500000000C2086CDF28326CFFFF046D1B07CB29

magcode, 2022-09-11:
I see!

Then I will just ignore the unknown telegram for now.
Thanks for having a look!

closed 2022-09-11 by magcode