wmbusmeters/wmbusmeters #610: Support MQTT discovery outside the HA addon

patrislav1, 2022-09-11:
I'm using wmbusmeters with HA - but it runs outside of HassOS, due to limitations of Proxmox USB passthrough (the SDR stick cannot be passed through to the HassOS VM, so I have to run wmbusmeters in a separate LXC container, where USB passthrough is no problem).

So now I want to populate the sensors in HA using MQTT discovery. But in my standalone setup, I can't run the [ha-addon](https://github.com/weetmuts/wmbusmeters/tree/master/ha-addon) discovery script, as it depends on `bashio` which is not available on a regular system outside HassOS.

Looking at [mqtt_discovery.sh](https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/mqtt_discovery.sh) it seems like the "heavy lifting" for the actual discovery can be neatly separated from the interfacing to the HassOS/add-on environment. As a "proof of concept" I copy/pasted the lines [48](https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/mqtt_discovery.sh#L48)-78 from the original script and feed it with keys/values retrieved from `/etc/wmbusmeters.d` files from a standalone wmbusmeters installation.

I think the "heavy lifting" part that I copy/pasted can be refactored into a dedicated script (like `discover_meter.sh` or similar), and this script can then be called from `mqtt_discovery.sh` as well as from `mqtt_discovery_standalone.sh` (or so) to avoid code duplication.

If something like this has any chance to be upstreamed, I'd be willing to make a PR. But due to my setup I can't test if my changes break anything in the HassOS add-on.

WDYT?

Related to #575

```bash
#!/bin/bash

MQTT_ARGS="-h homeassistant.fritz.box -u mqtt-username -P mqtt-password"
SENS_CONF_DIR=/etc/wmbusmeters.d
MQTT_DISCOVERY_PREFIX="homeassistant"

for sen_conf in $SENS_CONF_DIR/*; do
    declare -A aryKV
    for line in $(cat $sen_conf); do
        IFS='=' read -r -a kv <<< "$line"
        aryKV[${kv[0]}]=${kv[1]}
    done
    echo name: ${aryKV['name']} driver: ${aryKV['driver']}

    if [ "${aryKV['id']+true}" ] && [ "${aryKV['driver']+true}" ] ; then
        file="mqtt_discovery/${aryKV['driver']}.json"
        if test -f "$file"; then
            for attribute in $(jq --raw-output -c -M '. | keys[]' $file)
            do
                echo "Attribute: ${attribute}"
                aryKV["attribute"]=$attribute

                filter=".${attribute}.component"
                component=$(jq --raw-output -c -M $filter $file)
                if [[ ! -z "$component" ]] && [ $component != "null" ] ; then
                    topic="$MQTT_DISCOVERY_PREFIX/$component/wmbusmeters/${aryKV['id']}_$attribute/config"

                    filter=".${attribute}.discovery_payload"
                    payload=$(jq --raw-output -c -M $filter $file)

                    for key in ${!aryKV[@]}; do                                                              
                        payload="${payload//\{$key\}/${aryKV[${key}]}}"
                        #echo ${key} ${aryKV[${key}]}                                                             
                    done
                    echo "  Add/update topic: $topic"
                    echo $payload |jq
                    aryDiscTopic+=($topic)
                    /usr/bin/mosquitto_pub $MQTT_ARGS -r -t "${topic}" -m "${payload}"
                fi
            done
        else
            echo "  File $file not found."
        fi
    fi
done
```

KrzysztofHajdamowicz, 2023-02-03:
#TODO: Provide a pull request with bash script for running outside HA, as a docker container. Provide json for `amiplus` meter.

BIBOLV, 2023-05-13:
@patrislav1 did you manage to create script for your environment? Can you create PR for this?

jannickfahlbusch, 2023-08-21:
Hi,

I recently moved my wmbusmeters installation from the HomeAssistant to a dedicated VM (Because of #989).
For this I adapted the [autodiscovery script of the addon](https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/1019c3b31298114cd77b882fe81f3077b9be275b/wmbusmeters-ha-addon/mqtt_discovery.sh) to work outside of HomeAssistant.

You can find the script here: https://gist.github.com/jannickfahlbusch/f623dd8f056d03d0517e182cd493e81b