wmbusmeters/wmbusmeters #612: apator162

kitor, 2022-09-15:


kitor, 2022-09-15:
The old one I had (and worked properly?) reports `--version` as ga5f078, but I can't find it in repo.

Though I must run new version for a bit longer, as after running old one for a few minutes I didn't get a single message decoded, just spamming console with `tpl-ci-field b6`, while new one "decodes" data badly every minute or two.

kitor, 2022-09-20:
ah, as I forgot, aes key for those telegrams is `00000000000000000000000000000000`

kitor, 2022-09-21:
@weetmuts I'm not sure how to progress here.

All "broken" telegrams start with `B6` field (followed by number of bytes, see #598).
Then there's a usual `7Axx004085xxxx` as in "regular" telegram.
However later looks like a random noise. I attached over 1000 telegrams from my two meters: [apator162_telegrams.txt](https://github.com/weetmuts/wmbusmeters/files/9619151/apator162_telegrams.txt)

What is weird, since 3 days I'm not getting a single "regular" telegram from one of my meters.

weetmuts, 2022-09-23:
Your telegrams do not look like apator162 telegrams, in fact the auto-mapping to driver says that
these are apt_07_05, which is a not yet working driver. It reports the fixed value 4711 for all readings! :-)
You can see this because the telegram is in fact not encrypted according to the header.
The content could be encrypted using a mfct specific encryption though. 

https://wmbusmeters.org/analyze/5a441486128805000507B60afffff5450106f41BB0767a50004085ee8c318c97BB9e01929519e82d56de6fdB31dBf5ac13628f3B380f3d4e71f3a6d97e8cc7f8dB18340eefd2991fc5Bdee1a80781a37493a7cdc7d8e81308ac7e9

A normal apator162 is indeed encrypted, you can see the AES_CBC_IV here

https://wmbusmeters.org/analyze/5A441486128805000507B60AFFFFF5450106F41BB0767A50004085EE8C318C97BB9E01929519E82D56DE6FDB31DBF5AC13628F3B380F3D4E71F3A6D97E8CC7F8DB18340EEFD2991FC5BDEE1A80781A37493A7CDC7D8E81308AC7E9

So, I think you have never seen values for these telegrams before, but when wmbusmeters started parsing them,
with the b6 fix, then you started seeing them. They were silently discarded before. Confusing things....

weetmuts, 2022-09-23:
If you filter on exactly one of your meters and do logtelegrams do you get both the weird telegram from that meter and the normal apator162 telegram?

kitor, 2022-09-23:
As you can see in my post above, I catch exactly two IDs (hot and cold water) and for both IDs I get both types of telegrams - ones that are decoded properly by apator162 (and my meters are 16 2) and those I attached.

weetmuts, 2022-09-23:
```
Either way, in case value is not possible to parse, IMO wmbusmeters shall either not call shell at all, or add extra field that tells the script that parsing indeed failed.
```

Yes, this is a use case that I have not thought about before. Lets put this into this issue as well:
https://github.com/weetmuts/wmbusmeters/issues/615#issuecomment-1255887526

weetmuts, 2022-09-23:
Can you collect telegrams using plain rtlwmbus and just greping for your two meters.
See if you find two distinct types of telegrams, and post them here.

kitor, 2022-09-23:
No problem, I'll update you when it's done.

weetmuts, 2022-09-23:
The apt_07_05 can be found with 441486
and the apator162 can be found with 440106

kitor, 2022-09-23:
Hmm, it is possible that there's a conflict of IDs.
I moved antenna earlier this morning and I started receiving more of "regular" frames that are parsed by apator162...

I just received a proper dipole (I was using tv antenna until now). Now I see I receive always two telegrams, one after another, for both sensors (this is still from wmbusmeters log):

```
telegram=|4E4401061288050005077A200040852F2F_0FDF57B9961000008447BCB6DC35D9FF92050010627F010041555B423221D500436D037301357501008C620100F94E0100143B0100A03E40F003FFFFE62C|+75
telegram=|5A441486128805000507B6_0AFFFFF5450106F41BB5767A20004085B874ED8805E0CCA24076E64CF40207693BB197BAD88FA1CD119BED40E719907D18A694D25AEAED16A3F40010F5BF7D0E8CEDD0C0FD91F708314E8C7FD472A889|+80
(apator162) telegram contains a register (7a) with unknown size.
```
and

```
telegram=|25441486475036350408B6_0AFFFFF5450186F41BA953A0A10000000000000300372DAC158417|+212
telegram=|4E4401068686140005077A650040852F2F_0FE457B99600000010B954000041545A426B20D7004379037301C5500000564B00009E4600006A410000A01778EC03FFFFFFFFFFFFFFFFFFFFFFFFFFAD14|+258
```

This repeats, and obviously since `wmbusmeters` keeps last value I see duplicates in log, though with different RSSI values:

```
{"media":"water","meter":"apator162","name":"woda","id":"00058812","total_m3":98.146,"timestamp":"2022-09-23T10:06:43Z","device":"rtlwmbus[00000001]","rssi_dbm":27}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:07:43Z","device":"rtlwmbus[00000001]","rssi_dbm":12}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:07:44Z","device":"rtlwmbus[00000001]","rssi_dbm":27}
{"media":"water","meter":"apator162","name":"woda","id":"00058812","total_m3":98.146,"timestamp":"2022-09-23T10:08:36Z","device":"rtlwmbus[00000001]","rssi_dbm":12}
{"media":"water","meter":"apator162","name":"woda","id":"00058812","total_m3":98.146,"timestamp":"2022-09-23T10:08:36Z","device":"rtlwmbus[00000001]","rssi_dbm":28}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:09:16Z","device":"rtlwmbus[00000001]","rssi_dbm":13}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:09:17Z","device":"rtlwmbus[00000001]","rssi_dbm":27}
{"media":"water","meter":"apator162","name":"woda","id":"00058812","total_m3":98.146,"timestamp":"2022-09-23T10:10:11Z","device":"rtlwmbus[00000001]","rssi_dbm":11}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:10:34Z","device":"rtlwmbus[00000001]","rssi_dbm":13}
{"media":"water","meter":"apator162","name":"woda","id":"00148686","total_m3":21.689,"timestamp":"2022-09-23T10:10:35Z","device":"rtlwmbus[00000001]","rssi_dbm":26}
```

So either this device emits two frames, or in fact there's a double ID conflict in my neighborhood.

weetmuts, 2022-09-23:
Indeed! It seems like your meters sends two telegrams, an older format and a new format. 

I have seen this in some heat cost allocators. Probably to support older data collection infrastructures
while still upgrading the meter protocol. Weird though, since it takes more battery life.

Heh, this has never been detected before since accidentally the wmbusmeters parser diead at b6
and those telegrams were ignored....

This telegram that you posted clearly does not look encrypted:
25441486475036350408B6_0AFFFFF5450186F41BA953A0A10000000000000300372DAC158417
since you have all those zeroes.

Can you collect some telegrams both old and new, from just one of your meters?

Probably the same information is in the weird telegram somewhere so we could merge the decoding
of the other telegram into the apator162 driver. This way we could understand even older meters
that only send the old telegram type.

kitor, 2022-09-23:
Ah sorry, this unencrypted telegram is not relevant, I copied a wrong line from log. Yes, I can collect both telegrams, in fact I have telegram logging enabled so I'll just grep them from logs this evening (which should give us a lot of data + some value changes to work with).

kitor, 2022-09-27:
Pairs of messages, filtered out form logs for each meter.
I first filtered out meter ID, then filtered only where two consecutive messages were received (to limit data only to message pairs)

[00148686_pairs.txt](https://github.com/weetmuts/wmbusmeters/files/9652530/00148686_pairs.txt)
[00058812_pairs.txt](https://github.com/weetmuts/wmbusmeters/files/9652531/00058812_pairs.txt)

kitor, 2022-09-27:
Also a quick note: my meters were replaced recently (less than a year ago) so if `apt_07_05` is older meter, theory about sending same data twice in different format makes sense.

weetmuts, 2022-09-27:
Ok, I took a quick look at the telegrams from apt_07_05 and they seem to be encrypted. Most of the databytes change between every telegram. Though perhaps there is some leading data after b6 that does not change that much.

Maybe they are encrypted using the same key as the apator162, ie zero?

I have moved the detection of the apt_07_05 mfct/type/version into the apator162 driver and deleted the apt_07_05
driver. When a telegram with tpl_ci == 0xb6 is found, then it will be discarded and not trigger any shell update.
Check processContent inside driver_apator162.cc

So, all should be good again for apator162 setups now. The really old telegrams are explicitly ignored, insteadof implicitly ignored as they were before. :-)

If someone wants to decipher the apt_07_05 telegrams then they are free to do so. Just improve processContent.
Unfortunately I do not have the time to do that. :-) Busy working with OMS compliant meters instead. 

Please test!

BIBOLV, 2023-01-02:
Closing due to inactivity. Please reopen issue if you want to continue to work on topic.

closed 2023-01-02 by BIBOLV