wmbusmeters/wmbusmeters #615: Is there initialization lacking or is there sound reasoning to mix data from different telegrams?

FredFirestone, 2022-09-17:
I'd like to read Qundis watermeters and internet led me to rtl_sdr and wmbusmeters. i'm fairly new to all of this (sorry, if I missed on a concept or use incorrect terminology) but I did manage to receive radio telegrams (`rtl_sdr -f 868.625M -s 1.6M - `) and extract detail values (`wmbusmeters --ignoreduplicates=true --format=json stdin:rtlwmbus ...`) as I intended.

I was however puzzled when receiving seemingly inconsistent data: A "newly" received telegram showed non current values. When extracting just  this one telegram, the non current values where actually null.

A  testcase is this:
> (echo "S1;1;1;2022-09-17 05:07:08.000;146;146;12345678;0x2d4465327856341218077a570000000c13665101004c1351810000426cbf2c02bb560000326cffff046d0606d129"; echo "S1;1;1;2022-09-17 09:02:08.000;145;146;12345678;0x244465327856341218077a580000008c04136548010082046cdf2801fd0c11046d010ad129") | /tmp/wmbusmeters --ignoreduplicates=true stdin:rtlwmbus RoomTest lse_07_17 12345678 NOKEY 2>&-
> (echo "S1;1;1;2022-09-17 09:02:08.000;145;146;12345678;0x244465327856341218077a580000008c04136548010082046cdf2801fd0c11046d010ad129"; echo "S1;1;1;2022-09-17 05:07:08.000;146;146;12345678;0x2d4465327856341218077a570000000c13665101004c1351810000426cbf2c02bb560000326cffff046d0606d129"; ) | /tmp/wmbusmeters --ignoreduplicates=true stdin:rtlwmbus RoomTest lse_07_17 12345678 NOKEY 2>&-

Output is in the first case:
`
RoomTest        12345678        15.166 m³       8.151 m³        2021-12-31      OK      2127-15-31      2022-09-17 06:06        2022-09-17 13:21.29
RoomTest        12345678        15.166 m³       8.151 m³        2021-12-31      OK      2127-15-31      2022-09-17 10:01        2022-09-17 13:21.29`

Whereas changing the order yields:
`
`RoomTest        12345678        null m³ null m³ null    null    null    2022-09-17 10:01        2022-09-17 13:39.17
RoomTest        12345678        15.166 m³       8.151 m³        2021-12-31      OK      2127-15-31      2022-09-17 06:06        2022-09-17 13:39.17
`

It's of course just a simplified test case but does show that telegrams are not interpreted independent: The actual telegram containing just the reading of 2022-08-31 is - depending on order - displayed with seemingly current values actually taken from a previous datagram.
 
This seems not correct to me, as it makes it fairly difficult to distinguish between "real" values and values just taken from an arbitrary previous telegram. I'd also argue it would lead to wrong interpretations - in the testcase one might wrongly conclude, no water was consumed between 06:06 h and 10:01 h.

To me it seems as if some initialization is lacking. I'm sorry, if there is some sound reason to it (being entirely possible) I hadn't thought of.

wmbusmeters --version yields
`wmbusmeters: 1.9.0-25-gc113f0e
c113f0e665e2b9e62b4510b989e397bab714b85c
`

weetmuts, 2022-09-17:
wmbusmeters maintains the meter state in between telegrams, this is necessary since several meters
send multiple different telegrams each containing parts of the total state. In this case you have the 
shorter telegram, which only contains dates! If unknown purpose to us, check the analysis:
https://wmbusmeters.org/analyze/244465327856341218077a580000008c04136548010082046cdf2801fd0c11046d010ad129

And then you have the longer telegram with readings:
https://wmbusmeters.org/analyze/2d4465327856341218077a570000000c13665101004c1351810000426cbf2c02bb560000326cffff046d0606d129

So everything behaves as expected, except that you didnt expect your meter to send telegrams without consumption.
:-)

weetmuts, 2022-09-17:
There is a consumption value in the short telegram but we do not know what storage 8 is used for.

FredFirestone, 2022-09-18:
Very fast response, thank you.

Thank you for pointing out different meters behaving different and thus wmbusmeters merging data from different telegrams.

Thank's for the analyze hint. I've used it before to find the correct driver, but the web service is cool, too. As for the fields I'm using json and reformat using jq.

I didn't see "storage 8" at a quick glance but output in json is quite self explanatory:
`{"media":"water","meter":"lse_07_17","name":"RoomTest","id":"12345678","total_m3":null,"due_date_m3":null,"due_date":null,"what_date_m3":14.865,"what_date":"2022-08-31","error_code":null,"error_date":null,"device_date_time":"2022-09-17 10:01","meter_version":"11","timestamp":"2022-09-18T19:50:38Z","device":"rtlwmbus[]","rssi_dbm":145}
{"media":"water","meter":"lse_07_17","name":"RoomTest","id":"12345678","total_m3":15.166,"due_date_m3":8.151,"due_date":"2021-12-31","what_date_m3":14.865,"what_date":"2022-08-31","error_code":"OK","error_date":"2127-15-31","device_date_time":"2022-09-17 06:06","meter_version":"11","timestamp":"2022-09-18T19:50:38Z","device":"rtlwmbus[]","rssi_dbm":146}`

My meter sends information only a few times a day and my receiver seemingly misses some, maybe due to inference. Thus for my use case it its unfortunate wmbusmeters outputs a "old" value from several hours ago (and thus implying "non consumtion") when actually it just can't know the current value as what it received was a periodic frame of previous "end of month" values. 

It might be worthwhile to think of a flag to turn off aggregation of values / internal state. On the other hand maybe not many users are interested in tracking not only consumption but also exact reading dates.

As the issue may be viewed as "works as designed", do as you deem fit and close if you please.

weetmuts, 2022-09-19:
Yes, this is an unresolved question. Some users wants no json, unless all partial telegrams have arrived.
Some wants only json after the first consumption value has arrived. Your use case is new however.

We could keep track of when a value was most revently updated, ie:
"total_m3" : 123
"total_m3_timestamp" : "2022-03-13 15:43:12"

We could have a setting that will only relay json when certain fields are received in a telegram:
--relayonfieldsreceived=total_m3,target_m3

And/or we could have a setting:
--relaywaitforallfields # This would wait for all non-optional fields to receive a value before relaying.
--relayalways # This is the current behaviour.
--relayonlyimportantfields # Relay all data, received or null, but only when telegrams actually receive at least one of the important fields.

What do you think?

FredFirestone, 2022-09-19:
Thank you for asking, but take my answer with a grain of salt as my knowledge here is limited.

Having a timestamp like "total_m3_timestamp" would solve my issue. I'm not quite sure if the timestamp should be built at receiving time or if it should be taken from device_date_time (maybe as  total_m3_ddt).
Sorry if I'm nitpicking: But it might as well apply to more (all?) other values (like having a what_date_m3_timestamp or even a what_date_timestamp).
But then again it seems much easier to me having a "raw" flag circumventing state aggregation completely: Any and all data in a telegram is thus current as of device_data_time and was received at timestamp, no ambiguity there.

As to when should output occur: I have no clue whatsoever if there is a specification to the number and / or fields in a telegram or any other "proper" way to know when "all data is fully received" (like mandatory / optional fields). As my knowledge is limited my advice is very likely of limited value, too.

Making output decision configurable via switches would certainly be most versatile. Whether that is feasible (for all current and future devices, for devices with individual configuration, ...) and how well it could be abstracted and maintained is beyond me. As (wireless) telegrams may be dropped it could at times also lead to an unpreferable delay. And finally: If you were to wait for fields A and B: Is there any way to know which order you expect them? Like a meter sending a telegram with A followed by one with B, then pausing a long time before starting over again . If you by chance start with receiving telegram B (1) you may wait a long time and output as soon as you receive A (2), after which you receive B (3) and wait again for the next A (4) leading to always outputing non matching values. If on the other hand you don't wait for the next A at (3) there seems little difference to "relayalways".
With my little understanding I'd probably sticked to the current "relayalways". If all the detail to a telegram (like the timestamps) is unambiguously there, flexible and arbitrary filtering can occur externally (e.g. with an awk script), if neccessary.

kitor, 2022-09-23:
My two cents (as I recently found out that apator162 probably also have "multi part" data): 

IMO for preserving backward compatibility `--relay=always`, `--relay=waitforall` could be implemented, with `waitforall` as a default. Having timestamp for each field will add a ton of unnecessary data, not needed for many sensors.

With this approach, with `always` user is responsible for checking what keys exists and acting accordingly. And single timestamp is enough to describe all the data in given telegram.

This may be important if you have a very poor reception and you just miss some of the telegrams... in such a case part of data can be very outdated in existing behaviour.

I also feel this should be probably configurable per sensor config, not as a global switch. This would allow to fine tune it for every sensor of interest.

weetmuts, 2022-09-23:
An additional important point made by @kitor 

> Either way, in case value is not possible to parse, 
> IMO wmbusmeters shall either not call shell at all, 
> or add extra field that tells the script that parsing indeed failed.

weetmuts, 2022-09-23:
The apt_07_05 can be found with 441486
and the apator162 can be found with 440106

weetmuts, 2023-01-21:
This is an important question, I haven't figured out the solution yet though..

weetmuts, 2023-01-21:
An additional point. The meters can send data with all bits 1, ie 0xffff
this means that there is no valid data. This is typically used for the flow_temperature_c
when the water meter has been dry for a while.  Or for the error bit and error date
if there has been no errors....

Now, the default is OPTIONAL for all fields, which means that if target_m3 is missing
it will not be printed. For fields that are not-optional, then "target_m3":null will
be printed in the json. 

However right now if "target_m3" is 0xffff in the telegram, this will be printed
as a very negative value in the telegram instead. But should it be printed as null
or not printed at all.....

weetmuts, 2023-04-04:
Lets keep this open.

closed 2023-07-05 by github-actions[bot]