wmbusmeters/wmbusmeters #616: Command line usage is quite cryptic

travnick, 2022-09-19:
Command line usage is very confusing. Looks like sometimes `=` is mandatory, sometimes not. I't really hard to figure out proper format of parameters.

Not working:
```
$ wmbusmeters --device=stdin:rtlwmbus
Unknown option "--device=stdin:rtlwmbus"

$ wmbusmeters --device stdin:rtlwmbus
Unknown option "--device"
```
working
```
wmbusmeters stdin:rtlwmbus
```

Not working:
```
$ wmbusmeters --useconfig /etc --device stdin:rtlwmbus
Usage error: you must supply the pid file as the last argument to wmbusmetersd.

$ wmbusmeters --useconfig /etc --device=stdin:rtlwmbus
Usage error: you must supply the pid file as the last argument to wmbusmetersd.

$ wmbusmeters --useconfig=/etc --device stdin:rtlwmbus
Usage error: --useconfig=... can only be used in combination with:
--device= --listento= --exitafter= --oneshot= --logfile= --silent --normal --verbose --debug --trace

$ wmbusmeters stdin:rtlwmbus --useconfig=/etc
Usage error: you must supply the pid file as the last argument to wmbusmetersd.
```
working
```
wmbusmeters --useconfig=/etc --device=stdin:rtlwmbus

wmbusmeters --device=stdin:rtlwmbus --useconfig /etc

wmbusmeters --device=stdin:rtlwmbus --useconfig=/etc
```

weetmuts, 2022-09-21:
Yeah, so --device= has only a very specific use case. To override the config files.
In any other situation the devices are the first parameters after the options.
All options that take arguments are of the long form standard --xxx=yyy

Usage: wmbusmeters {option}+ [device]+ {meter quadruplets}+

I could add a helper test that will print a meaningful message if you do:
wmbusmeters --device=xyz
instead
of wmbusmeters xyz

I see why it fails to recognize --device=xyz it is because the --useconfig=zzz
will move into the parser that understands --device. :-) 

I should have used --overridedeviceinconfig=xyz instead of just  --device=xyz
oh well....

weetmuts, 2022-09-21:
Fix pushed. It now the options is "--overridedevice=xyz"  (the old one still works for a while.)
Better error message when supplying --overridedevice= or --device when not supplying --useconfig=
Detection of superfluous command line arguments when using --useconfig=xyz
this caused the strange " you must supply the pid file as the last argument to wmbusmetersd"
message, which should be fixed now.

Please test!

travnick, 2022-09-27:
Looks like working.

closed 2022-09-27 by travnick