wmbusmeters/wmbusmeters #620: Bug: Aborted (core dumped) when encoding with key

Schwonsen, 2022-09-27:
Used version:
```
Using: wmbusmeters: 1.9.0-34-gf5f4e83
f5f4e8350555f4bd06b431873a32476f93c63c97
```

Payload:
```
wmbusmeters --analyze=859A9D0F5DC2BAD679644E4FB6F9CE29 a944c5149564252200047a10009025a8431d8Bee683c2471d99f2B943e739f39aff1a257d056cBB97d6Ba1ce2a85c01455Bef04a0BB199582803174de070957378350c6419c5Bf499Bf8e089390B4aB4f610c0BB334ad2B810994Be18fcf0d8B12a3BBB3e8cBe90009df3d7cfc8B62d89923e1a0d207BBdd4B453491a00d3942fe1B8d0d03492ad8cce69c8968B157d6f25ad1a541aafB0787aae06d6c9Bf903f903fd0c05010002fd0B
```                                       

Output:
`Assertion failed: (found.count(&fi) == 0), function printMeter, file meters.cc, line 2082.`

With the latest Version the decoding of some 'Engelmann Sensors' meter payloads fails,  if i use a key.  
The last Version i tested with where the payload from the example was decrypted correctly was the Version: `wmbusmeters: 1.9.0-1-gaa2f602
aa2f602f611b162f724e33fe71c5d187ebc35cf7` with the output:
```
Auto driver  : sensostar
Best driver  : enercal 22/71
Using driver : sensostar 00/00
000   : a9 length (169 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : c514 dll-mfct (EFE)
004   : 95642522 dll-id (22256495)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 10 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 9025 tpl-cfg 2590 (synchronous AES_CBC_IV nb=9 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 6D vif (Date and time type)
019 C!: 002DCC29 ("meter_timestamp":"2022-09-12 13:00")
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 06 vif (Energy kWh)
025 C!: 06000000 ("total_kwh":6)
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : 13 vif (Volume l)
031 C!: DC020000 ("total_water_m3":0.732)
035   : 01 dif (8 Bit Integer/Binary Instantaneous value)
036   : FD vif (Second extension FD of VIF-codes)
037   : 17 vife (Error flags (binary))
038 C!: 00 ("current_status":"OK")
039   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
040   : 6C vif (Date type G)
041 C?: DF28
043   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
044   : 06 vif (Energy kWh)
045 C?: 00000000
049   : 84 dif (32 Bit Integer/Binary Instantaneous value)
050   : 01 dife (subunit=0 tariff=0 storagenr=2)
051   : 06 vif (Energy kWh)
052 C?: 00000000
056   : 84 dif (32 Bit Integer/Binary Instantaneous value)
057   : 02 dife (subunit=0 tariff=0 storagenr=4)
058   : 06 vif (Energy kWh)
059 C?: 00000000
063   : 84 dif (32 Bit Integer/Binary Instantaneous value)
064   : 03 dife (subunit=0 tariff=0 storagenr=6)
065   : 06 vif (Energy kWh)
066 C?: 00000000
070   : 84 dif (32 Bit Integer/Binary Instantaneous value)
071   : 04 dife (subunit=0 tariff=0 storagenr=8)
072   : 06 vif (Energy kWh)
073 C?: 00000000
077   : 84 dif (32 Bit Integer/Binary Instantaneous value)
078   : 05 dife (subunit=0 tariff=0 storagenr=10)
079   : 06 vif (Energy kWh)
080 C?: 00000000
084   : 84 dif (32 Bit Integer/Binary Instantaneous value)
085   : 06 dife (subunit=0 tariff=0 storagenr=12)
086   : 06 vif (Energy kWh)
087 C?: 00000000
091   : 84 dif (32 Bit Integer/Binary Instantaneous value)
092   : 07 dife (subunit=0 tariff=0 storagenr=14)
093   : 06 vif (Energy kWh)
094 C?: 00000000
098   : 84 dif (32 Bit Integer/Binary Instantaneous value)
099   : 08 dife (subunit=0 tariff=0 storagenr=16)
100   : 06 vif (Energy kWh)
101 C?: 00000000
105   : 84 dif (32 Bit Integer/Binary Instantaneous value)
106   : 09 dife (subunit=0 tariff=0 storagenr=18)
107   : 06 vif (Energy kWh)
108 C?: 00000085
112   : C1 dif (8 Bit Integer/Binary Instantaneous value storagenr=1)
113   : 33 dife (subunit=0 tariff=3 storagenr=7)
114   : 7F vif (Manufacturer specific)
115 C?: 2E
116   : 47 dif (64 Bit Integer/Binary Instantaneous value storagenr=1)
117   : A6 vif (Operating time hours)
118   : 4E combinable vif (DateTimeExceedsUpperLastEnd)
119 C?: 109FE8D75ADAE277
127   : 0C dif (8 digit BCD Instantaneous value)
128   : 06 vif (Energy kWh)
129 C!: 0000B66A ("total_kwh":78860000)
133   : 23 dif (24 Bit Integer/Binary Minimum value)
134   : 64 vif (External temperature 10⁻³ °C)
135 C?: DCF098

{
    "media":"heat",
    "meter":"sensostar",
    "name":"",
    "id":"22256495",
    "meter_timestamp":"2022-09-12 13:00",
    "total_kwh":78860000,
    "total_kwh":78860000,
    "total_water_m3":0.732,
    "current_status":"OK",
    "timestamp":"2022-09-27T10:31:02Z"
}
```

weetmuts, 2022-09-27:
Thanks! This is a weird telegram, it contains two instantaneous energy fields. the first at offset 25 encoded using 32 bits.
The second at offset 129 encoded using bcd, but with a broken bcd number, since it contains a letter.
There is also an illegal 0x7f dif.

I think the telegram has been corrupted from offset 112 and to the end. 

I have fixed so that wmbusmeters gracefully handles this telegram.

Here is the telegram decoded and pre-decrypted:
https://wmbusmeters.org/analyze/A944C5149564252200047A100090252F2F046D002DCC290406060000000413DC02000001FD1700426CDF28440600000000840106000000008402060000000084030600000000840406000000008405060000000084060600000000840706000000008408060000000084090600000085C1337F2E47A64E109FE8D75ADAE2770C060000B66A2364DCF0987FD410920600000000840F06000000002F2F2F2F2F03F903FD0C05010002FD0B

Do you have any idea why it looks like this? The decryption looks ok all the way to 2f2f2f and the length byte looks ok.

weetmuts, 2022-09-27:
What is the product name? Is it sensostar product using wmbus instead of mbus?

Schwonsen, 2022-09-28:
> What is the product name? Is it sensostar product using wmbus instead of mbus?

Hi thank you for the fix. 
The product name is 'Engelmann Sensostar I', it should use wmbus telegramms.

There is still one error, the attribute "total_kwh" is incorrectly interpreted as null in some telegrams. 

**'total_kwh':null payloads:**
https://wmbusmeters.org/analyze/a944c5149564252200047a5d009025aa5d3614331a4e5c3316ae3f016Bea63B1999e17029355a4ae0f498f936e876c1ad9B4a973642c10c384f6f74c7de852f811ffB8acBB36245015726d1a75a6ad2d94ff8e19a093f6e4e010178B41e26468ca8d560B5c571ff8e047ec7d1f3a72B06B63cdcB2B86ff1df9f6de62831d19B109c13438e7851aB5890d540c06de90628B4e8388381B6e0fc96d36d577e462036203fd0c05010002fd0B:auto:859A9D0F5DC2BAD679644E4FB6F9CE29

or

https://wmbusmeters.org/analyze/A944C5149564252200047AAC00902572135D2BF4F30FA1EE914037F58255B74237096D38F606A4F2DEA7C2D28502DFE86B89A3E48D8CDE258980B38064AC145CA1D0EDDD63B10112F60ABDDED73493AE1C21201F32CA69CEFE2700C58752149E54E3D1D57ACF9ECE777B003956E79E4F578F01799BEFFC3A19416FBE0DF5E64848ECFCE7E3C18B48A293C6526BBAD7FB7AB5619E8A032AFDACDF022F1D2FB603B603FD0C05010002FD0B:auto:859A9D0F5DC2BAD679644E4FB6F9CE29

**'total_kwh' correct:**
https://wmbusmeters.org/analyze/A944C5149564252200047AF900902581C4E44ED19FDD1D4962B67979D7420000D087CFD025BED32598910C8A676EF0AE67E5D10F11367F12FE466CC7053284158A13FE5664F27C10857780273BB3F129BDB968A01D1CA1FBABBF646D03AF080B6EF0B6065880CA9D83B2D458669598DE9CAE924917D704E64AEF68E279EC643F067BFF40A8BDB443BD9F0ACA6BA1D019F1F6B9A58A65C709A71FBCA3A38DB903B903FD0C05010002FD0B:auto:859A9D0F5DC2BAD679644E4FB6F9CE29

weetmuts, 2022-09-28:
Thanks! Your telegram where half of the data (offset 112 and to the end) is random triggers all sorts of interesting corner case. Either I am missing something important in how telegrams can be constructed, or there is something seriously wrong with your meter.

Anyway, the random data contained energy difvif pairs, but without actual data. I.e. 0000 or 0084 etc.
The sort order for which dv_entry to check first for a match against total_kwh was not specified so sometimes
it matched the no-data in the random part, and sometimes in the good part.

I added a sort order so your examples above now work.

But we really need to understand why data from offset 112 is so weird.

weetmuts, 2022-11-13:
Can we close this now?

Schwonsen, 2022-11-16:
Yes thank you.

closed 2022-11-16 by Schwonsen