wmbusmeters/wmbusmeters #621: device override does not work as expected

travnick, 2022-09-27:
There is an error `You can only specify one stdin or one file or one command!` when I try to use `--overridedevice=anything`/`--device=anything` with `--useconfig=/etc/` if the device is already defined in config file.

I know it's defined, I do not care since I want to override it :)

BIBOLV, 2023-01-02:
Working for me as described in readme - `wmbusmeters --useconfig=/home/me/.config/wmbusmeters --overridedevice=rtlwmbus`
Please double check and report if any issues

closed 2023-04-04 by github-actions[bot]

travnick, 2023-07-21:
do you have `device` defined in your configuration file?
I have `device=stdin:hex" in config file, and it fails with override
```
echo "3c4401068561940105077abc0030859168a14672a1c99fa77c5de302f6867ed95a14c1c5d9f2baafa28ebde817d103a151ce69b347e3a4561925083916d9750b15" | wmbusmeters  --useconfig=/etc --overridedevice="stdin:hex"
(config) "device" "stdin:hex"
(config) "logtelegrams" "false"
(config) "format" "json"
(config) "meterfiles" "/var/lib/wmbusmeters/meter_readings"
(config) "meterfilesaction" "append"
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log"
(config) "shell" "echo "$METER_JSON""
(config) "" ""
(config) loading meter file /etc/wmbusmeters.d/apator01946185.conf
(config) name=apator01946185
(config) type=apator162
(config) id=01946185
(config) key=<notprinted>
(config) overriding device with "stdin:hex"
You can only specify one stdin or one file or one command!
```

weetmuts, 2023-07-21:
@travenick Ah, yes, normally the  config has the daemon settings (like im871a:c1,t1 or something)
because stdin:hex does not make sense for the daemon. Since stdin is closed for the daemon. :-)
You can either just not use --overridedevice or remove device=stdin:hex in the config file. Either works.

However, the error is a bit counter-intuitive so I would like to fix/improve the use case.

Can you explain your use case to me first?

travnick, 2023-07-24:
I have a config file with defined values. And since I play with an external python daemon which listens on MQTT and executes wmbusmetters (with stdin data) to decode incoming readings from rtl_433.

In this case, I have to remove device from the config file because of the error.
Such override is also useful for other testing purposes, where one would like to provide data by hand and see the results without having to stop daemon and mess up with configuration file