wmbusmeters/wmbusmeters #625: mystery QHeat

MightyPork, 2022-09-29:
`41449344796550674637727965506793444604dc0000200c0d000000004c0d00000000426cffffcc080d00000000c2086cdf2802fd170000326cffff046d3a0ddb29`

67506579 mfct: (QDS) Qundis, Germany (0x4493) type: Heat meter (0x04) ver: 0x46

I should have more details about this meter next week.

It was supposed to be ENBRA Sharky 744 but I have some doubts if they didn't install something else. It also should be encrypted but isn't.

MightyPork, 2022-09-30:
Ok I have a photo, it's QHeat 5.5, but some different version than what is supported. Should have more info soon...

[image]

MightyPork, 2022-10-07:
I now have the super secret specification, i'll try to compare it with the current driver...

MightyPork, 2022-10-18:
here's a datasheet: https://qundis.it/wp-content/uploads/2022/03/Data_sheet_Qheat5.5US_R_DST1-00HE-GB0-USFU1-A-220210.pdf

weetmuts, 2022-10-24:
I pushed an update to the qheat driver which seemed to be the best match for your telegram.
Thanks for the heads up on the problem with the Joule detection! It is fixed now.

Please test!

MightyPork, 2022-10-24:
Tested, works great! The consumption is now converted to kwh and reported as kwh. Thanks for fixing it :+1:

closed 2022-10-24 by MightyPork