wmbusmeters/wmbusmeters #626: Telegram (maybe) incorrectly attributed to the wrong device

Contik, 2022-09-29:
Hi,

I've been running `wmbusmeters` in version `1.9.0-3-g27a77b5` in its Docker Hub container flavor as `weetmuts/wmbusmeters:latest` with SHA-256 digest `weetmuts/wmbusmeters@sha256:639b4ea19f1aaf8ef308084193314d9ddd1b4e27cab8ed4515224d1c570c920f` from image creation timestamp Tuesday, September 6, 2022 at 8:28:15 pm UTC for a couple of weeks now.

Today `wmbusmeters` apparently received a telegram that by all intents and purposes was not sent by one of the meter for which I have meter files in `wmbusmeters.d`, however `wmbusmeters` apparenty treated the telegram as if it belonged to one of my meter files and effectively passed an outlier through the meter file's `shell=` command.

For context my `wmbusmeters.conf` reads:
```
loglevel=normal
device=auto
listento=c1,t1,s1
logtelegrams=true
logtimestamps=always
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
```

From `/wmbusmeters_data/logs/wmbusmeters.log` the relevant section of telegrams reads:
```
[2022-09-29_05:44:44] telegram=|33446850637985029480A2_0F9F2B9301D013130111340AD70B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1662917
[2022-09-29_05:48:52] telegram=|33446850637985029480A2_0F9F2B9301D013130111330AD60B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1663165
[2022-09-29_05:52:55] telegram=|33446850637985029480A2_0F9F2B9301D013130111360AD70B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1663408
[2022-09-29_06:04:51] telegram=|294468506379850294087A650010252F2F_27896D7A0345E2E692C7157A301D0F0FDEA25BBD91D109D693|+1664124                      <--- Outlier
[2022-09-29_06:05:02] telegram=|33446850637985029480A2_0F9F2B9301D0131301113B0AD80B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1664135
[2022-09-29_06:09:10] telegram=|33446850637985029480A2_0F9F2B9301D0131301113A0AD60B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1664383
[2022-09-29_06:13:13] telegram=|33446850637985029480A2_0F9F2B9301D0131301113D0AD70B010000000000000000000E101F20261E1D1F191C221E1B20221F02|+1664626
```
Where the middle telegram according to `wmbusmeters --analyze '<telegram_string>'` turns into:
```
Auto driver  : fhkvdataiv
Best driver  : hydroclima 08/18
Using driver : fhkvdataiv(driver should be upgraded) 00/00
000   : 29 length (41 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 6850 dll-mfct (TCH)
004   : 63798502 dll-id (02857963)
008   : 94 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 65 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 1025 tpl-cfg 2510 (synchronous AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
015   : 2f2f already decrypted check bytes
017   : 27 dif (64 Bit Integer/Binary Minimum value)
018   : 89 vif (Energy 10¹ J)
019   : 6D combinable vif (None)
020 C?: 7A0345E2E692C715
028   : 7A dif (4 digit BCD Value during error state storagenr=1)
029   : 30 vif (Power J/h)
030 C?: 1D0F
032  ?: 0F manufacturer specific data DEA25BBD91D109D693

{
    "media":"heat cost allocation",
    "meter":"fhkvdataiv",
    "name":"",
    "id":"02857963",
    "current_consumption_hca":0,
    "set_date":"",
    "consumption_at_set_date_hca":0,
    "set_date_1":"",
    "consumption_at_set_date_1_hca":0,
    "set_date_17":"",
    "consumption_at_set_date_8_hca":0,
    "error_date":"",
    "device_date_time":"",
    "timestamp":"2022-09-29T21:08:48Z"
}
```
The analysis lists `dll-id` as `02857963` which actually is the exact ID from my meter file. The meter file in question reads:
```
name=living-room-1-rhca-1-gen-3.iot.location.example.de
id=02857963
key=NOKEY
driver=fhkvdataiii
shell=value="$(printf -- '%s' "${METER_JSON}" | sed -r -e 's'$'\x1''"'$'\x1''\\"'$'\x1''g')"; data='{"request":"sender data","data":[{"host":"living-room-1-rhca-1-gen-3.iot.location.example.de","key":"wmbus.meter.raw","value":"'"${value}"'"}]}'; datalen=$(printf "%08x" ${#data}); datalen="\\x${datalen:6:2}\\x${datalen:4:2}\\x${datalen:2:2}\\x${datalen:0:2}"; printf "ZBXD\1${datalen}\0\0\0\0%s" "$data" | nc zabbix-server-1.back.location.example.de 10051
```
Where I'm essentially piping the `printf` output of any given `${METER_JSON}` into `nc` which transmits the payload (in a slightly massaged format for protocol compatibility) into a Zabbix monitoring instance.

The device in question is a Techem Energy Services electronic heating cost allocator taped to a radiator. The vendor Techem has a datasheet on the pages of their [Techem Middle East and North Africa](https://www.techem.com/ae/en/about-us/techem-mena) subsidiary aka Techem MENA, see [PDF file "Electronic_heating_cost_allocator_with_radio.pdf" via techem.com](https://www.techem.com/content/dam/techem-ae/documents/Electronic_heating_cost_allocator_with_radio.pdf.coredownload.pdf).

Zabbix saw this packet comin as (prettified for readability):
```
{
    "media": "heat cost allocation",
    "meter": "fhkvdataiii",
    "name": "living-room-1-rhca-1-gen-3.iot.location.example.de",
    "id": "02857963",
    "current_hca": 37606,                                          <--- Way off
    "current_date": "2022-01-04T02:00:00Z",
    "previous_hca": 890,                                           <--- Way off
    "previous_date": "2054-12-09T02:00:00Z",                       <--- Way off
    "temp_room_c": 312.53,                                         <--- Way off
    "temp_radiator_c": 74.72,                                      <--- Way off
    "timestamp": "2022-09-29T04:04:51Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 125
}
```
Where the `id` is correct at `02857963` and matches both the meter file and all other prior and later readings but other values are way off. A normal telegram a few seconds later looks like this in JSON format (also prettified for readability):
```
{
    "media": "heat cost allocator",
    "meter": "fhkvdataiii",
    "name": "living-room-1-rhca-1-gen-3.iot.location.example.de",
    "id": "02857963",
    "current_hca": 275,
    "current_date": "2022-09-29T02:00:00Z",
    "previous_hca": 403,
    "previous_date": "2021-12-31T02:00:00Z",
    "temp_room_c": 26.19,
    "temp_radiator_c": 30.32,
    "timestamp": "2022-09-29T04:05:02Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 117
}
```
Which creates a statistical outlier like so:
[image]

Was this a corrupt (partially transmitted?) telegram? Is this misbehavior or a bug maybe? Is there anything I can do to make this not happen?

Thanks you so much and I appreciate all the effort that goes into `wmbusmeters`.

Kind regards

weetmuts, 2022-09-30:
Very weird. Which wmbus device do you use to receive the telegrams?

weetmuts, 2022-09-30:
Is this happening regularly ?

weetmuts, 2023-01-21:
Can you print these three telegrams in full here:
# [2022-09-29_05:52:55] [Expected interval] telegram=|334468...|+1663408
# [2022-09-29_05:56:59] [No telegram here ]
# [2022-09-29_06:01:00] [No telegram here ]
_ [2022-09-29_06:04:51]                     telegram=|294468...|+1664124  <--- Outlier
# [2022-09-29_06:05:02] [Expected interval] telegram=|334468...|+1664135

So that we can see the contents of the outlier.

Contik, 2023-04-28:
I do not have these logs anymore, will keep my eyes peeled and check back in with more detailed telegrams in case I observe this again. Sorry for letting this get stale.

closed 2023-07-29 by github-actions[bot]