wmbusmeters/wmbusmeters #627: Waterstarm showing incorrect meter_timestamp

JensHoRi, 2022-09-29:
Today (30th of september 0:00 until 0:05) some of my meters configured as waterstarm where read with meter_timestamp
2022-10-00 00:00 or 2022-10-00 00:05, depending on the exact time

But 0:00 of Day 0 of Month Oktober is not a valid date.

Could this be a failure in wmbusmeters or is it a wrong telegram (bug in firmware of meters)?

weetmuts, 2022-10-08:
Do you have an example telegram?

JensHoRi, 2022-10-08:
Because it was only seen the first minutes after midnight sadly not :(

weetmuts, 2022-10-08:
See if you can enable logtelegrams to catch one. Its hard to tell what the problem is otherwise.

Here is an example:
https://wmbusmeters.org/analyze/3944FA122162992202067A360420252F2F046D282A9E2704136A00000002FD17400004933C000000002F2F2F2F2F2F03FD0C08000002FD0B0011

The datetime:
282A9E27 ("meter_timestamp":"2020-07-30 10:40")

which is decoded in dvparser.cc bool DVEntry::extractDate(struct tm *out)
which goes into v.size() == 4 which does

       ok &= ::extractDate(v[3], v[2], out);
       ok &= ::extractTime(v[1], v[0], out);

And then in extractDate
    int day   = (0x1f) & lo; 

and in extractTime:
    int min  = (0x3f) & lo;
    int hour = (0x1f) & hi;

So the code assumes that the number is actually starting 1.
However if it starts with 0 then the code should add one to day etc.

I have to check the spec.

weetmuts, 2022-10-08:
The spec says that the meter should store 1-31 in the day bits.
since the number 0 is used to signify every day. 

The minute should be encoded as 0-59 (63 means every minute) 
and hour as 0-23 (31 means every hour)

Maybe there is a bug in the meter?

JensHoRi, 2022-10-08:
Normally date is fine. I only saw this during change of month the first few minutes about midnight. Maybe I can watch at 31 of October again

JensHoRi, 2022-10-08:
Just looked: oldest log was still there:

waterstarm) 019 C!: 0020C02A ("meter_timestamp":"2022-10-00 00:00")

JensHoRi, 2023-01-27:
I'll recheck on 1 of march, because until yet I never saw this again, but otherwise December and January both have 31 days .-/

JensHoRi, 2023-02-06:
Just for information: 
Saw this on 1st of february directly after midnight again:

(waterstarm) 019 C!: 0420E022 ("meter_timestamp":"2023-02-00 00:04")

some later in log

(meters) render field meter_timestamp(Text txt)[0] with dventry @19 key 046D data 0420E022
(meters)           "meter_timestamp":"2023-02-00 00:04"

so seems to be every change of month directly after midnight.

See another meter same minute showing 0420E122 then shown as correct timestamp 2023-02-01 00;04

Very interesting: All meters are from same manufacturer...

Curious

closed 2023-05-08 by github-actions[bot]