wmbusmeters/wmbusmeters #632: dongle gets extremely hot and breaks

joergbeicht, 2022-10-04:
Hello (help, help, help),
at first, wmbusmeter docker works fine.
My dongle gets extremely hot and breaks.
So far these two have been destroyed (found in wmbusmertrs.log):
DEV=rtlwmbus[00000001]
DEV=im871a[00103086]

What settings do I have to avoid this?
Thank you in advance for your help.

Jörg

weetmuts, 2022-10-08:
The rtlsdr dongle can get hot, do you have one of those that are in plastic or metal? The metal ones probably are better do dissipate the heat.

The im871a dongle should not get hot at all.....is it heated by the other dongle?

joergbeicht, 2022-10-08:
Good morning,
Thank you for your answer.
That helps me alot.

Both dongles were on a rasp pi4 with wmbusmeters docker.
The pi4 is on my desk. The only thing on the pi4 is the dongle and the network cable.

This is my rtl dongle:
https://www.amazon.de/gp/product/B01B4L48QU/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

The im871a alone on the USB port
https://www.piotek.de/iM871A?googlede=15732&wsa=h922j4h2hcad6&gclid=Cj0KCQjwnP-ZBhDiARIsAH3FSRcI-aTO5z7XKeOeRNHlXmf_PWQVjl9EQAO-L52cyp5yZGV5r9JYQBAaAqHSEALw_wcB

weetmuts, 2022-10-13:
I bought the rtlsdr dongle the NESDR Nano 2+ that you linked to, got it today.
So small and looks nice! But it gets really hot!

I ran it for 7 minutes and then the plastic was quite warm to the touch and the usb contact metal was really hot!
It even heats up my laptop from the inside where it is connected... yikes.

Tried my other rtlsdr dongles and none of them was even close to generating the same heat after 7 minutes.

I suppose it depends on the computer chassis wether it can dissipate the heat. The raspberry might not have
enough metal to get rid of the heat so it heats up the other dongle through the usb contact and everything breaks.

https://www.reddit.com/r/stratux/comments/400tja/heatsink_for_nooelec_nesdr_nano_2/

Get another dongle. :-)

BIBOLV, 2023-01-02:
Closing as answer was provided. Please reopen issue if any further help is needed.

closed 2023-01-02 by BIBOLV