wmbusmeters/wmbusmeters #634: Can't get readings out of my Multical21

AythamiEsp, 2022-10-10:
Device: Raspberry PI 3b+, im871a, Kamstrup Multical21
Location: Belgium, Antwerpen

Hi, These meters have been installed for a couple of years in Flanders or Antwerp by the company Water-Link or Waterlink.
I've been at this over a week and can't find any multical21 around. 
Installed wmbusmeters on a Rpi3b+ got the stick running fine, I've been finding some kind of smart gas meter from (FLO) Flonidan, Denmark (0x198f).  
And as the multical21 are mandatory all the houses in my vicinity should have them and got nothing on the "radar".

Today I've been looking more closely at the type and it says 02113COE862

As I understand the meter uses Sigfox, don't know what that is but seems to only send the values once a day...
Seems like a RF communication

Can this work with wmbusmeters?

ElKaPa, 2022-10-10:
Sigfox is a different transmission system. Kamstrup made these special Multical21 for Water-Link

https://www.kamstrup.com/en-en/news-and-events/news/sigfox-mc21

AythamiEsp, 2022-10-10:
Yeah, apparently so. Also works in the same spectrum: 868MHz. Matybe something can be programmed for it.

Checked the watermeter at my brothers house and he has the 02140COG867. So thisone does work with C1 telegram.
Would the deployment of the mandatory watermeters be with mixed Multical21 types?
I'll get there soon to check for C1 telegrams, if so maybe I can convince a technician to swap it...

Can anything be done with the Sigfox signals?

BIBOLV, 2023-01-02:
Unfortunately sigfox signals can`t be processed using wmbusmeters. Unless payload is wmbus message that are transmitted using sigfox, but it is really doubtful. Closing issue for now.

closed 2023-01-02 by BIBOLV