wmbusmeters/wmbusmeters #635: Numer of Kamstrup Multical21 fields are missing

raszuk, 2022-10-12:
Hi, 

Is there any plan to add available Multical21 reported fields like max_water_temp_month, min_water_temp_month, reverse_volume, remaining battery for either C1 or T1 telegrams ? 

[image]
[image]

Thx,R.

raszuk, 2022-10-12:
> Absolutely! I just need example telegrams. Can you produce some telegrams
> with all the different possible fields?

I can produce all telegrams listed if it is ok to have fields with empty
data.

How should I capture those telegrams so they are decoded for your in the
best way ?

Many thx,
R.

weetmuts, 2022-10-12:
Yep, empty fields are ok. You can collect and predecrypted telegrams like this:

wmbusmeters --logtelegrams auto:c1 Water multical21 meterid meteraeskey
you will get lines like:
telegram=|.......|

These are the predecrypted telegrams that you can share here, without having to worry about the key.

If you can set a temporary key that you can share here, then you can also do just:
wmbusmeters --logtelegrams auto:c1 
Then the telegram=|...|  lines are encrypted and you need to post the temp key here as well.

Both methods work.

Since the multical21 sends both compressed and uncompressed telegras, please collect both types!

raszuk, 2022-10-12:
Just to confirm before I spent more time later this week to capture all
modes. Is this below two lines sufficient for each mode ?

Started auto rtlwmbus[77771111153705700] listening on c1
telegram=|2E442D2C654864771B167AC61020252F2F|02FF20710004134E060000046D3B34CC2A51671B04933C040000002F2F2F|+7
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:09.37

Thx !

weetmuts, 2022-10-12:
That is a full telegram, you can check its contents here:
https://wmbusmeters.org/analyze/2E442D2C654864771B167AC61020252F2F02FF20710004134E060000046D3B34CC2A51671B04933C040000002F2F2F

The compressed telegram is shorter, here is a compressed telegram from my meter.
https://wmbusmeters.org/analyze/23442D2C998734761B168D2087D19EAD217F1779EDA86AB6710008190000081900007F13

Notice that there are no field indicators/separators. These have been collected from a previous full telegram.
The format signature crc is used to identify the set of mbus-content expected.

In this case, wmbusmeters can decode the compressed telegram simply because I added some
known format signatures to it. Normally you have to wait for a full telegram before wmbusmeters can 
start decoding the compressed telegrams.

weetmuts, 2022-10-12:
Every 8th telegram is a full telegram, the rest compressed. At least for my multical21.

raszuk, 2022-10-12:
I am really not seeing any other telegram type ...

***@***.***:~/kamstrup$ wmbusmeters --logtelegrams auto:c1 Water
multical21 77644865 6A01AD9FF79981F8BF90044087544D89
Started auto rtlwmbus[77771111153705700] listening on c1
telegram=|2E442D2C654864771B167AF81020252F2F|02FF20710004134E060000046D0D35CC2A51671B04933C040000002F2F2F|+13
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:23.09
telegram=|2E442D2C654864771B167AF91020252F2F|02FF20710004134E060000046D0D35CC2A51671B04933C040000002F2F2F|+29
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:23.25
telegram=|2E442D2C654864771B167AFA1020252F2F|02FF20710004134E060000046D0D35CC2A51671B04933C040000002F2F2F|+46
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:23.42
telegram=|2E442D2C654864771B167AFB1020252F2F|02FF20710004134E060000046D0D35CC2A51671B04933C040000002F2F2F|+62
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:23.58
telegram=|2E442D2C654864771B167AFC1020252F2F|02FF20710004134E060000046D0E35CC2A51671B04933C040000002F2F2F|+79
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:24.15
telegram=|2E442D2C654864771B167AFD1020252F2F|02FF20710004134E060000046D0E35CC2A51671B04933C040000002F2F2F|+95
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:24.31
telegram=|2E442D2C654864771B167AFE1020252F2F|02FF20710004134E060000046D0E35CC2A51671B04933C040000002F2F2F|+112
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:24.48
telegram=|2E442D2C654864771B167AFF1020252F2F|02FF20710004134E060000046D0E35CC2A51671B04933C040000002F2F2F|+128
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:25.04
telegram=|2E442D2C654864771B167A001020252F2F|02FF20710004134E060000046D0F35CC2A51671B04933C040000002F2F2F|+145
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:25.21
telegram=|2E442D2C654864771B167A011020252F2F|02FF20710004134E060000046D0F35CC2A51671B04933C040000002F2F2F|+161
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:25.37
telegram=|2E442D2C654864771B167A021020252F2F|02FF20710004134E060000046D1035CC2A51671B04933C040000002F2F2F|+178
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:25.54
telegram=|2E442D2C654864771B167A031020252F2F|02FF20710004134E060000046D1035CC2A51671B04933C040000002F2F2F|+194
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:26.10
telegram=|2E442D2C654864771B167A041020252F2F|02FF20710004134E060000046D1035CC2A51671B04933C040000002F2F2F|+211
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:26.27
telegram=|2E442D2C654864771B167A051020252F2F|02FF20710004134E060000046D1135CC2A51671B04933C040000002F2F2F|+227
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:26.43
telegram=|2E442D2C654864771B167A061020252F2F|02FF20710004134E060000046D1135CC2A51671B04933C040000002F2F2F|+244
Water   77644865        1.614 m3        0 m3    0 m3/h  127 °C  27 °C
DRY(dry 22-31 days)     2022-10-12 22:27.00

raszuk, 2022-10-12:
Hi,

So with different meter I see exactly as you have described. But running
into different problem - online tool you sent URLs to is failing to decode
using Kamstrup provided AES key :(

***@***.***:~/kamstrup$ wmbusmeters --logtelegrams auto:c1 Water
multical21 77332649 7A5DCAC91991878266C6E29E89724D6E
Started auto rtlwmbus[77771111153705700] listening on c1
(meter) newly created meter (Water 77332649 multical21) did not handle
telegram!

*telegram=|2E442D2C492633771B168D28A611252320E7C678|02FF201100041307000000623B0000615B7F515B8161671651671A|+59Water
  77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C   DRY(dry
1-8 hours)      2022-10-12 23:13.38*
telegram=|23442D2C492633771B168D28A712252320079F79327F8E8D|11000700000000007F81161A|+74
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:13.53
telegram=|23442D2C492633771B168D28A813252320079F79327F8E8D|11000700000000007F81161A|+90
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:14.09
telegram=|23442D2C492633771B168D28A920252320079F79327F8E8D|11000700000000007F81161A|+106
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:14.25
telegram=|23442D2C492633771B168D28AA21252320079F79327F8E8D|11000700000000007F81161A|+122
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:14.41
telegram=|23442D2C492633771B168D28AB22252320079F79327F8E8D|11000700000000007F81161A|+138
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:14.57
telegram=|23442D2C492633771B168D28AC23252320079F79327F8E8D|11000700000000007F81161A|+153
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:15.12
telegram=|23442D2C492633771B168D28AD30252320079F79327F8E8D|11000700000000007F81161A|+169
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:15.28

*telegram=|2E442D2C492633771B168D28AE31252320E7C678|02FF201100041307000000623B0000615B7F515B8161671651671A|+185Water
  77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C   DRY(dry
1-8 hours)      2022-10-12 23:15.44*
telegram=|23442D2C492633771B168D28AF32252320079F79327F8E8D|11000700000000007F81161A|+201
Water   77332649        0.007 m3        0 m3    0 m3/h  129 °C  26 °C
DRY(dry 1-8 hours)      2022-10-12 23:16.00

weetmuts, 2022-10-13:
Yep, last examples are compressed telegram! :-)
https://wmbusmeters.org/analyze/23442D2C492633771B168D28AF32252320079F79327F8E8D11000700000000007F81161A

When using the online service it cannot be decoded, but it decodes fine as you can see in your log, as long
as wmbusmeters has received the long telegram before. :-)

The telegrams you provided are great. Can you just briefly comment on each telegram type what the selected fields were in the meter config? 

For example in this telegram: https://wmbusmeters.org/analyze/2E442D2C492633771B168D28A611252320E7C67802FF201100041307000000623B0000615B7F515B8161671651671A

It is probably the case that it sends the max/min for flow/extern temperature, but is it the average over the previous month?

And here.
https://wmbusmeters.org/analyze/2E442D2C654864771B167AF81020252F2F02FF20710004134E060000046D0D35CC2A51671B04933C040000002F2F2F

Is it sending the meters current datetime? Total reverse flow accumulated?

I am not sure why the online tool does not decrypt for you. It is the same wmbusmeters binary, just running in the cloud. :-) If you want, you could post some encrypted telegrams and I can test.

raszuk, 2022-10-13:
Hi,

> When using the online service it cannot be decoded, but it decodes fine as
> you can see in your log, as long
> as wmbusmeters has received the long telegram before. :-)
>
I can see how compressed telegrams are cool for permanent reading of the
meters. But how do they help for walk by reading ? I presume the operator
must wait anyway at each readout location to get full telegram ...

 > Can you just briefly comment on each telegram type what the selected
fields were in the meter config?

Of course. I was just preparing here to capture useful data. I am not near
meter today but perhaps till next week I will report some more.

I am not sure why the online tool does not decrypt for you. It is the same
> wmbusmeters binary, just running in the cloud. :-) If you want, you could
> post some encrypted telegrams and I can test.
>

It all works fine. My mistake. I was confused about encrypted and decrypted
telegrams. Was pasting wrong one.

Many thx,
Robert

Message ID: ***@***.***>
>

weetmuts, 2022-10-13:
> I can see how compressed telegrams are cool for permanent reading of the
> meters. But how do they help for walk by reading ? I presume the operator
> must wait anyway at each readout location to get full telegram ...

The format signature is independent of the meter identity. So you only
need to recognize the meter type+version+format_sig to know the layout of the data.

Either you can collect and store some format signatures in advance 
(which wmbusmeters does for a few types) or you wait for a single full 
telegram, then you do the drive by and you can decode each compressed telegram.

raszuk, 2022-10-19:
Hello, 

Sorry for delay. I prepared three modes 40 C1, 41 T1 OMS and 48 C1  in the attached files. 
[telegrams_Kamstrup_Mode40.pdf](https://github.com/weetmuts/wmbusmeters/
[telegrams_Kamstrup_Mode41.pdf](https://github.com/weetmuts/wmbusmeters/files/9822219/telegrams_Kamstrup_Mode41.pdf)
files/9822218/telegrams_Kamstrup_Mode40.pdf)
[telegrams_Kamstrup_Mode48.pdf](https://github.com/weetmuts/wmbusmeters/files/9822221/telegrams_Kamstrup_Mode48.pdf)

Thank you,
R

raszuk, 2022-10-21:
Adding few more captured telegrams from different modes: 

[https://drive.google.com/drive/folders/1xhSwvoFHXsu4UF8CxGhgBUlk-z4_6Nlx?usp=sharing](https://drive.google.com/drive/folders/1xhSwvoFHXsu4UF8CxGhgBUlk-z4_6Nlx?usp=sharing)

raszuk, 2023-03-05:
Hi, 

I provided all requested on Oct 12'2022 captures from Kamstrup meters. I am under impression that those were requested to be added to wmbusmeters decoder and would be included in general distribution. 

Please kindly indicate if this is not the case ? 

Requested quote: 

> [weetmuts](https://github.com/weetmuts) commented [on Oct 12, 2022]
> Absolutely! I just need example telegrams. Can you produce some telegrams with all the different possible fields?

weetmuts, 2023-03-05:
It is indeed on the todo list. Thanks for the telegrams! The bot is automatic.

raszuk, 2023-03-05:
Excellent. Thank you !!!! I was just afraid the bot will close this issue :)

raszuk, 2023-05-06:
Keepalive msg :) To prevent stale status. 

It would be awesome to get some ETA when the described extensions will be added. 

many thx !

closed 2023-08-06 by github-actions[bot]