wmbusmeters/wmbusmeters #636: Cannot install HA Addon

shaggee, 2022-10-13:
When Im trying to install the addon from the HA addon list, it cannot be installed.
Error info:
 Failed to to call /addons/cada3e3b_wmbusmeters/install - The command '/bin/ash -o pipefail -c make' returned a non-zero code: 2 

`Step 8/30 : RUN make
 ---> Running in 9a0148ba34f7
New version number generates new build/version.h
Building 1.9.0-68-g9152e28
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aes.cc -c -E > build/aes.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aes.cc -MMD -c -o build/aes.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aescmac.cc -c -E > build/aescmac.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/aescmac.cc -MMD -c -o build/aescmac.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/bus.cc -c -E > build/bus.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/bus.cc -MMD -c -o build/bus.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/cmdline.cc -c -E > build/cmdline.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/cmdline.cc -MMD -c -o build/cmdline.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/config.cc -c -E > build/config.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/config.cc -MMD -c -o build/config.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/dvparser.cc -c -E > build/dvparser.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/dvparser.cc -MMD -c -o build/dvparser.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/formula.cc -c -E > build/formula.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/formula.cc -MMD -c -o build/formula.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/mbus_rawtty.cc -c -E > build/mbus_rawtty.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/mbus_rawtty.cc -MMD -c -o build/mbus_rawtty.o
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/meters.cc -c -E > build/meters.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/meters.cc -MMD -c -o build/meters.o
src/meters.cc: In member function 'void MeterCommonImplementation::addNumericFieldWithCalculator(std::string, std::string, PrintProperties, Quantity, std::string, Unit)':
src/meters.cc:982:13: error: converting to 'bool' from 'std::nullptr_t' requires direct-initialization [-fpermissive]
  982 |             ));
      |             ^
In file included from src/wmbus.h:21,
                 from src/config.h:23,
                 from src/bus.h:21,
                 from src/meters.cc:18:
src/dvparser.h:388:23: note:   initializing argument 1 of 'FieldMatcher::FieldMatcher(bool)'
  388 |     FieldMatcher(bool act) : active(act) { }
      |                  ~~~~~^~~

make: *** [Makefile:130: build/meters.o] Error 1

Removing intermediate container 9a0148ba34f7`

RiddleMan, 2022-10-13:
I experienced the same behavior.

weetmuts, 2022-10-13:
Fix pushed. I hope, since this error does not show up with my compiler, g++ (Ubuntu 11.2.0-19ubuntu1) 11.2.0.

In my compiler version the NULL was translated into a bool false, before used to create a FieldMatcher(bool b).

So, good thing that the latest compilers catches this nowadays, since I did use NULL because I thought I was feeding a ptr.... :-/ but it accidentally worked because the false generated the expected FieldMatcher object. Yay.

RiddleMan, 2022-10-13:
I confirm it works! Thanks a lot!

shaggee, 2022-10-13:
> Fix pushed. I hope, since this error does not show up with my compiler, g++ (Ubuntu 11.2.0-19ubuntu1) 11.2.0.
>
> In my compiler version the NULL was translated into a bool false, before used to create [...]

Tried but there is another problem:

22-10-13 21:31:32 INFO (SyncWorker_1) [supervisor.docker.addon] Starting build for cada3e3b/amd64-addon-wmbusmeters:0.2.23
22-10-13 21:34:51 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:34:51 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/56: 
22-10-13 21:34:51 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:34:51 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/56: 
22-10-13 21:34:51 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:34:51 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/59: 
22-10-13 21:35:03 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:35:03 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/61: 
22-10-13 21:35:03 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:35:03 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/61: 
22-10-13 21:35:03 ERROR (MainThread) [supervisor.utils.dbus] D-Bus method org.freedesktop.DBus.Properties.call_get_all not exists!
22-10-13 21:35:03 WARNING (MainThread) [supervisor.dbus.network] Can't process /org/freedesktop/NetworkManager/Devices/62:

weetmuts, 2022-10-15:
Thanks for the quick issue report! Closing this now!

closed 2022-10-15 by weetmuts