wmbusmeters/wmbusmeters #638: Client not reconnect to MQTT broker

jacekmarek, 2022-10-13:
Home Assistant 2022.9.7
Home Assistant Community Add-on: wmbusmeters,  version: 0.2.23
Home Assistant Add-on: Mosquitto broker, version: 6.1.3 

When MQTT broker is temporarily unavailable (stop working - crashed, start again), the client does not reconnect to MQTT broker and stop publish messages. You need to restart add-on wmbusmeters for it to work properly

Configuration add-on:
```
conf: |-
  loglevel=normal
  device=rtlwmbus:868.9M:t1
  logtelegrams=false
  format=json
  logfile=/dev/stdout 
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/"$METER_ID" "$METER_JSON"
```

Log add-on:
```
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Adding meter-0002 ...
Adding meter-0003 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
MQTT Discovery cleanup...
Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on t1 using fq 868.9M
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1

```

jacekmarek, 2022-10-28:
@weetmuts do you need additional information to troubleshoot this issue?

ch3funio, 2023-02-11:
Hi, i got the same problem. 
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:08:00] INFO: Syncing wmbusmeters configuration ...
[15:08:00] INFO: Registering meters ...
[15:08:01] INFO: Adding meter-0001 ...
[15:08:01] INFO: Adding meter-0002 ...
[15:08:01] INFO: Generating MQTT configuration ... 
[15:08:02] INFO: Broker mqtt://core-mosquitto:1883 will be used.
MQTT Discovery cleanup...
[15:08:04] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on s1,t1,c1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1` and so on.

ch3funio, 2023-02-11:
``aaaand it's workign after some config tweaking, i see some differences in your here:

```
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
  meterfiles=/config/wmbusmeters/logs/meter_readings/
  meterfilesaction=overwrite
```
and mqtt:
```
mqtt:
  basic_topic: wmbusmeters
  port: 1883
  host: core-mosquitto
  user: mqtt-user
  password: "!secret mqtt_password"
```
what i change was host (previously got whole mqtt://core:port etc)
and quotation marks in shell

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]