wmbusmeters/wmbusmeters #639: HackRF for use with 2.4Ghz Trilliant network (Hydro One)

gregology, 2022-10-15:
Hydro One (electricity provider in Ontario Canada) uses smart Trilliant meters which communicate using an IEEE 802.15.4 mesh standard with a ZigBee variation. IEEE 802.15.4-conformant devices may use one of three possible frequency bands for operation (868/915/2450 MHz). From my research, the Hydro One network is using the 2.4Ghz band.

I have a HackRF One SDR which can receive at 2.4Ghz. Is wmbusmeters compatible with a HackRF One? If so, is there documentation on how to setup my device? If not, does anyone have any other recommendations on options to collect this data with a HackRF One? The final goal is to have this data in Home Assistant to monitor the efficiency of our heat pump.

BIBOLV, 2022-10-30:
In case RTLSDR is being used then `rtlwmbus` reads input from `rtlsdr` and feeds it to wmbusmeters for payload decryption and decoding. I do not know if `rtlsdr` can read from HackRF, but as it is popular, I guess someone has added support. Now moving to data - wmbusmeters will not be able to parse ZigBee messages, at least out of the box.
There are other project for zigbee devices, like zigbee2mttq or I use deconz to read all my zigbee devices and display in HA,

BIBOLV, 2023-01-02:
Closing issue for now. Please reopen if any other questions.

closed 2023-01-02 by BIBOLV