wmbusmeters/wmbusmeters #640: Set device by ID/path

shaggee, 2022-10-16:
Is it possible to use the device addres by ID/path not by ttyUSB0/ttyUSB1? Sometimes the devices are chenging their path ttyUSB0<->ttyUSB1 and causing problems.
The same should be applied to "donotprobe" option.

weetmuts, 2022-10-16:
What does an ID/path look like? You know you can specify im871a etc to scan only for that dongle?

weetmuts, 2022-10-16:
Have you read the README from "As device you can use:"
after which all the ways it is possible to specify the dongle.

shaggee, 2022-10-16:
Im using nanoCUL. Should I use cul:c1?
ID and path:
  /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  /dev/serial/by-path/pci-0000:00:12.0-usb-0:1.4:1.0-port0

MightyPork, 2022-10-18:
@shaggee  Try adding a udev rule, you can create an alias tied to the device by its VID:PID or a serial number.

Here's some examples: http://dingo-iot.io/help/dingo/Content/Software/DINGO%20Stack/File%20system/etc/udev/rules.d/99-usb-serial.rules.htm

closed 2022-10-28 by shaggee