wmbusmeters/wmbusmeters #642: libwmbusmeters as shared library

heeplr, 2022-10-19:
wmbusmeters is an awesome project and I think it could unleash it's full potential only by providing a shared library.
I found the discussion in #104 and like to add to this suggestion.

I'm currently monitoring a lot of different 433/868 MHz devices using rtl_433 (garage door opener, temperature monitor, tire pressure sensors and wmbus meters). 
Some WMBUS devices are already fully supported by rtl_433, some only provide the hex data. Currently I'm using a jq/grep quirky pipeline to pass wmbus hex datagrams to wmbusmeters which kind of works.

Turns out, [wmbus code](https://github.com/merbanan/rtl_433/blob/master/src/devices/m_bus.c) in rtl_433 is quite elementary compared to wmbusmeters.

Would it be too much to ask, to build/install/version a shared library and link all current tools against it? While that's trivial to do (i assume the API is pretty stable at this point), I can see that it is quite a lot of work. I see a multitude of benefits here:

* 3rd party tools could decode WMBUS easily without manually inlining wmbusmeters code
* no breaking changes needed
* the wmbusmeters codebase could be slimmed down if that is wanted, as some things are already done by other well maintained tools (e.g. rtl433/rtlwmbus maybe)
* as wmbusmeters could be used in more projects, wider adoption could lead to more driver/meter support contributions
* encapsulation, separation of concerns and unix philosophy rock ;)

It would be awesome if you could consider such a step in the not-so-far future.

Thanks everyone for their contribution to this *awesome* project.

MightyPork, 2022-10-20:
This would be a great addition, especially if the library could be linked statically as well.

My current setup with a third party tool is that it spawns a wmbusmeters instance in the "stdin" mode and then writes the hex telegrams to this pipe and parses JSON that comes out the other end. It's quirky, but works. Having it all in one binary would significantly ease the upgrade process, for example - provided the library version was relicensed to LGPL to allow that.

MightyPork, 2022-10-20:
The software is written in Rust, so JSONs are not a problem, it's a bit complicated to set up but works okay in this "slave process" configuration. 

One thing I'm probably not doing the best way is passing the meter quadruples as CLI params (`meter1 auto 19044480 NOKEY`). The meters can be added or removed at run time and are defined elsewhere in a database, some with encryption keys, so each time the config changes, I restart the slave process.

I thought of using a temporary config files, perhaps in /tmp, but creating a whole folder with one file per meter is not practical. So, I went with the easiest way. It's reading an apartment building with 100+ meters, so you can imagine the command line gets pretty long, but so far it worked okay. One of the meters there is the new QHeat 55 (US)R driver I'm trying to add

weetmuts, 2022-10-20:
Interesting, so how do you get the telegrams into rust? I.e. before you send them to wmbusmeters.

heeplr, 2022-10-20:
> I do not think that integrating wmbusmeters in rtl_433 will help you that much, since you also need to integrate the key management into rtl_433.

It's true that rtl_433 with libwmbusmeters support would need some changes to rtl_433. Simplest way of key management without explicit rtl_433 key support could use environment variables or an external config file.

But since encryption becomes more and more ubiquitous even with cheap ISM hardware anyway, I could imagine that rtl_433 adds a way of handling decryption keys sooner or later and wmbusmeters integration could be the precedence.

weetmuts, 2022-10-20:
@MightyPork Yes, it seems to me that it would make sense for wmbusmeters to understand the NEP protocol, just like wmbusmeters understand the other proprietary (but documented) apis (im871a,amb8465 etc) to receive wmbus telegrams.
Then the udp can enter wmbusmeters and then send the decoded data to your application?

I found this, but I cannot find the protocol itself in there...
https://www.softlink.cz/wp-content/uploads/2016/12/Manual_WB868-RFE-R.pdf

MightyPork, 2022-10-20:
I have now created a new issue about NEP: #643

closed 2023-02-19 by github-actions[bot]