wmbusmeters/wmbusmeters #648: Techem Watermeter Radio3 / Current consumption not present in JSON

Snowrisk, 2022-10-27:
Hi,

I have two Techem Watermeters (warm & cold) with Radio3-Modules.
These are recognized and read out with the mkradio3 driver. The current consumption (055) is also displayed. 
However, this is then not available in the JSON (marked with "<<<<<<<<<"):

warm:
```
(meter) Warmwasser_Wohnung: for me? 63063736 in 63063736
(meter) Warmwasser_Wohnung: yes for me
(meter) Warmwasser_Wohnung(1) mkradio3  handling telegram from 63063736
(meter) Warmwasser_Wohnung 63063736 "2F446850363706637462A2069F2B3207B015860100000D090911100F11130E1714161918181813111B19141A15171518"
(wmbus) parseDLL @0 48
(telegram) DLL L=2f C=44 (from meter SND_NR) M=5068 (TCH) A=63063736 VER=74 TYPE=62 (Warm water) (driver mkradio3) DEV=cul RSSI=-60
(wmbus) parseELL @10 38
(wmbus) parseNWL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(telegram) TPL CI=a2
telegram=|2F446850363706637462A2_069F2B3207B015860100000D090911100F11130E1714161918181813111B19141A15171518|+117
(mkradio3) 000   : 2f length (47 bytes)
(mkradio3) 001   : 44 dll-c (from meter SND_NR)
(mkradio3) 002   : 6850 dll-mfct (TCH)
(mkradio3) 004   : 36370663 dll-id (63063736)
(mkradio3) 008   : 74 dll-version
(mkradio3) 009   : 62 dll-type (Warm water)
(mkradio3) 010   : a2 tpl-ci-field (Mfct specific)
(mkradio3) 011 C?: 069F2B3207B015860100000D090911100F11130E1714161918181813111B19141A15171518 mfct specific
(mkradio3) 049 C!: 2b9f previous date (2021-12-31T02:00:00Z)
(mkradio3) 051 C!: 3207 prev consumption (184.200000 m3)
(mkradio3) 053 C!: 15b0 current date (2022-10-27T02:00:00Z)
(mkradio3) 055 C!: 8601 curr consumption (39.000000 m3) <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(meters) render field total(Volume)[0] without dventry
(meters)             "total_m3":223.2
(meters) render field target(Volume)[1] without dventry
(meters)             "target_m3":184.2
(meters) render field current_date(Text)[2] without dventry
(meters)             "current_date":"2022-10-27T02:00:00Z"
(meters) render field prev_date(Text)[3] without dventry
(meters)             "prev_date":"2021-12-31T02:00:00Z"
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/usr/bin/mosquitto_pub -h 10.10.10.241 -u MQTT -P 23cn64W2mA3f2K26 -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
(shell) env "METER_JSON={"media":"warm water","meter":"mkradio3","name":"Warmwasser_Wohnung","id":"63063736","total_m3":223.2,"target_m3":184.2,"current_date":"2022-10-27T02:00:00Z","prev_date":"2021-12-31T02:00:00Z","timestamp":"2022-10-27T16:24:30Z","device":"cul","rssi_dbm":-60}"
(shell) env "METER_ID=63063736"
(shell) env "METER_NAME=Warmwasser_Wohnung"
(shell) env "METER_MEDIA=warm water"
(shell) env "METER_TYPE=mkradio3"
(shell) env "METER_TIMESTAMP=2022-10-27T16:24:30Z"
(shell) env "METER_TIMESTAMP_UTC=2022-10-27T16:24:30Z"
(shell) env "METER_TIMESTAMP_UT=1666887870"
(shell) env "METER_TIMESTAMP_LT=2022-10-27 18:24.30"
(shell) env "METER_DEVICE=cul"
(shell) env "METER_RSSI_DBM=-60"
(shell) env "METER_TOTAL_M3=223.2" <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(shell) env "METER_TARGET_M3=184.2" <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(shell) env "METER_CURRENT_DATE=2022-10-27T02:00:00Z"
(shell) env "METER_PREV_DATE=2021-12-31T02:00:00Z"
(shell) waiting for child 1860 to complete.
(shell) /bin/sh: return code 0
```

cold:
```
(meter) Kaltwasser_Wohnung: for me? 70864007 in 70864007
(meter) Kaltwasser_Wohnung: yes for me
(meter) Kaltwasser_Wohnung(2) mkradio3  handling telegram from 70864007
(meter) Kaltwasser_Wohnung 70864007 "2F446850074086707472A2069F2B7D0DB0159F02000016121222222326251C2B1F23292B23241D1A25261F27232B2124"
(wmbus) parseDLL @0 48
(telegram) DLL L=2f C=44 (from meter SND_NR) M=5068 (TCH) A=70864007 VER=74 TYPE=72 (Cold water) (driver mkradio3) DEV=cul RSSI=-46
(wmbus) parseELL @10 38
(wmbus) parseNWL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(telegram) TPL CI=a2
telegram=|2F446850074086707472A2_069F2B7D0DB0159F02000016121222222326251C2B1F23292B23241D1A25261F27232B2124|+181
(mkradio3) 000   : 2f length (47 bytes)
(mkradio3) 001   : 44 dll-c (from meter SND_NR)
(mkradio3) 002   : 6850 dll-mfct (TCH)
(mkradio3) 004   : 07408670 dll-id (70864007)
(mkradio3) 008   : 74 dll-version
(mkradio3) 009   : 72 dll-type (Cold water)
(mkradio3) 010   : a2 tpl-ci-field (Mfct specific)
(mkradio3) 011 C?: 069F2B7D0DB0159F02000016121222222326251C2B1F23292B23241D1A25261F27232B2124 mfct specific
(mkradio3) 049 C!: 2b9f previous date (2021-12-31T02:00:00Z)
(mkradio3) 051 C!: 7d0d prev consumption (345.300000 m3)
(mkradio3) 053 C!: 15b0 current date (2022-10-27T02:00:00Z)
(mkradio3) 055 C!: 9f02 curr consumption (67.100000 m3) <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(meters) render field total(Volume)[0] without dventry
(meters)             "total_m3":412.4
(meters) render field target(Volume)[1] without dventry
(meters)             "target_m3":345.3
(meters) render field current_date(Text)[2] without dventry
(meters)             "current_date":"2022-10-27T02:00:00Z"
(meters) render field prev_date(Text)[3] without dventry
(meters)             "prev_date":"2021-12-31T02:00:00Z"
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/usr/bin/mosquitto_pub -h 10.10.10.241 -u MQTT -P 23cn64W2mA3f2K26 -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
(shell) env "METER_JSON={"media":"cold water","meter":"mkradio3","name":"Kaltwasser_Wohnung","id":"70864007","total_m3":412.4,"target_m3":345.3,"current_date":"2022-10-27T02:00:00Z","prev_date":"2021-12-31T02:00:00Z","timestamp":"2022-10-27T16:25:34Z","device":"cul","rssi_dbm":-46}"
(shell) env "METER_ID=70864007"
(shell) env "METER_NAME=Kaltwasser_Wohnung"
(shell) env "METER_MEDIA=cold water"
(shell) env "METER_TYPE=mkradio3"
(shell) env "METER_TIMESTAMP=2022-10-27T16:25:34Z"
(shell) env "METER_TIMESTAMP_UTC=2022-10-27T16:25:34Z"
(shell) env "METER_TIMESTAMP_UT=1666887934"
(shell) env "METER_TIMESTAMP_LT=2022-10-27 18:25.34"
(shell) env "METER_DEVICE=cul"
(shell) env "METER_RSSI_DBM=-46"
(shell) env "METER_TOTAL_M3=412.4" <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(shell) env "METER_TARGET_M3=345.3" <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
(shell) env "METER_CURRENT_DATE=2022-10-27T02:00:00Z"
(shell) env "METER_PREV_DATE=2021-12-31T02:00:00Z"
(shell) waiting for child 1862 to complete.
(shell) /bin/sh: return code 0
```

Is this an error? What can I do to make it work?

Best regards
Jens

cloom, 2022-11-01:
Hello @Snowrisk 

What command did you run to have this detailed view? Sorry I just got my RTL-SDR key and start to learn about the amazing wmbusmeter :)
I have a Techem Radio 4 and I can get the data, it seems it's the same with the Techem Radio 3 and there is only a precision of 0.1 cubic meter sent over the radio :( 
I would love to see if I can get more data from it.

Snowrisk, 2022-11-01:
Hi @cloom 

this was the "debug"-setting at loglevel in the config file:

````
loglevel=debug
# Remember to change auto here to the device you are going to use in production.
device=/dev/ttyUSB0:cul:c1
donotprobe=/dev/ttyAMA0
logtelegrams=true
format=json
# meterfiles=/var/log/wmbusmeters/meter_readings
# meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
````

weetmuts, 2022-11-01:
You can also use --logtelegrams to get
telegram=|xx44xxxxxxxx| data then 
run wmbusmeters --analyze xx44xxxxxxx
or do it online using https://wmbusmeters.org

If you post a couple of telegrams here, then we can take a look at the resolution of the measurements.

weetmuts, 2022-11-01:
@Snowrisk Thanks, I will take a look.

jsuchal, 2022-11-02:
The current period consumption is not reported. Not sure why but the you get the total and target (previous period). Its easy to calculate though:
`"total_m3":412.4,"target_m3":345.3` =>  345.3 + 67.1 = 412.4

You can easily do a helper in HA to get this back to current period. total - target.

Snowrisk, 2022-11-06:
@jsuchal 
I have done this already, but it would be nice if it's also in the json as for the other meters.

Snowrisk, 2023-03-12:
@weetmuts @BIBOLV Any news ?

closed 2023-06-17 by github-actions[bot]