wmbusmeters/wmbusmeters #649: Missing Driver for HeatMeter

Smart86, 2022-10-28:
Hello and first thank you for this great project!

I have a heat meter that unfortunately cannot find a suitable driver. This is a Sensus IEC 870-5.

Here is the log:

(meter) HeatMeter: meter detection could not find driver for id: xxxxxx mfct: (HYD) Hydrometer (0x2324) type: Heat meter (0x04) ver: 0x86
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|19442423860456219483A2_510E0006641AFB53B0904642ABE3EF|+2

On the device on site you can see the consumption up to last year, the current meter reading in m³, a consumption in m³ and m³ per hour.

I would be happy to provide further details if this could help in any way.

BIBOLV, 2022-10-30:
Hi,
Have you tried compact5 driver? Some values can be exctraced:
```
{
    "media":"heat",
    "meter":"compact5",
    "name":"",
    "id":"83942156",
    "total_kwh":70745,
    "current_kwh":45139,
    "previous_kwh":25606,
    "timestamp":"2022-10-30T20:21:46Z"
}
```

Smart86, 2022-11-01:
Thank you, the suggested driver compact5 I tried of course. It says that some data is extracted, but unfortunately the measured values do not match the Real values in any way :-(

When I check da izar driver, in reality it is MWh but the value is at least correct. All remaining values are unfortunately missing

BIBOLV, 2022-11-22:
Please collect more telegrams using --logtelegrams and post them here together with current reading of meter.

closed 2023-04-05 by github-actions[bot]