wmbusmeters/wmbusmeters #650: Wmbusmeters works with my raspberry but not on my synology using Docker

gd35, 2022-10-29:
I made my own nanoCUL USB dongle. It works perfectly well with my rpi4 (without docker) based on the latest version of wmbusmeters I got from GitHub. However, I plan to use this nanoCUL with my Synology using Docker. I solved the driver issue (lack of usb support with DSM 7.x) and was able to install the container and to configure it. However, it does not work. I got the following entries in the log file:

`(wmbusmeters) version: 1.8.0-13-ga5f0783
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: /dev/ttyUSB0:cul:38400:t1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" cul
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(115,212,92,99,54,213,97,14) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(115,212,92,99,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary "5620312E3637206E616E6F43554C3836385F723536380D0A"
(cul) probe response "V 1.67 nanoCUL868_r568<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(serial) EVENT thread interrupted
(cul) are you there? yes
(main) opening /dev/ttyUSB0:cul:38400:t1
Started config cul on /dev/ttyUSB0 listening on t1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(115,212,92,99,54,213,97,14) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(115,212,92,99,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740D0A"
(serial) EVENT thread interrupted
(wmbus device) timeout request id 1
(cul) received ""(cul) setting link mode(s) t1 is not supported for this cul device!
(serial) stopping manager
`

wmbusmeters states that my nanoCUL does not support t1 although it worked fine with the raspberry (without docker).
I tried wmbusmeters on my Mac and got a similar issue...

I will be happy to get some advises. 

Thanks in advance,

BIBOLV, 2022-10-30:
First log is from old version, so first, I would suggest to upgrade to latest version and try again. If it does not work then try to install docker on raspi and check if all works then.
Looking at logs it is visible that initial communication with dongle is successful in all cases, but it fails when mode is being set, so try to set different mode to check if it will fail also.
If nothing helps - my last suggestion would be to purchase IMST dongle to replace cul.

gd35, 2022-11-04:
Thanks for the advice. Just after sending my first message, I realized that the official docker container had an older version of wmbusmeters. So I generated my own container with the latest version. The issue remains. So I will test a docker container with my rpi4 to check whether it is a docker issue. I will also test another mode. I recently found an unused rpi3b that I can use to connect to my homemade cul. It will be a cheaper option compared to the buying of an IMST dongle.

BIBOLV, 2023-01-02:
Closing issue for now. Please reopen if you want to continue this topic or have any other news/questions.

closed 2023-01-02 by BIBOLV