wmbusmeters/wmbusmeters #656: multical21 "wrong decryption key" ?

koburg, 2022-11-04:
Hello

 I got the 32bit encryption key from water supply company and setup the meter but get this in the log

Received telegram from: 77605206
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: im871a[00102644]
                  rssi: -66 dBm
                driver: multical21
(wmbus) 000   : 23 length (35 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2d2c dll-mfct (KAM)
(wmbus) 004   : 06526077 dll-id (77605206)
(wmbus) 008   : 1b dll-version
(wmbus) 009   : 16 dll-type (Cold water meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 65 ell-acc
(wmbus) 013   : 704ccb21 sn (AES_CTR)
(wmbus) 017   : a3d2 payload crc (calculated d6b0 ERROR)
(wmbus) 019 CE: 95B63ECB80DEDAA4718740D5713D0CF7BD failed decryption. Wrong key?
(wmbus) telegram from 77605206 ignored by all configured meters!
(serial) received binary "A5C20309472D2C57277971340C4C6807"
(im871a) checkIM871AFrame "A5C20309472D2C57277971340C4C6807"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 76 (-84 dBm)
(im871a) got crc16 0768 expected 0768
(im871a) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 10
(wmbus) parseDLL @0 10
(telegram) DLL L=09 C=47 (from meter ACC_NR) M=2c2d (KAM) A=71792757 VER=34 TYPE=0c (Heat volume at inlet meter)

koburg, 2022-11-10:
anything I can do to investigate further ?

weetmuts, 2022-11-10:
It is must likely just the wrong decryption key. It has happened to other people that they have received the wrong key.

Double check the meter id and request the key again from the water company. 

It is a 128bit key though....

BIBOLV, 2023-01-02:
Closing issue for now as this is probably related to wrong decryption key. Please reopen issue if there are any news or questions.

closed 2023-01-02 by BIBOLV

big23, 2023-08-30:
Hello , same problem - Multical21 and "Wrong key?" message. My key i 100% correct (checked with HA-addon)
**Decrypted meter details:
id = 74685644
key = B91EAF561BEF664ADD91875125763485**

Log:
(serial) received ascii "C1;1;1;2023-08-30 17:35:03.000;29;35;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2023-08-30 17:35:03.000;29;35;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 36
(wmbus) parseELL @10 26
(wmbus) parseDLL @0 36
(telegram) DLL L=23 C=44 (from meter SND_NR) M=2c2d (KAM) A=74685644 VER=1b TYPE=16 (Cold water meter) (driver multical21) DEV=rtlwmbus[00000001] RSSI=29
(wmbus) parseELL @10 26
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=13 SN=c1875020 (AES_CTR session=1 time=329852) CRC=fabd
Received telegram from: 74685644
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: rtlwmbus[00000001]
                  rssi: 29 dBm
                driver: multical21
(wmbus) 000   : 23 length (35 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2d2c dll-mfct (KAM)
(wmbus) 004   : 44566874 dll-id (74685644)
(wmbus) 008   : 1b dll-version
(wmbus) 009   : 16 dll-type (Cold water meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 13 ell-acc
(wmbus) 013   : c1875020 sn (AES_CTR)
(wmbus) 017   : fabd payload crc (calculated e6a9 ERROR)
(wmbus) 019 CE: F7ED2AF48698B8B1BBDE661C9BCCA3820F failed decryption. Wrong key?
(wmbus) telegram from 74685644 ignored by all configured meters!
(serial) received ascii "C1;1;1;2023-08-30 17:35:03.000;33;28;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2023-08-30 17:35:03.000;33;28;74685644;0x23442d2c445668741b168d2013c1875020fabdf7ed2af48698b8b1bbde661c9bcca3820f<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=36.