wmbusmeters/wmbusmeters #657: rtlsdr issue, non-numeric bus id, device lost

kitor, 2022-11-04:
Putting as a screenshot due to non-ascii chars.

[image]

This happens on my proxmox host (debian 11.5 based), both from head and from e3e7b77 which I run on another device.

rtl-sdr and rtl-wmbus works fine:

```
root@pve09:/var/log/wmbusmeters# rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus
T1;0;0;2022-11-04 09:02:35.000;52;57;357481FF;0x29441486ff8174350408b60efffff445ffff0010a985ea12ff5da0ff00005a00ffff7400642dc4150015
T1;1;1;2022-11-04 09:02:35.000;166;131;00097195;0x4e4401069571090005077a77004085b4e03f1e9599966dff8f9d8ab15f83c09c62bd4947c9772215bfe7b171a903dfea16235c11000da34e2520a629ebbe9a559936388a0b628d8c11fcbb5881d531
T1;1;1;2022-11-04 09:02:35.000;138;160;00097195;0x4e4401069571090005077a77004085b4e03f1e9599966dff8f9d8ab15f83c09c62bd4947c9772215bfe7b171a903dfea16235c11000da34e2520a629ebbe9a559936388a0b628d8c11fcbb5881d531
T1;0;0;2022-11-04 09:02:35.000;42;21;FF752044;0x2d441486442075ff0408b612fffff64501860a10ae621410a780ea12a762a02100009f0200003f0664ff00149019
T1;1;1;2022-11-04 09
```

wmbusmeters config:

```
loglevel=normal
device=rtlwmbus:868.993M:t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

Any idea where to look?

BIBOLV, 2023-01-02:
Were you able to resolve issue? Did you tried to change serial of dongle?

kitor, 2023-02-24:
Yes, changing the serial number did the trick. Not sure what had happened since the dongle worked earlier with wmbusmeters.

Anyway, closing as solved. Thanks!

closed 2023-02-24 by kitor