wmbusmeters/wmbusmeters #658: Home Assistant Addon: Wiki cleanup

pascaltippelt, 2022-11-04:
Hi!

I am using the home assistant integration. It works good, but documentation is wrong / old.

# Wiki

The wiki at https://weetmuts.github.io/wmbusmeterswiki/HASS.html states to configure the shell like _shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON"_. This causes an error. Correct is:  _shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_ID $METER_JSON_

I suggest to add a full example-configuration to deliver more context:

```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: >-
  loglevel=normal\ndevice=auto:t1,c1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=false\nformat=json\nlogfile=/dev/stdout\nshell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_ID $METER_JSON
meters:
  - |-
    name=MyTapWater
    id=12345678
    key=00112233445566778899AABBCCDDEEFF
    driver=multical21
mqtt:
  host: 127.0.0.1
  port: 1883
  user: wmbusmeters
  password: wmbusmeters

```

There is also a problem with sending the JSON to MQTT, it is not always completely delivered. But I will add anither issue for that.

Jordan87, 2022-11-10:
Also from Add-on "Configuration" page we are not able to add meter, we have to select "Edit in YAML" first to see valid place for meter configuration.
[image]
[image]

I spent few hours to find how to configure it (I tried by editing files but ale changes were reverted after add-on restart)

BIBOLV, 2022-12-03:
[Documentation ](https://weetmuts.github.io/wmbusmeterswiki/HASS.html)has been updated.

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-17 by github-actions[bot]