wmbusmeters/wmbusmeters #663: Static USB device id

Jordan87, 2022-11-10:
Hi

Im using HA with wmbusmeters add-on.

Can you add static settings for device?
I have three different USB devices which are detected as a TTYUSB* and sometimes after reboot can change number.

For example for Modbus (main HA configuration file - configuration.yaml) or Zigbee2Mqtt i can configure IDs instead of device path so we can consider it as a static selection for device
[image]

When I configure wmbusmeters add-on after start it is quickly stoped.

> data_path: /config/wmbusmeters
> enable_mqtt_discovery: false
> conf: |-
>   loglevel=normal
>   device=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:t1
>   logtelegrams=true
>   format=json
>   logfile=/dev/stdout
>   shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"

Log:
> s6-rc: info: service s6rc-oneshot-runner: starting
> s6-rc: info: service s6rc-oneshot-runner successfully started
> s6-rc: info: service fix-attrs: starting
> s6-rc: info: service fix-attrs successfully started
> s6-rc: info: service legacy-cont-init: starting
> s6-rc: info: service legacy-cont-init successfully started
> s6-rc: info: service legacy-services: starting
> s6-rc: info: service legacy-services successfully started
> Syncing wmbusmeters configuration ...
> Registering meters ...
> Adding meter-0001 ...
> Generating MQTT configuration ... 
> Broker core-mosquitto will be used.
> MQTT Discovery cleanup...
> Running wmbusmeters ...
> (config) "device" "/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:t1"
> (wmbusmeters) version: 1.9.0-78-g86b497d
> (config) using device: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:t1 
> (config) number of meters: 1
> (lookup) with file/hex "/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0" unknown
> You have to specify the expected device type for the tty /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
> (serial) stopping manager
> (config) "logtelegrams" "true"
> (config) "format" "json"
> (config) "logfile" "/dev/stdout"
> (config) "shell" "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
> (config) "" ""
> (config) loading meter file /config/wmbusmeters/etc/wmbusmeters.d/meter-0001
> (config) name=MainWater
> (config) driver=izar> (config) "logfile" "/dev/stdout"
> (config) "shell" "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
> (config) "" ""
> (config) loading meter file /config/wmbusmeters/etc/wmbusmeters.d/meter-0001
> (config) name=MainWater
> (config) driver=izar
> (config) id=2119e135
> (config) key=<notprinted>
> (wmbusmeters) using log file /dev/stdout
> s6-rc: info: service legacy-services: stopping
> s6-rc: info: service legacy-services successfully stopped
> s6-rc: info: service legacy-cont-init: stopping
> s6-rc: info: service legacy-cont-init successfully stopped
> s6-rc: info: service fix-attrs: stopping
> s6-rc: info: service fix-attrs successfully stopped
> s6-rc: info: service s6rc-oneshot-runner: stopping
> s6-rc: info: service s6rc-oneshot-runner successfully stopped

magcode, 2022-11-10:
You may want to look into "udev". In Linux this allows you to have fixed USB device names based on the serial number of the device. This is handled on the OS level und you get reliable "/dev/myfirstusbdevice"

weetmuts, 2022-11-10:
What kind of wmbus dongle do you use?

weetmuts, 2022-11-10:
Assuming you are using an im871a then try this:
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:im871a:t1

Jordan87, 2022-11-11:
> You may want to look into "udev". In Linux this allows you to have fixed USB device names based on the serial number of the device. This is handled on the OS level und you get reliable "/dev/myfirstusbdevice"
Im using dedicated HA OS (HAOS) which AFAIK doesn't have standard cli access
https://developers.home-assistant.io/docs/operating-system/

> What kind of wmbus dongle do you use?
Its a Arduino Nano (clone) + CC1101 module

> Assuming you are using an im871a then try this: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:im871a:t1
With this log is showing

> No wmbus device detected, waiting for a device to be plugged in.
> Expected im871a on /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 but did not find it! Ignoring tty!

Jordan87, 2022-11-11:
Ok, i think this one is working for me.

> device=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:cul:t1

But without specifying type of donge shouldn't this work, as with "auto"?

weetmuts, 2022-11-11:
Yeah, sorry about that, I have changed the code so that auto only works, if you specify auto.
If you specify a serial port, then you have to specify the expected wmbus device.

The reason was that the auto scanning is really slow, so I want to avoid this as much as possible.

I have updated the README right now to reflect this.

weetmuts, 2022-11-13:
Can we close this now?

Jordan87, 2022-11-14:
If you don't want to introduce any change in code then yes. Thanks :)

closed 2023-01-02 by BIBOLV