wmbusmeters/wmbusmeters #664: Analyze command provokes crash (aborted core dumped) / WHE542 meter regression

mentulatore, 2022-11-11:
Hi, this is a single issue which contains multiple reports.
Environment: Fedora 36 x86_64, all updated / replicated on Ubuntu 20.04 too.

1. wmbusmeters, compiled from master (commit fadce1cb60c094144cb90edd73a351a6b6e0326d), crashes as shown below (same output if compiled with `DEBUG=true`):

```
$ ./configure; make
[...]
$ build/wmbusmeters --analyze '4844653268617784340878_0B6E0000000DFF5F2F00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD046D2A11C22B'
wmbusmeters: src/driver_bfw240radio.cc:77: int {anonymous}::getHistoric(int, std::vector<unsigned char>&): Assertion `content.size() == 40' failed.
Aborted (core dumped)
```

But it works using Fedora's 36 repository package:

```
$ wmbusmeters --analyze '4844653268617784340878_0B6E0000000DFF5F2F00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD046D2A11C22B'
Auto driver  : whe5x
Best driver  : fhkvdataiv(driver should be upgraded) 54/54
Using driver : whe5x(driver should be upgraded) 00/00
000   : 48 length (72 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 6532 dll-mfct (LSE)
004   : 68617784 dll-id (84776168)
008   : 34 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
011   : 0B dif (6 digit BCD Instantaneous value)
012   : 6E vif (Units for H.C.A.)
013 C!: 000000 current consumption (0.000000 hca)
016   : 0D dif (variable length Instantaneous value)
017   : FF vif (Manufacturer specific)
018   : 5F combinable vif (DurationExceedsUpperLast)
019   : 2F varlen=47
020 C!: 00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD vendor extension data
067   : 04 dif (32 Bit Integer/Binary Instantaneous value)
068   : 6D vif (Date and time type)
069 C!: 2A11C22B device datetime (2022-11-02 17:42)

{
    "media":"heat cost allocation",
    "meter":"whe5x",
    "name":"",
    "id":"84776168",
    "current_consumption_hca":0,
    "set_date":"",
    "consumption_at_set_date_hca":0,
    "set_date_1":"",
    "consumption_at_set_date_1_hca":0,
    "error_date":"",
    "device_date_time":"2022-11-02 17:42",
    "timestamp":"2022-11-11T16:34:36Z"
}
```
Package information:

```
$ sudo dnf info wmbusmeters
Installed Packages
Name         : wmbusmeters
Version      : 1.9.0
Release      : 1.fc36
Architecture : x86_64
Size         : 3.3 M
Source       : wmbusmeters-1.9.0-1.fc36.src.rpm
Repository   : @System
From repo    : updates
Summary      : Read the wireless mbus protocol to acquire utility meter readings
URL          : https://github.com/weetmuts/wmbusmeters
License      : GPLv3+
Description  : The program receives and decodes C1,T1 or S1 telegrams
             : (using the wireless mbus protocol) to acquire utility meter readings.
             : The readings can then be published using MQTT, curled to a REST api,
             : inserted into a database or stored in a log file.
```

If compiled from git after commenting out line 77 in "src/driver_bfw240radio.cc" this is the output:

```
$ build_debug/wmbusmeters --analyze '4844653268617784340878_0B6E0000000DFF5F2F00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD046D2A11C22B'
Auto driver  : qcaloric
Best driver  : fhkvdataiv(driver should be upgraded) 54/54
Using driver : qcaloric 00/00
000   : 48 length (72 bytes)(OK)
001   : 44 dll-c (from meter SND_NR)
002   : 6532 dll-mfct (LSE)
004   : 68617784 dll-id (84776168)
008   : 34 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
011   : 0B dif (6 digit BCD Instantaneous value)
012   : 6E vif (Units for H.C.A.)
013 C!: 000000 ("current_consumption_hca":0)
016   : 0D dif (variable length Instantaneous value)
017   : FF vif (Manufacturer specific)
018   : 5F combinable vif (DurationExceedsUpperLast)
019   : 2F varlen=47
020 C?: 00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD
067   : 04 dif (32 Bit Integer/Binary Instantaneous value)
068   : 6D vif (Date and time type)
069 C!: 2A11C22B ("device_date_time":"2022-11-02 17:42")

{
    "media":"heat cost allocation",
    "meter":"qcaloric",
    "name":"",
    "id":"84776168",
    "status":"OK",
    "current_consumption_hca":0,
    "set_date":null,
    "consumption_at_set_date_hca":null,
    "set_date_1":null,
    "consumption_at_set_date_1_hca":null,
    "device_date_time":"2022-11-02 17:42",
    "timestamp":"2022-11-11T16:33:16Z"
}
```

2. It seems the current version of the code doesn't handle correctly the "vendor extension data" field. Notice how the value of the timestamp differs from wmbusmeters' repository package for Fedora.

3. I think both versions fail to decode all the included information in the telegram, like the "set_date" field which should be "2022-09-30".

Additional information: the telegrams are generated by SIEMENS HCA model WHE542-D291S. They were recorded using wmbusmeters (rtlsdr + rtl_wmbus).

Here I provide 16 samples which may help debug / improve wmbusmeters (each telegram should be from a different device):

```
telegram=|4844653268617784340878_0B6E0000000DFF5F2F00019380813708B06EFFFFDE2913070000DF2A0000000098000801CC001A0019002A000000000000000000000037FD046D2A11C22B|
telegram=|4844653269617784340878_0B6E0000000DFF5F2F00019280810108B06EFFFFDE2972070000DF2A000000004600F300E20077006A00080000000000000000000000FCFC046D2111C22B|
telegram=|4844653270617784340878_0B6E0000000DFF5F2F00019380803D08B06EFFFFDE2997060000DF2A000000004700D300BF007D003F0024000000000000000000000047FD046D1F11C22B|
telegram=|4844653271617784340878_0B6E0000000DFF5F2F00019288800708B06EA72BDE2903000000DF2A0000000003000000000000000000000000000000000000000000FDFF046D2311C22B|
telegram=|4844653272617784340878_0B6E0000000DFF5F2F000192807E0008B06EFFFFDE2943010000DF2A0000000016000E0046000F00140002000000000000000000000071FF046D2611C22B|
telegram=|4844653273617784340878_0B6E0000000DFF5F2F000192807E0108B06EFFFFDE2950000000DF2A00000000000000001E0014000000000000000000000000000000CEFF046D3611C22B|
telegram=|4844653274617784340878_0B6E0000000DFF5F2F00019280000F08B06EFFFFDE2961010000DF2A00000000000000006800390000000000000000000000000000005FFF046D3811C22B|
telegram=|4844653275617784340878_0B6E0000000DFF5F2F000192807F0008B06EFFFFDE2957130000DF2A00000000430042015D0100013E012D0000000000000000000000B3FA046D3611C22B|
telegram=|4844653276617784340878_0B6E0000000DFF5F2F000192807F0208B06EFFFFDE2978070000DF2A000000002E006A008200FD00D7001C0000000000000000000000F6FC046D3711C22B|
telegram=|4844653277617784340878_0B6E0000000DFF5F2F000192807F0508B06EFFFFDE2914180000DF2A000000001D01A2018E0161012A013E0000000000000000000000EAF8046D3611C22B|
telegram=|4844653278617784340878_0B6E0000000DFF5F2F00019380FF3108B06EFFFFDE2920110000DF2A00000000C100F900F800DB00BD00160000000000000000000000A0FB046D3811C22B|
telegram=|4844653279617784340878_0B6E0000000DFF5F2F00019380802508B06EFFFFDE2953030000DF2A000000002000B3006500270002000000000000000000000000009FFE046D2011C22B|
telegram=|4844653280617784340878_0B6E0000000DFF5F2F00019380FE2508B06EFFFFDE2900000000DF2A00000000000000000000000000000000000000000000000000000000046D2111C22B|
telegram=|4844653281617784340878_0B6E0000000DFF5F2F00019380011208B06EFFFFDE2900000000DF2A00000000000000000000000000000000000000000000000000000000046D2611C22B|
telegram=|4844653282617784340878_0B6E0000000DFF5F2F00019380012808B06EFFFFDE2914010000DF2A0000000006000F0032002B0000000000000000000000000000008EFF046D1F11C22B|
telegram=|4844653283617784340878_0B6E0000000DFF5F2F000193807E1F08B06EFFFFDE2927060000DF2A000000002600D900ED00570030000000000000000000000000008DFD046D2211C22B|
```

Thanks

mentulatore, 2022-11-11:
According to SIEMENS' WHE5 family System Manual:

**Useful data content of the AMR telegrams (S-mode)**

- Device number (8-digit)
- Device type/software version
- Time/date
- Error status
- Error date
- Current consumption
- Due date
- Due date value
- Counter status at end of last month
 
**Useful data content of the walk-by telegrams (S-mode)**

- Device number (8-digit)
- Device type/software version
- Time/date
- Error status
- Error date
- Current consumption
- Due date
- Due date value
- Counter status at end of last month
- 15 Statistic values

The explanation of the statistic values is included in SIEMENS' ACT50 help file:

> Specifies the meter readings on the last day of the month on the designated date for the last 15 months.

Regarding to the 16 telegram samples included in my previous post, I'm not sure but I suppose those were walk-by telegrams.

weetmuts, 2022-11-11:
Thank you very much for the detailed bug report! This was a very silly bug in bfw240radio. 

When you run analyze, it will actually try all drivers, to see which driver decodes the most fields.
The bfw240radio had a too aggressive assert, that was triggered by your telegram. 

Fix pushed in bfw240radio:
assert(content.size() == 40); 
to
assert(content.size() >= 40);

Please test. Also note that the qcaloric driver is the new merged driver for qcaloric, whe5x  and whe46x.

weetmuts, 2022-11-11:
Note also that these meters send two types of telegrams. One is the one that you have provided above, there are others with more information. Please check the actual driver here:
https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_qcaloric.cc

mentulatore, 2022-11-11:
> Note also that these meters send two types of telegrams.

Yeah, there are the AMR telegrams too (will try to receive them next time).
The meters near me are configured for S-mode: theoretically they should be capable of operating in C-mode as alternative (if programmed by the technicians). Don't know if C-mode telegrams differ in content from the S-mode ones.

> When you run analyze, it will actually try all drivers, to see which driver decodes the most fields.

Understood but please note it seemed to happen even when instructing the analyze switch to use the specific `qcaloric` driver.

> 2\. It seems the current version of the code doesn't handle correctly the "vendor extension data" field.

Should that field be decoded and produce visible values?

> Notice how the value of the timestamp differs from wmbusmeters' repository package for Fedora.

Is it normal?

> 3\. I think both versions fail to decode all the included information in the telegram, like the "set_date" field which should be "2022-09-30".

What about this?

Maybe I'm wrong but I expected to be able to read all the previously mentioned values (the ones shown in the manual) in the decoded telegram.

Thanks for the prompt reply!

weetmuts, 2022-11-12:
>  2. It seems the current version of the code doesn't handle correctly the "vendor extension data" field.
> Should that field be decoded and produce visible values?

If you can help in decoding it, then please help us! I do not have time to decode vendor specific data. 
I focus on the (w)mbus standard compliant fields. :-)

>  Notice how the value of the timestamp differs from wmbusmeters' repository package for Fedora.
> Is it normal?

Yes timestamp is when wmbusmeters received the telegram for decoding. 

> I think both versions fail to decode all the included information in the telegram, like the "set_date" field which should be "2022-09-30".

You have to provide a telegram that actually contains the set_date and we can have a look. The telegrams above do not (or maybe it is in the vendor proprietary data), as I mentioned the same meter sends multiple types of telegrams. You have to catch them all!

closed 2023-03-18 by github-actions[bot]