wmbusmeters/wmbusmeters #667: Many `internal error after select! errno=Invalid argument` after logtimestamps=always

Hadatko, 2022-11-13:
Hello, it looks like the parameter set in config is working but anyway i have plenty errors in log. Without using mentioned parameter log is without datetimestamp, but no errors.

`config`
```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: |-
  loglevel=debug
  logtimestamps=always
  device=/dev/ttyUSB1:im871a:t1,c1
  donotprobe=/dev/ttyAMA0
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters: []
mqtt: {}
```
log
```txt
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
MQTT Discovery ...
MQTT Discovery cleanup...
Running wmbusmeters ...
(config) "logtimestamps" "always"
[2022-11-13_12:05:14] (wmbusmeters) version: 1.9.0-71-ge01782d
[2022-11-13_12:05:14] (config) using device: /dev/ttyUSB1:im871a:t1,c1 
[2022-11-13_12:05:14] (config) number of meters: 0
[2022-11-13_12:05:14] (lookup) with file/hex "/dev/ttyUSB1" im871a
[2022-11-13_12:05:14] (serial) check if /dev/ttyUSB1 can be accessed
[2022-11-13_12:05:14] (serial) tty /dev/ttyUSB1 can be accessed
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) before config: /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,253) modem(DTR|RTS)
[2022-11-13_12:05:14] (serial) after config:  /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,160,231,75,220,127,248) modem(DTR|RTS)
[2022-11-13_12:05:14] (serialtty) opened /dev/ttyUSB1 fd 3 (detect im871a)
[2022-11-13_12:05:14] (serial) received binary ""
[2022-11-13_12:05:14] (serial /dev/ttyUSB1) sent "A5010F00"
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) received binary "A581100833001601443C0000C614"
[2022-11-13_12:05:14] (im871a) checkIM871AFrame "A581100833001601443C0000C614"
[2022-11-13_12:05:14] (im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[2022-11-13_12:05:14] (im871a) endpoint 1
[2022-11-13_12:05:14] (im871a) msgid 16
[2022-11-13_12:05:14] (im871a) got crc16 14c6 expected 14c6
[2022-11-13_12:05:14] (im871a) received full frame
[2022-11-13_12:05:14] (device info bytes) "33001601443C0000"
[2022-11-13_12:05:14] (im871a/im170a) info: type=im871a mode=collector firmware=16 hci=01 uid=00003c44
[2022-11-13_12:05:14] (serial /dev/ttyUSB1) sent "A5010500"
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) received binary "A5810616FF000A00B32547371000010001FF07023200010000003620"
[2022-11-13_12:05:14] (im871a) checkIM871AFrame "A5810616FF000A00B32547371000010001FF07023200010000003620"
[2022-11-13_12:05:14] (im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[2022-11-13_12:05:14] (im871a) endpoint 1
[2022-11-13_12:05:14] (im871a) msgid 6
[2022-11-13_12:05:14] (im871a) got crc16 2036 expected 2036
[2022-11-13_12:05:14] (im871a) received full frame
[2022-11-13_12:05:14] (serialtty) closed /dev/ttyUSB1 (detect im871a)
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (device config bytes) "FF000A00B32547371000010001FF0702320001000000"
[2022-11-13_12:05:14] (im871a/im170a) config: other link_mode=ct_n1a id=00103747 media=00 version=01 c_field=00 auto_rssi=01
[2022-11-13_12:05:14] (im871a/im170a) are you there? yes 00103747 im871a firmware: 16
[2022-11-13_12:05:14] (main) opening /dev/ttyUSB1:im871a:t1,c1
[2022-11-13_12:05:14] Started config im871a[00103747] on /dev/ttyUSB1 listening on t1,c1
[2022-11-13_12:05:14] (im871a) on /dev/ttyUSB1
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) before config: /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,253) modem(DTR|RTS)
[2022-11-13_12:05:14] (serial) after config:  /dev/ttyUSB1 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,192,227,75,220,127,248) modem(DTR|RTS)
[2022-11-13_12:05:14] (serialtty) opened /dev/ttyUSB1 fd 3 (im871a)
[2022-11-13_12:05:14] (im871a) get config
[2022-11-13_12:05:14] (serial /dev/ttyUSB1) sent "A5010500"
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) received binary "A5810616FF000A00B32547371000010001FF07023200010000003620"
[2022-11-13_12:05:14] (im871a) checkIM871AFrame "A5810616FF000A00B32547371000010001FF07023200010000003620"
[2022-11-13_12:05:14] (im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[2022-11-13_12:05:14] (im871a) endpoint 1
[2022-11-13_12:05:14] (im871a) msgid 6
[2022-11-13_12:05:14] (im871a) got crc16 2036 expected 2036
[2022-11-13_12:05:14] (im871a) received full frame
[2022-11-13_12:05:14] (im871a) get config completed
[2022-11-13_12:05:14] (im871a) got device id 00103747
[2022-11-13_12:05:14] (main) regular reset of im871a  on /dev/ttyUSB1 will happen every 82800 seconds
[2022-11-13_12:05:14] (im871a) get device info
[2022-11-13_12:05:14] (serial /dev/ttyUSB1) sent "A5010F00"
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) received binary "A581100833001601443C0000C614"
[2022-11-13_12:05:14] (im871a) checkIM871AFrame "A581100833001601443C0000C614"
[2022-11-13_12:05:14] (im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[2022-11-13_12:05:14] (im871a) endpoint 1
[2022-11-13_12:05:14] (im871a) msgid 16
[2022-11-13_12:05:14] (im871a) got crc16 14c6 expected 14c6
[2022-11-13_12:05:14] (im871a) received full frame
[2022-11-13_12:05:14] (im871a) device info completed
[2022-11-13_12:05:14] (im871a) device info: type=im871a mode=collector firmware=16 hci=01 uid=00003c44
[2022-11-13_12:05:14] (im871a) set config to set link mode 0a
[2022-11-13_12:05:14] (serial /dev/ttyUSB1) sent "A501030600020A300100"
[2022-11-13_12:05:14] (serial) EVENT thread interrupted
[2022-11-13_12:05:14] (serial) internal error after select! errno=Invalid argument
[2022-11-13_12:05:14] (serial) received binary "A58104009CF7"
[2022-11-13_12:05:14] (im871a) checkIM871AFrame "A58104009CF7"
[2022-11-13_12:05:14] (im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[2022-11-13_12:05:14] (im871a) endpoint 1
[2022-11-13_12:05:14] (im871a) msgid 4
[2022-11-13_12:05:14] (im871a) got crc16 f79c expected f79c
[2022-11-13_12:05:14] (im871a) received full frame
[2022-11-13_12:05:14] (im871a) set config completed
[2022-11-13_12:05:14] (wmbus) no alarm (expected activity) for im871a
[2022-11-13_12:05:14] (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
[2022-11-13_12:05:14] No meters configured. Printing id:s of all telegrams heard!
[2022-11-13_12:05:14] (serial) waiting for stop
```

Hadatko, 2022-11-14:
Hi, yes and i would say so.

Hadatko, 2023-03-08:
?

weetmuts, 2023-03-08:
It is indeed really strange, the logtimestamps only print a timestamp on the log line...
How could that influence the select function that listens on file descriptors....

Hadatko, 2023-03-15:
Don't ask me :D

closed 2023-06-17 by github-actions[bot]