wmbusmeters/wmbusmeters #668: (apator162) telegram contains a register (80) with unknown size.

pedroke, 2022-11-14:
Hi all,

First of all, many, many thanks for this outstanding project!

Just last week I got the water meter modules installed and one of them is always returning 0 as a total m3.
My investigation:

(wmbusmeters) started meter 4 (Tepla_voda 04578225 auto)
telegram=|704401062582570405077A0E0060852F2F_0F766DFB96010000430600**80**8F67DB8F67DB01010000**10**2F1F01007B01000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000A05F5C1804FFFFFFFFFFFFFFFFFFFF26BCD649|+69
[2022-11-13_13:54:41] (apator162) telegram contains a register (80) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/

So it always reports 0 (zero) as a total m3.
It seems there is indeed **0x80** in the telegram present and the current driver_apator162.cc the register 0x80 is commented out, resulting in size = -1 which in turn breaks the iteration over telegram.
I'm not aware of all the historical reasons why it is commented out, but I suppose, it should be OK to have it there, but I'm not sure what size it should return.. 
In my case the sequence goes like this:

**80** 8F67DB8F67DB01010000 **10**

So there is 0x80 then 10 unknown bytes and then 0x10 which is followed by correct consumption value.
The strange thing is that my other meter doesn't contain 0x80 in its telegrams at all.
They are both from the same series marked with the same HW and FW version on the sticker..

Maybe the ugly fix for me would be to return size = 10 for 0x80, but it would be nice to hear also other opinion :)

Thanks a lot.

pedroke, 2022-11-14:
.. one additional thought..  what about returning size = 1 instead of -1 when register identifier is not recognized? This way it would continue and not fail in any case?

weetmuts, 2022-11-14:
No, you cannot continue without knowing the size of the register. You do not know where the next register starts without it.

However the fix, is indeed as you suggested 0x80 returns 10. I have fixed this.
The comment was an old remnant from before we knew that the first 8 bytes should be skipped before finding registers.

Please test!

weetmuts, 2022-11-14:
Their protocol is dynamic, and the registers come and go. Also I am guessing that there are sub-bits inside the registers as well. But since I am not really interested in decoding non-mbus stuff, I am not going to spend more stuff figuring out what the different registers mean as long as wmbusmeters can reliably pick out the consumption. 

Other developers are of course welcome to help out decoding the protocol more if they like. :-)

pedroke, 2022-11-14:
Thanks a lot. An idea to return size=0 for unknown registers would ensure that the whole datagram would be parsed, even when some identifiers are changed in the future.. Simply the iteration would go to next byte.. I think it would be failsafe for future changes, at least partialy :) I will also try to monitor the datagrams to see if I can find anything useful inside :) Thanks again..

closed 2022-11-15 by weetmuts