wmbusmeters/wmbusmeters #669: primary address not functional

autega, 2022-11-14:
@autega Great that you got it working! 

Actually wmbusmeters already supports the primary address. :-)
Use id = 17 for wmbusmeters to use the primary address 17 instead.

 Please test if you have the time.

_Originally posted by @weetmuts in https://github.com/weetmuts/wmbusmeters/issues/532#issuecomment-1298335790_

autega, 2022-11-14:
When testing I´ve got the following:
Primary Address of the device is 4.

```
/wmbusmeters # ./wmbusmeters  --pollinterval=15s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter sensostar:MAIN:mbus 21950207 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
MyTempMeter     21950207        1949 kWh        1047.549 m³     OK      2022-11-01 14:08.35
/wmbusmeters # ./wmbusmeters  --pollinterval=15s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter sensostar:MAIN:mbus 4 NOKEY

Not a valid id nor a valid meter match expression "4"
```

Any ideas what I´m doing wrong.

autega, 2022-12-06:
@weetmuts Could you please take a look?

weetmuts, 2022-12-06:
Yes, its on my todo now that all the drivers have been refactored.

autega, 2022-12-06:
Great to hear. Let me know if there's anything I can do.

weetmuts, 2023-02-05:
on the todo!

autega, 2023-03-07:
Hey. Any updates on this issue?
Here´s the log with the newest version of wmbusmeters.
Still the same...

`/wmbusmeters # ./wmbusmeters --pollinterval=6s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter sensostar:MAIN:mbus 21950207 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
MyTempMeter     21950207        12596 kWh       4622.799 m³     OK      2022-12-31      6916 kWh        2023-03-07 07:49.34
/wmbusmeters # ./wmbusmeters --pollinterval=6s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter sensostar:MAIN:mbus 4 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
/wmbusmeters # ./wmbusmeters --debug --pollinterval=6s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter sensostar:MAIN:mbus 4 NOKEY
(wmbusmeters) version: 1.11.0-142-gc84c4d3
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400 
(config) number of meters: 1
(meter) created MyTempMeter sensostar  encrypted
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(152,236,6,100,246,101,165,9) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(152,236,6,100,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(meter) not polling from bad id  "4" with wrong length
(meter) not polling from bad id  "4" with wrong length
(meter) not polling from bad id  "4" with wrong length
(meter) not polling from bad id  "4" with wrong length
(meter) not polling from bad id  "4" with wrong length
(meter) not polling from bad id  "4" with wrong length`

weetmuts, 2023-03-07:
Yes, I know. Still working on it. It is a pretty complicated fix because the whole address (id) code must be rewritten.

weetmuts, 2023-05-07:
Still on the todo!

weetmuts, 2024-03-03:
@autega its finally done, you can now use p0 to p250 for the primary addressing.
`wmbusmeters --pollinterval=5s /dev/ttyUSB1:mbus:2400 TEMP piigth:mbus p0 NOKEY`

autega, 2024-03-11:
@weetmuts now using primary addressing for my three electricity meters. 
Thank you for that complicated fix.

closed 2024-05-11 by weetmuts