wmbusmeters/wmbusmeters #670: Problem with the heat meter in Home Assistant

barcik75, 2022-11-14:
So far I have configured two water meters (hot, cold) via wmbusmeters to mqtt. Both worked fine, although in the logs I had this information:
```Broker ha.barcik.synology.me will be used.
MQTT Discovery ...
 Adding meter: MainColdWater ...
  File /config/wmbusmeters/etc/mqtt_discovery/apator162.json not found.
 Adding meter: MainHotWater ...
  File /config/wmbusmeters/etc/mqtt_discovery/apator162.json not found.
```
And they show the correct values.
Configuration of the wmbusmeters plugin:
```data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: |-
  loglevel=normal
  device=auto:t1
  donotprobe=/dev/ttyAMA0
  logtelegrams=false
  format=json
  logfile=/config/wmbusmeters/logs/wmbusmeters.log
  meterfiles=/config/wmbusmeters/logs/meter_readings
  meterfilesaction=append
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=MainColdWater
    driver=apator162
    id=02820633
    key=00000000000000000000000000000000
  - |-
    name=MainHotWater
    driver=apator162
    id=03986285
    key=00000000000000000000000000000000
mqtt:
  host: ha.barcik.synology.me
  port: 1883
  user: mqtt
  password: topsecret
```
in configuration.yaml I have:
```mqtt:
  sensor:
    - name: Cold water - main meter
      state_topic: "wmbusmeters/MainColdWater"
      json_attributes_topic: "wmbusmeters/MainColdWater"
      value_template: '{{ value_json["total_m3"] }}'
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: water
      unique_id: 02820633
    - name: Hot water - main meter
      state_topic: "wmbusmeters/MainHotWater"
      json_attributes_topic: "wmbusmeters/MainHotWater"
      value_template: '{{ value_json["total_m3"] }}'
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: water
      unique_id: 03986285
```
And as I mentioned, everything shows correctly.
Today I got the key to decode the frame from the heat meter and added it to the HA. After entering the name, driver, id and key:
```  - |-
    name=MainHeat
    driver=multical603
    id=82750409
    key=00000000000000000000000000000000
``` 
to configuration file, I got in logs:
```
Adding meter: MainHeat ...
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_energy_forward_kwh/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_energy_returned_kwh/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_EXTERNAL_ALARM/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_LOW_BATTERY_LEVEL/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_SENSOR_T1_ABOVE_MEASURING_RANGE/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_SENSOR_T1_BELOW_MEASURING_RANGE/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_SENSOR_T2_ABOVE_MEASURING_RANGE/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_SENSOR_T2_BELOW_MEASURING_RANGE/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_TEMP_DIFF_WRONG_POLARITY/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/82750409_status_VOLTAGE_INTERRUPTED/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_t1_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_t2_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_total_energy_consumption_kwh/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_total_volume_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/82750409_volume_flow_m3h/config
MQTT Discovery cleanup...
```
And the following entities appeared on the dashboard:
[image]
But most of the values are set to 0 except one. When I display the attributes of the MainHeat entity total energy consumption I have the following values:
```state_class: total
media: heat
meter: multical303
id: '82750409'
total_energy_kwh: 60
total_volume_m3: 1.78
forward_energy_m3c: 92
return_energy_m3c: 76
forward_c: 35.37
return_c: 31.42
actual_flow_m3h: 0
status: OK
meter_date: '2022-11-14'
target_energy_kwh: 0
target_volume_m3: 0
target_date: '2022-11-01'
timestamp: '2022-11-14T22:53:22Z'
device: rtlwmbus[00000001]
rssi_dbm: 146
unit_of_measurement: kWh
device_class: energy
icon: mdi:home-lightning-bolt-outline
friendly_name: MainHeat total energy consumption
```
As I added logging values from counters, I have in a separate file:
```
{"media":"heat","meter":"multical303","name":"MainHeat","id":"82750409","total_energy_kwh":60,"total_volume_m3":1.78,"forward_energy_m3c":92,"return_energy_m3c":76,"forward_c":34.84,"return_c":31.01,"actual_flow_m3h":0,"status":"OK","meter_date":"2022-11-14","target_energy_kwh":0,"target_volume_m3":0,"target_date":"2022-11-01","timestamp":"2022-11-14T22:56:00Z","device":"rtlwmbus[00000001]","rssi_dbm":144}
```
Which indicates that the frame is decoded correctly and the values by wmbusmeters are sent to mqtt. However, they are not received correctly by HA.
I also tried to add additional sensors in the mqtt section in configuration.yaml, similar to water sensors:
```
    - name: TestHeat
      state_topic: "wmbusmeters/MainHeat"
      json_attributes_topic: "wmbusmeters/MainHeat"
      value_template: '{{ value_json["total_energy_kwh"] }}'
      unit_of_measurement: "kWh"
      state_class: total_increasing
      device_class: energy
      unique_id: 82750409
```
However, after the HA restart, the sensor did not appear and it is not in the list of all entities.
And to be honest, I don't know what to do next to make the heat meter readings show up in HA. I'm running out of ideas...

barcik75, 2022-11-17:
If it can help:
```
 Logger: homeassistant.helpers.template
Source: helpers/template.py:583
First occurred: 15:35:14 (30 occurrences)
Last logged: 15:36:34

    Template variable warning: 'dict object' has no attribute 't1_temperature_c' when rendering '{{ value_json.t1_temperature_c }}'
    Template variable warning: 'dict object' has no attribute 't2_temperature_c' when rendering '{{ value_json.t2_temperature_c }}'
    Template variable warning: 'dict object' has no attribute 'volume_flow_m3h' when rendering '{{ value_json.volume_flow_m3h }}'
    Template variable warning: 'dict object' has no attribute 'current_status' when rendering '{{ 'LOW_BATTERY_LEVEL' in value_json.current_status }}'
    Template variable warning: 'dict object' has no attribute 'total_energy_consumption_kwh' when rendering '{{ value_json.total_energy_consumption_kwh }}'

```

Volcano0, 2022-12-01:
I had similar issue, it seems a general problem. Try to downgrade Wmbusmeters to 0.2.23.

barcik75, 2022-12-10:
I managed to do a workaround. I added topics that interested me in the MQTT section. Everything worked until today, until the update. Now discovery and entered topics do not work. Do you check the code before releasing it?

BIBOLV, 2023-01-03:
Hi,
Meter drivers were rewritten and multical was one that had changes that were not backwards compatible for some fields and meters - more [here](https://github.com/weetmuts/wmbusmeters/issues/734#issuecomment-1346111003).

closed 2023-04-05 by github-actions[bot]

thecem, 2024-07-07:
> If it can help:
>
> ```
>  Logger: homeassistant.helpers.template
> Source: helpers/template.py:583
> First occurred: 15:35:14 (30 occurrences)
> Last logged: 15:36:34
>
>     Template variable warning: 'dict [...]

I createt a pull request https://github.com/wmbusmeters/wmbusmeters-ha-addon/pull/41