wmbusmeters/wmbusmeters #671: Not enough data for decoding?

JeromeLeary, 2022-11-17:
Please what am I doing wrong? I'm using an AMB8665 dongle on Fedora 37 laptop
I feel I am being stupid...

This is what I type... but it aborts

$ wmbusmeters --debug /dev/ttyUSB0:amb8465:c1

(wmbusmeters) version: rpm-build_1.9.0
(config) using device: /dev/ttyUSB0:amb8465:c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" amb8465
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary "96FF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF01090000FFFFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
/usr/include/c++/12/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

weetmuts, 2022-11-18:
? it looks like a problem to me. did you solve it or did you accidentally close it?

JeromeLeary, 2022-11-19:
I didn't solve it. Its still a problem.. I just re-opened it...  I be very grateful if you can help solve it. Thank you

weetmuts, 2022-11-28:
Can you try the latest version on github? I just made some fixes to the amber dongle driver.

JeromeLeary, 2022-11-29:
Ok I tried to install the latest version from github on my Fedora 37 linux and got the following error message:

[jeromeleary@fedora wmbusmeters-master]$ ./configure
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for rtlsdr_get_device_count in -lrtlsdr... no
configure: error: Could  not find rtlsdr library. Try: sudo apt install librtlsdr-dev

[jeromeleary@fedora wmbusmeters-master]$ sudo yum install librtlsdr-dev
[sudo] password for jeromeleary: 
Last metadata expiration check: 0:14:36 ago on Mon 28 Nov 2022 16:41:05 EST.
No match for argument: librtlsdr-dev
Error: Unable to find a match: librtlsdr-dev

[jeromeleary@fedora ~]$ uname -r
6.0.9-300.fc37.x86_64

What am I doing wrong?

weetmuts, 2022-12-29:
Perhaps the missing package is called rtl-sdr-devel ?

closed 2023-03-31 by github-actions[bot]