wmbusmeters/wmbusmeters #672: Techem Radio4 meters for cold & warm water /  Techem vario 4.1.1  (heat meter) decoding

thisandre, 2022-11-17:
We are using these meters in or house (all data is encrypted but I got the keys from Techem)
I made following observations:
- Warm and cold meters are looking the same (Both marked as Techem-Radio4), but the JSON decoding for cold water indicates the meter as Weh_07, as the warm water meter decodes as "warm water - driver not found". 
Question: Which driver / meter to use??
- The Techem Vario 4.1.1 is indicated as "heat", which is correct, but does not find a driver.
Same question: which driver to use??
According to the public available info about these meters, the information that should be available is the consumption of the previous period and the previousdate.  
The heat consumption is recorded in the telegram as "4406", followed by 4 bytes of consumption value (in kWh). 
For the watermeters, the consumption is recorded as "4315", followed by 3 bytes of consumption value (in m3/10). 
For all meters, the previous period is indicated as "426", followed by 2 bytes with the date.  
All other data in the telegrams is manufacturer specific and obviously hidden with a key.  (see attached file with the telegram details)
Do I have to write a new driver for these meters, or is there one which I can use? 

[logtel1.txt](https://github.com/weetmuts/wmbusmeters/files/10031652/logtel1.txt)

MartinBachmannHD, 2022-11-17:
...as I am struggling with Techem as well - may I ask which hardware and which protocol you are using? I used the CUL with 'T' and that was somehow unsuccessful with radio 4 / vario 4...

thisandre, 2022-11-17:
I'm using just a simple DVB-T/FM/DAB USB stick (820T2 & SDR) with rtlwmbus.   With this I can receive all meters from 1 floor down and 1 floor up in the house.

BIBOLV, 2022-11-18:
Try mkradio4
```
{
    "media":"heat",
    "meter":"mkradio4",
    "name":"",
    "id":"67627875",
    "total_m3":4902.9,
    "target_m3":2.5,
    "timestamp":"2022-11-18T22:26:53Z"
}
```

thisandre, 2022-11-19:
I tried already near all drivers!  The "total_m3" and "target_m3" don't match. Neither is the previous date indicated. 
The only values that are available in this message (apart from the "vendor secrets") are the previous consumption time and previous consumption m3, as I described in the issue. 
I'm intended to create a driver for this but I'm struggling with the set-up.  According to the instructions, one should create a file called "driver_xxx.cc" but in the src directory, lots of "meter_xxx.cc" files also exist.  Do we need both, or can everything be put together in de driver_xxx.cc file??

BIBOLV, 2022-11-20:
"driver_xxx.cc" is correct one, "meter_xxx.cc" are old standard files left that are being refactored to new style.

thisandre, 2022-11-20:
The driver for Techem Vario451mid is using the same codes, with a different interpretation of the data fields as the Techem Vario411. (both using 0x04 / 0x17 in di.addDetection). How can we differentiate between the 2 devices and send corresponding JSON messages?

BIBOLV, 2022-11-20:
Diver differentiates message using manufacturer, media type and media version. Could you please specify where do you see overlap - I did a quick search and could not find any. For 67627875 meter detection line would be:
```
di.addDetection(MANUFACTURER_TCH, 0x04,  0x28);
```

thisandre, 2022-11-21:
Sorry, my mistake, I was working on an elderly version, where the file "meter_detection.h" was still there and claimed the media type & version for the vario451mid. This file seems to be gone in the actual version.

hemonu, 2022-11-23:
Look's like I have the same combination of meter manufacturers and "Techem Radio4" as thisandre. How did you get the AES key from Techem?

thisandre, 2022-11-23:
Just asked Techem and got them after a long procedure of safety aspects (mails with separated passwords etc.). I also had to sign that I would not disclose the keys to third parties.  I'm doing the "Hausverwaltung" in our house myself and we rent the meters from Techem. I needed the info to do my job. I also asked them for a technical description of the telegram format describing the manufacturer specific information.  Let's see what comes out ...

MartinBachmannHD, 2022-11-23:
I contacted Techem via the contact page on their public website a few days ago - but without success until now. But I am not a direct customer of Techem, only indirectly via our 'Hausverwaltung'. Is there a special page / special email to better get in contact with Techem about the keys? Ideally it would be a self-service... :-)

BIBOLV, 2023-01-03:
@thisandre - how tests are progressing?

closed 2023-04-05 by github-actions[bot]

k1tfh, 2023-11-21:
@thisandre  - i got 2 keys from techem. which one should i use? one labeled "Keys A" and the other "Key S".

k1tfh, 2023-11-21:
I am confused. I checked some of your telegrams in your logtel1.txt with [https://wmbusmeters.org](url) without any Decryption key. 

This is what comes out: 

Auto driver  : mkradio4a
Best driver  : topaseskr 05/47
Using driver : mkradio4a 00/00

{
    "media":"warm water",
    "meter":"mkradio4a",
    "name":"",
    "id":"66953825",
    "target_m3":16.1,
    "target_date":"2021-12-31",
    "timestamp":"2023-11-21T21:50:21Z"
}

So how did you manage to collect such large numbers of telegrams? It seems like the meters from techem are both encrypted and decrypted. Using no key or the key from techem leading to a block.

thisandre, 2023-11-22:
I don't know what meters you're using, but in my case, I was not able to read the data without decoding.  Logtel1.txt is already decoded by the wmbus program using the "Keys A" data for that specific meter (identified with the Id).  As I mentioned, I was not able to decode the rest of the message, only the part giving the end-of-year values. In my case, the encryption key is meter specific, meaning I need a key for every meter (Which I got from Techem). I don't know whether Techem is using other setups for other properties (as common keys for the property, keys per floor, or something like that). In my case, I'm collecting data from over 30 meters which are definitely not readable without the keys!

k1tfh, 2023-11-22:
Perhaps this could help: [https://github.com/wmbusmeters/wmbusmeters/issues/1102#issuecomment-1823358611](url)

I now receive the current values with a delay of a few seconds to a few minutes. That's enough for me in combination with Home Assistant. I still get a few error messages, but the values are coming in. I hope it stays that way.

thisandre, 2023-11-23:
Glad that you made it!  It is correct that the devices send out the info roughly at some minutes interval.
It however does not solve the problem of decoding the 'proprietary part' of the message, which should disclose the current consumption and other info. As I said, I already tried various combinations of aes-keys and message lengths but did not get any further as getting the yearly values.

fhb, 2023-12-15:
I'm living in a rental and am trying to get the keys for my water and heat counters. I just sent an email to service@techem.de requesting the keys providing all the information from my end of year bill. I'll report back wether they refuse my request or provide the keys.

@k1tfh re you also a direct customer of techem? 
@MartinBachmannHD did you get the keys?

fhb, 2023-12-19:
So they won't give the keys to a third party:

> vielen Dank für Ihre Anfrage vom 15. Dezember 2023.
> 
> Bitte haben Sie Verständnis, dass wir die von Ihnen gewünschten Informationen nur unserem Vertragspartner zur Verfügung stellen können.
> 
> Bitte wenden Sie sich an Ihre Hausverwaltung.
> 
> Diese kann die OMS Schlüssel gerne bei uns anfordern.

jsuchal, 2025-10-15:
@PetrStepka this looks pretty good! Could you share the request text template & how long it took to respond?

Nopel2020, 2025-12-30:
Hello @PetrStepka!

Did you get a response from Techem? I am in the same situation, trying to get my keys from Techem here in Belgium. So far no response.