wmbusmeters/wmbusmeters #674: energy meter EGM Elgama

palibrzuch, 2022-11-18:
Hello.
I have a counter as in the attachment.
Is it possible to add support to him?
Logs also in the attachment.
If there is anything else to read, I'll do it.
Regards

[egm.txt](https://github.com/weetmuts/wmbusmeters/files/10040259/egm.txt)
[image]

BIBOLV, 2022-11-20:
Hi,
Do you have encryption key? Do you have more details about this meter, like are wmbus telegrams OMS compliant? 
I looked at meter manufacturer site but could not find this model.

palibrzuch, 2022-11-22:
Unfortunately, I don't have a decryption key. I substitute the standard 32x0.
I also added a second counter without the decryption key.

Logs below.
Unfortunately, I cannot find any information about this counter.
I wrote to the operator for read access - impossible.

https://wmbusmeters.org/analyze/1E44ED146390821301027A740010052BF31A1A50A8402ABD4BB61316E6F319

(meter) licznik_eon2: for me? 13829220 in 13829063
(meter) licznik_eon2: not for me: not my id
(wmbus) telegram from 13829220 ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-11-22 08:35:32.000;48;11;13829220;0x1e44ed142092821301027a640010052c3f95ce1c88c37e4c55692af78e970a<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=31.
(serial) received ascii "T1;1;1;2022-11-22 08:35:32.000;98;95;13829063;0x1e44ed146390821301027a740010052bf31a1a50a8402abd4bb61316e6f319<0A>T1;1;1;2022-11-22 08:35:32.000;82;94;13829063;0x1e44ed146390821301027a740010052bf31a1a50a8402abd4bb61316e6f319<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-11-22 08:35:32.000;98;95;13829063;0x1e44ed146390821301027a740010052bf31a1a50a8402abd4bb61316e6f319<0A>T1;1;1;2022-11-22 08:35:32.000;82;94;13829063;0x1e44ed146390821301027a740010052bf31a1a50a8402abd4bb61316e6f319<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) licznik_eon: for me? 13829063 in 13829063
(meter) licznik_eon: yes for me
(meter) licznik_eon(1) auto  handling telegram from 13829063
(meter) licznik_eon 13829063 "1E44ED146390821301027A740010052BF31A1A50A8402ABD4BB61316E6F319"
(wmbus) parseDLL @0 31
(telegram) DLL L=1e C=44 (from meter SND_NR) M=14ed (EGM) A=13829063 VER=01 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=98
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 13829063 mfct: (EGM) Elgama Elektronika, Lithuania (0x14ed) type: Electricity meter (0x02) ver: 0x01
(telegram) TPL CI=7a ACC=74 STS=00 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) licznik_eon2: for me? 13829063 in 13829063
(meter) licznik_eon2: yes for me
(meter) licznik_eon2(2) auto  handling telegram from 13829063
(meter) licznik_eon2 13829063 "1E44ED146390821301027A740010052BF31A1A50A8402ABD4BB61316E6F319"
(wmbus) parseDLL @0 31
(telegram) DLL L=1e C=44 (from meter SND_NR) M=14ed (EGM) A=13829063 VER=01 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=98
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "2BF31A1A50A8402ABD4BB61316E6F319"
(TPL) IV ED146390821301027474747474747474
(TPL) decrypted  "85BEDD505A8276051A1A46BDF8EEB73F"
(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 13829063 mfct: (EGM) Elgama Elektronika, Lithuania (0x14ed) type: Electricity meter (0x02) ver: 0x01
(telegram) TPL CI=7a ACC=74 STS=00 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21

BIBOLV, 2022-11-22:
Without key it will not possible to read values.

palibrzuch, 2022-11-23:
Obtaining such a key is unlikely to be possible. I wrote to the operator and producer. We'll see.
We won't do anything without this key?

[image]

BIBOLV, 2023-01-02:
It is not possible to proceed without encryption key that would be used to decrypt telegrams and process values in them. Closing issue for now. Please reopen if you manage to get key and collect decrypted telegrams.

closed 2023-01-02 by BIBOLV