wmbusmeters/wmbusmeters #675: Is the Sharky775 encrypted?

JeromeLeary, 2022-11-18:
I've just installed 32 Sharky 775 meters and I cant get any response from them using an AMB8665-M USB dongle. Does anyone know what the problem might be? Are these meters encrypted? 

% wmbusmeters --debug --logtelegrams --oneshot /dev/ttyUSB0:amb8465:t1 heatmeter sharky 72503500 NOKEY

(wmbusmeters) version: rpm-build_1.9.0
(config) using device: /dev/ttyUSB0:amb8465:t1 
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0" amb8465
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary "2EFF8A7A00780080710200010000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF01090000FFFFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
/usr/include/c++/12/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

weetmuts, 2022-11-25:
Can you check from a manual how long the the config response is for the 8665-M dongle? It looks like it gives a shorter config response that 8465?

JeromeLeary, 2022-11-25:
Hello, I actually gave up with the 8665-M because I was receiving corrupted transmissions with non ascii bytes. Even a terminal capture program (Minicom) failed to sync with it. 
So I bought a RTL-SDR which worked brilliantly out of the box with your wonderful software.  Anyway thank you for your hard work.

weetmuts, 2022-12-05:
Can you test with the latest version on github 1.10.1 I think I have fixed that bug.

JeromeLeary, 2022-12-07:
Will do. I'm using Fedora, will this install the latest version?  
  **dnf install wmbusmeters**
..or do I need to rebuild it from github? Thanks.

closed 2023-03-18 by github-actions[bot]