wmbusmeters/wmbusmeters #676: Decoding Flowis

vierundvierzig, 2022-11-19:
Hello,

I have water meters from Flowis ([some datasheet](https://odczytyradiowe.pl/Katalogi/eng/LibraRS-walkby-radio-system.pdf)) and wmbusmeteres can't read it. I can clearly see signals on 868.0MHz using SDR#. I tried commands:

- rtl_433 -f 868.0M
- rtl_sdr -f 868.0M -s 1600000 - 2>/dev/null | rtl-wmbus/build/rtl_wmbus
- wmbusmeters auto:c1,t1

but there is no signals from Flowis.

Can you point me what should I do to start new driver's decoding process?

Many thanks!

BIBOLV, 2022-11-19:
You can try different frequencies and just use rtl_sdr|rtl_wmbus like this:

rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s

Move up/down from the proper frequency using steps of 0.05.

vierundvierzig, 2022-11-23:
Thank you BIBLOV for your response.

Unfortunately, no result. I even moved freq by 0.02.

Can I record somehow this signals (I can see them on SDR #) for further investigation? 

PS. It only transmits from Monday to Friday, 7-15.

BIBOLV, 2023-01-03:
Hi,
Did you manage to get signal?

vierundvierzig, 2023-01-08:
Hello,
unfortunately no, I have no idea where to start. I will really appreciate if you point me some tutorials or tips.
Thank you!

BIBOLV, 2023-01-08:
There is one more open [issue](https://github.com/weetmuts/wmbusmeters/issues/713)  mentioning similar meter as your. Does your meter also has OMS marking? I did a quick search about this module and did not find that wmbus would be mentioned anywhere, just lora.

vierundvierzig, 2023-01-08:
No, there is no OMS marking, take a look at the picture (black cable is for external antenna):

[image]

BIBOLV, 2023-01-09:
It could be then that those modules are not using wmbus but lora or some other protocol. That could be why see activity in freqency but nothing when looking for wmbus. You can try to talk to [them](https://metering.com.pl/oferta/doradztwo-techniczne/) as they are manufacturers if I'm not mistaken. Or try to find some written confirmation that wmbus is being used. Or, if in picture are your old modules, try to disassemble them and look at some chips, it could give maybe some hints what is being used there.

vierundvierzig, 2023-01-10:
I have TTGO Lora, tried today but no success - no messages on LoRa and FSK modulation.
I can not disassemble them.
Still looking how to recognize signal.

vierundvierzig, 2023-01-15:
Manufacturer said that it is closed protocol and it's impossible to understand it without source code.... Anyway :)

I can spot the signal on spectrogram:
[image]
As you can see, it is very short and I think it is FSK modulation. Any new ideas?

BIBOLV, 2023-01-15:
I would bet on lora as it does not make sense to create new protocol in times when there are so many already created, but that`s just my guess.

vierundvierzig, 2023-01-15:
Do you have any guess how to decode it?

weetmuts, 2023-01-15:
@merbanan do you have any input on a mysterious fsk radio protocol? :-)

merbanan, 2023-01-15:
@vierundvierzig use rtl_433 and follow the https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md instructions. This device is highly likely not wmbus compatible and then you are welcome over to the rtl_433 project for support.

vierundvierzig, 2023-01-15:
Thank you very much for this link! I will follow instruction on Tuesday morning (water meter has specific time when it works) and post some result. 
Can we continue this topic here or should I open new one in rtl_433 project?

merbanan, 2023-01-15:
Open a new issue there and then add a link to it from here.

vierundvierzig, 2023-01-16:
Topic will be continued [here](https://github.com/merbanan/rtl_433/issues/2317). Thank you very much for help!

closed 2023-01-16 by vierundvierzig

merbanan, 2023-01-16:
So far we know that the Flowis protocol is using 100 kbps FSK with a 200 bit message excluding the preamble. It is not compatible with anything wmbus. Hopefully it will be added to the rtl_433 project if we get enough sample messages to analyse.

merbanan, 2023-02-04:
This device is now supported in the rtl_433 project.