wmbusmeters/wmbusmeters #677: NexMeter Gas Meter support - 169 Mhz N mode

optiluca, 2022-11-20:
Hi.  My gas meter appears to be a 169 Mhz N mode device (this seems to be the case for every gas meter in my area).

[image]

I see quite a few issues regarding support for devices on this band, but none seem to have been resolved.

I have an rtl-sdr,  patience, but no competence. :-)  What should I do to gather data to allow for this device to be supported?

Does it make sense to keep this ticket open, or should I chime in an already-existing one, for a different 169 Mhz N mode device?

Thanks!

BIBOLV, 2022-11-20:
Hi,
Firstly support for N must be implemented in rtl_wmbus, so if you can, you can record some signal samples and post them [here](https://github.com/xaelsouth/rtl-wmbus/issues/26).
You can close this issue for now and open once more when support in rtl_wmbus is implemented and we can move forward.

optiluca, 2022-11-20:
OK, thanks.  I'll try recording some data and posting it on the rtl_wmbus issue.

closed 2022-11-20 by optiluca

weetmuts, 2023-07-02:
Note that there is a 169MHz hardware dongle that works with wmbusmeters: amb3665
(Search for Amber AMB 3665-M)
Make sure you buy a matching suitable antenna as well.