wmbusmeters/wmbusmeters #681: Support for Kamstrup Wireless M-Bus Dongle?

thorsten, 2022-11-20:
My local water company rejected by wish to get the AES key, but it's possible to get a Wireless M-Bus Dongle directly from Kamstrup. It's this dongle: https://www.kamstrup.com/en-en/heat-solutions/meters-devices/accessories/wireless-mbus-dongle

Does wmbusmeters work with this dongle?

weetmuts, 2022-11-20:
Yes, it should work!

weetmuts, 2022-11-20:
But check if it comes with an mbus master dongle. otherwise you have to buy one.

thorsten, 2022-11-20:
Thanks, I have to talk to Kamstrup support first 🙂

weetmuts, 2022-11-20:
https://m.aliexpress.com/item/1005004540180026.html?spm=a2g0n.detail.1000014.11.4881623f7252Ad&gps-id=platformRecommendH5&scm=1007.18499.315613.0&scm_id=1007.18499.315613.0&scm-url=1007.18499.315613.0&pvid=73b385d3-13c6-404d-90fa-efa6949beac1&_t=gps-id:platformRecommendH5,scm-url:1007.18499.315613.0,pvid:73b385d3-13c6-404d-90fa-efa6949beac1,tpp_buckets:668%232846%238111%231996&pdp_npi=2%40dis%21SEK%21280.92%21252.82%21%21%21%21%21%40210323a516689627951238785e0495%2112000029535143565%21rec

weetmuts, 2022-11-20:
make sure you buy the usb mbus master.

weetmuts, 2022-11-20:
you select the type using the color option! :-)

closed 2023-02-19 by github-actions[bot]

PatrickAmann, 2024-04-05:
Since my local water company wont provide the key but offered to get a wireless M bus dongle configured by kamstrup directly to be used with my water meter, I was wondering if the key can somehow be extracted from this OEM dongle? 

The kamstrup software describes a function to "Get Meter X Settings" (left screenshot on the bottom left:

https://documentation.kamstrup.com/docs/Wireless_M_Bus_Dongle/en-GB/User_guide/CONT74D29DEC0A39465BB9DA4221045D4C02/?page=10 

This makes me wonder if the key can be extracted by either using this method or some other methods since I dont feel like using a second dongle for the exact same thing as my other one already does.