wmbusmeters/wmbusmeters #682: Adding Landis+Gyr Ultraheat T230

thomasa88, 2022-11-20:
Hi,

I have a Landis+Gyr Ultraheat T230 that I'm trying to read using rtl-sdr and wmbusmeters. I'd like to add it to wmbusmeters.

```
build/wmbusmeters --logtelegrams auto:t1
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 69153171
          manufacturer: (LUG) Landis Gyr, Germany (0x32a7)
                  type: Heat meter (0x04)
                   ver: 0x07
                device: rtlwmbus[00000001]
                  rssi: 114 dBm
                driver: unknown!
telegram=|_3A44A7327131156907047A0A0000202F2F0C06408703000C14267763000B3B1600F00B2D0100000B5A0702000B5E660200046D2F12D42B0F3C0000|+501
```

I saw that qheat_55_us identified as Landis Gyr, but with another version: !546

It failed quite badly when I tried using it:

```
wmbusmeters --analyze=:qheat_55_us simulation_t230.txt 
Auto driver  : not found!
Best driver  : c5isf 24/28
Using driver : qheat_55_us 00/00
000   : 3a length (58 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a732 dll-mfct (LUG)
004   : 71311569 dll-id (69153171)
008   : 07 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 0a tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 0C dif (8 digit BCD Instantaneous value)
018   : 06 vif (Energy kWh)
019 C!: 40870300 ("total_energy_consumption_kwh":38740)
023   : 0C dif (8 digit BCD Instantaneous value)
024   : 14 vif (Volume 10⁻² m³)
025 C?: 26776300
029   : 0B dif (6 digit BCD Instantaneous value)
030   : 3B vif (Volume flow l/h)
031 C?: 1600F0
034   : 0B dif (6 digit BCD Instantaneous value)
035   : 2D vif (Power 10² W)
036 C?: 010000
039   : 0B dif (6 digit BCD Instantaneous value)
040   : 5A vif (Flow temperature 10⁻¹ °C)
041 C?: 070200
044   : 0B dif (6 digit BCD Instantaneous value)
045   : 5E vif (Return temperature 10⁻¹ °C)
046 C?: 660200
049   : 04 dif (32 Bit Integer/Binary Instantaneous value)
050   : 6D vif (Date and time type)
051 C!: 2F12D42B ("device_date_time":"2022-11-20 18:47")
055 C?: 0F manufacturer specific data 3C0000

{
    "media":"heat",
    "meter":"qheat_55_us",
    "name":"",
    "id":"69153171",
    "device_date_time":"2022-11-20 18:47",
    "total_energy_consumption_kwh":38740,
    "key_date":null,
    "key_date_kwh":null,
    "prev_month":null,
    "prev_1_month_kwh":null,
    "prev_2_month_kwh":null,
    "prev_3_month_kwh":null,
    "prev_4_month_kwh":null,
    "prev_5_month_kwh":null,
    "prev_6_month_kwh":null,
    "prev_7_month_kwh":null,
    "prev_8_month_kwh":null,
    "prev_9_month_kwh":null,
    "prev_10_month_kwh":null,
    "prev_11_month_kwh":null,
    "prev_12_month_kwh":null,
    "prev_13_month_kwh":null,
    "timestamp":"2022-11-20T21:00:32Z"
}
```

So I just let analyze run without options and got this:

```wmbusmeters --analyze simulation_t230.txt 
Auto driver  : not found!
Best driver  : c5isf 24/28
Using driver : c5isf 24/28
000   : 3a length (58 bytes)(OK)
001   : 44 dll-c (from meter SND_NR)
002   : a732 dll-mfct (LUG)
004   : 71311569 dll-id (69153171)
008   : 07 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 0a tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 0C dif (8 digit BCD Instantaneous value)
018   : 06 vif (Energy kWh)
019 C!: 40870300 ("total_energy_consumption_kwh":38740)
023   : 0C dif (8 digit BCD Instantaneous value)
024   : 14 vif (Volume 10⁻² m³)
025 C!: 26776300 ("total_volume_m3":6377.26)
029   : 0B dif (6 digit BCD Instantaneous value)
030   : 3B vif (Volume flow l/h)
031 C!: 1600F0 ("volume_flow_m3h":2200.016)
034   : 0B dif (6 digit BCD Instantaneous value)
035   : 2D vif (Power 10² W)
036 C!: 010000 ("power_kw":0.1)
039   : 0B dif (6 digit BCD Instantaneous value)
040   : 5A vif (Flow temperature 10⁻¹ °C)
041 C!: 070200 ("flow_temperature_c":20.7)
044   : 0B dif (6 digit BCD Instantaneous value)
045   : 5E vif (Return temperature 10⁻¹ °C)
046 C!: 660200 ("return_temperature_c":26.6)
049   : 04 dif (32 Bit Integer/Binary Instantaneous value)
050   : 6D vif (Date and time type)
051 C!: 2F12D42B ("last_month_date":"2022-11-20 18:47")
055 C?: 0F manufacturer specific data 3C0000

{
    "media":"heat",
    "meter":"c5isf",
    "name":"",
    "id":"69153171",
    "total_energy_consumption_kwh":38740,
    "total_volume_m3":6377.26,
    "status":"OK",
    "volume_flow_m3h":2200.016,
    "power_kw":0.1,
    "last_month_date":"2022-11-20 18:47",
    "flow_temperature_c":20.7,
    "return_temperature_c":26.6,
    "timestamp":"2022-11-20T20:57:47Z"
}

```

This looks pretty fine to me - are the fields shown common/default fields?

I'm not used to this bus, so I could not figure out the manufacture specific part, but I think it is in this document:
https://www.etaz.rs/imgDocuments/696/Tehni%C4%8Dke%20karakteristike%20T230.pdf

Any pointers? :)

[simulation_t230.txt](https://github.com/weetmuts/wmbusmeters/files/10050678/simulation_t230.txt)

dzikiBOB, 2023-01-04:
I do have same telegrams. Hope to see as new driver soon :)

thomasa88, 2023-03-06:
Not stale. I still need some pointers.

closed 2023-06-26 by github-actions[bot]

gertjanathome, 2024-04-22:
Is there already a solution for this driver/meter. I heva a Siemens WSN515-FE wich is a Landis&Gyr T230. With the Im871a the readings are good, now the JSON driver is failing.