wmbusmeters/wmbusmeters #683: No telegrams

HAPSagan, 2022-11-21:
Hi
I'm having trouble and hope that someone can help me.

I have:
Odroid Blue (N2+)
Home Assistant 2022.11.3
Supervisor 2022.10.2
Operating System 9.3
Frontend-version: 20221108.0 - latest

and a

iM871A-USB - Wireless M-Bus USB-adapter 868 MHz connected via an USB-hub.

https://shop.imst.de/wireless-modules/usb-radio-products/10/im871a-usb-wireless-m-bus-usb-adapter-868-mhz

In config I have:

wmbusmeters > etc 
wmbusmeters.conf

looking like this:

loglevel=normal
device=auto:t1
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
logfile=/dev/stdout
shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"

When I start the add on I get the following log:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
MQTT Discovery cleanup...
Running wmbusmeters ...
Started auto im871a[00103839] on /dev/ttyUSB2 listening on t1
No meters configured. Printing id:s of all telegrams heard!

But I can not find a trace of any message. What am I doing wrong?

/Peter

HAPSagan, 2022-11-22:
Don't know if this is relevant, but I forgot to mention the drive path:

ttyUSB2
/dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_02759436-if00-port0

DEVLINKS: >-
  /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_02759436-if00-port0
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.4.3:1.0-port0
DEVNAME: /dev/ttyUSB2
DEVPATH: >-
  /devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.1.auto/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.3/1-1.1.4.3:1.0/ttyUSB2/tty/ttyUSB2
ID_BUS: usb
ID_MODEL: WiMOD_iM871A-usb
ID_MODEL_ENC: WiMOD\x20iM871A-usb
ID_MODEL_ID: ea60
ID_PATH: platform-xhci-hcd.1.auto-usb-0:1.1.4.3:1.0
ID_PATH_TAG: platform-xhci-hcd_1_auto-usb-0_1_1_4_3_1_0
ID_REVISION: '0100'
ID_SERIAL: Silicon_Labs_WiMOD_iM871A-usb_02759436
ID_SERIAL_SHORT: '02759436'
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_VENDOR: Silicon_Labs
ID_VENDOR_ENC: Silicon\x20Labs
ID_VENDOR_ID: 10c4
MAJOR: '188'
MINOR: '2'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '235096913859'
/dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_02759436-if00-port0

closed 2022-11-22 by HAPSagan

HAPSagan, 2022-11-22:
Hi
After desperate attempts I managed to get telegrams by changing from T1 to C1. I realize that it is due to a lack of knowledge on my part, but would be grateful if it were more simply described to make it easier for more beginners like me. Wouldn't it have been good if the default setting had been C1,T1?

weetmuts, 2022-11-22:
Great to hear that you solved it. 

The link-modes are not that easy......you just have to know how your meter transmits and have a dongle that can listen to that link-mode. Please make a pull request to the README with improved description about the link-modes for meters if you like!

Also the old im871a firmware could only listen to one of the linkmodes, the new firmware can listen to both c1 and t1 at the same time.

HAPSagan, 2022-11-22:
Thank you for your reply!
Is it possible to create a sensor for my "fjärrvärme" at the same time? I could see the telegrams.

weetmuts, 2022-11-22:
Sure, email me the telegrams or post them in an issue here. Then I can check if they are encrypted or not.