wmbusmeters/wmbusmeters #685: Support for ABB B23 111-100/Elvaco CMeX20w

snapspizza, 2022-11-21:
Hello

Thank you for this excellent piece of software. It is greatly appreciated!

There seems to be support for all of my meters, except for my electricity meter.

My electricity meter is a ABB B23 111-100 that's coupled with a Elvaco CMeX20w Meter Connectivity Module, which sends out the Wireless M-Bus telegrams.

[image]

A manual for the ABB meter can be found [here](https://search.abb.com/library/Download.aspx?DocumentID=2CMC485003M0201&LanguageCode=en&DocumentPartId=&Action=Launch) and a manual for the Elvaco MCM can be found [here](https://www.elvaco.se/Image/GetDocument/en/157/cmex20w%20users%20manual%20english.pdf).

Here's a few telegrams sent by the meter:
```
Started config rtlwmbus[00000001] listening on t1
(meter) electricity: meter detection could not find driver for id: 00674565 mfct: (ABB) ABB, Sweden (0x442) type: Electricity meter (0x02) ver: 0x20
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|844442046545670020027A3E000020_0E840017495200000004FFA0150000000004FFA1150000000004FFA2150000000004FFA3150000000007FFA600000000000000000007FFA700000000000000000007FFA800000000000000000007FFA90000000000000000000DFD8E0007302E38322E31420DFFAA000B3030312D313131203332421F|+0
telegram=|844442046545670020027A58000020_0E840019495200000004FFA0150000000004FFA1150000000004FFA2150000000004FFA3150000000007FFA600000000000000000007FFA700000000000000000007FFA800000000000000000007FFA90000000000000000000DFD8E0007302E38322E31420DFFAA000B3030312D313131203332421F|+0
telegram=|844442046545670020027A75000020_0E840021495200000004FFA0150000000004FFA1150000000004FFA2150000000004FFA3150000000007FFA600000000000000000007FFA700000000000000000007FFA800000000000000000007FFA90000000000000000000DFD8E0007302E38322E31420DFFAA000B3030312D313131203332421F|+0
```

Could anybody help me decode these into something intelligible and create a driver for the meter?

weetmuts, 2022-11-23:
@snapspizza There is more stuff in these telegrams.

https://wmbusmeters.org/analyze/844442046545670020027A3E0000200E840017495200000004FFA0150000000004FFA1150000000004FFA2150000000004FFA3150000000007FFA600000000000000000007FFA700000000000000000007FFA800000000000000000007FFA90000000000000000000DFD8E0007302E38322E31420DFFAA000B3030312D313131203332421F

But most values are zero right now. Can you checkout the content later when there are values that can be correlated with real world values, then we can add the remaining values as well.

snapspizza, 2022-11-23:
@weetmuts Yeah, the length of the telegram does suggest that there should be more relevant information in the telegram. Will update later with a sample telegram and see if we can make sense of it.