wmbusmeters/wmbusmeters #692: Issue with heat meter

thebang2, 2022-11-23:
Have a "Engelmann Sensostar 1" heat meter - meter is recognized with scan:
`
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 94677139
          manufacturer: (EFE) Engelmann Sensor, Germany (0x14c5)
                  type: Heat meter (0x04)
                   ver: 0x00
                device: rtlwmbus[00000001]
                  rssi: 43 dBm
                driver: sensostar
telegram=|_B344C5143971679400047B0500A025F4281EA83EAB349CBEC2205052673B1EBED0F9356363C7803F8A2718A723F07B95AB455C138AE98F6D2C8EC8A1FC8CEE1BD7CE64986D183ECC9EC9AB0347F1AF610B654E46FEB4A7C75E610F091AFD63F8698474EBEF1B165AEEBE6D3A5C8E16B4CDDCE40CE13114FB37B0B630270EEB5576DFEEA407E6D061A181A7B69E1865EB9D88CEF0CCD96D572931966BF0938EE556B077F668CF301F803C099B9D360E0501000111|+35
`

Added device in wmbusmeters.d\heizung:
`
name=Heizung
id=94677139
driver=sensostar
`

Log says:

`Started auto rtlwmbus[00000001] listening on t1
(wmbus) Unknown tpl-ci-field 7b
(meter) newly created meter (Heizung 94677139 sensostar) did not handle telegram!
(wmbus) Unknown tpl-ci-field 7b
(wmbus) Unknown tpl-ci-field 7b`

What is to do here?

kroimon, 2022-11-25:
This looks like an encrypted telegram to me - way too much entropy in the data.
For example, look at this [example sensostar telegram](https://wmbusmeters.org/analyze/68B3B36808007257004820c51400046c100000047839803801040600000000041300000000042B00000000142B00000000043B00000000143B00000000025B1400025f15000261daff02235c00046d2c2ddc24440600000000441300000000426c000001fd171003fd0c05000084200600000000c420060000000084300600000000c430060000000084401300000000c44013000000008480401300000000c48040130000000084c0401300000000c4c0401300000000a216) from the test suite, it has clearly repeated patterns and a lot of zero segments.

You will need the key to decrypt your telegrams.

weetmuts, 2022-11-25:
Nice, this is a telegram that uses the ci-field 0x7b which means m-bus compact frame. This is not yet implemented.
I will take a look at this after I have refactored all the drivers!

weetmuts, 2022-11-25:
So, wmbusmeters already implements 0x79 which is compact frame without header. 0x7b means compact with short header. It might also be encrypted as @kroimon says.

thebang2, 2022-11-25:
Thanks guys, I have already suspected that the telegram is encrypted. :/

weetmuts, 2023-02-02:
If you can collect more telegrams, then we might find the non-compact telegram.
But I still think you need the decryption key.

afflux, 2023-02-10:
got some more telegrams from Engelmann Sensostar here:

```
telegram=|_B344C5148614559100047B2300A025CA784EC60BFC2BC4EED932495525A1B40A8E11E33CF46639CB4D67C0C5CCC57090A4A14960B34F627BAF1307C0226AE22F914CDBE7C472723AC8715199B2B8EA1BF98937CE61CD95CEA57121BDF1698663211F35561ABFDE0ED5F12693F29FC5704CF013F3096E7E0883E2609305D2AFF7A32EA0A475B28014F7B0EBC1917A0765DA5DB85C5DCB0BC2E033A59B8E0BEC183354477A2B8367182832B51D987B200501002111|+39
telegram=|_B344C5148614559100047B2400A0259D20C78034FBDAB5E9189C56709A3B937E0495A947A9539DDEDA157B63787E4DAA5C3CB0DEAFF4C9820F92D565D3794F8B10344BC2E756B2E3F01166650E572B8B4574D7CAF012B0BDEE0AB05578AD5DE2AE84D041A81D4A8EDA906EAB347E9ACBEF8BCB2553E06199D5501BD302F42FF6AD85953292EF1116F18D7F5D3C1FF09DBBBBEE7099AEB77FF2D13F3E26C536BE536B04DB6C38EEE257F122A6E89BD10501002111|+130
telegram=|_B344C5148614559100047B2700A025C7D456C65BF79EF870059A6051A475E41CF8DD4BDDDA1CA09C129103900A05F4752FA4D2A7E9B7A836EEEFE8A931969D87990E56640F8694B94DE3E8D44A9537977CB5362945C11E6DBC5463E6F54B9C2FCE9983773E03485668CB61ECB3151FE42B96311674163E6F56313F81C67A0C59FB4C00543EBCE144C0A678AC53368F7B47B2E5AB7B4417ECB4DB0913F42944B9E51977E059352F3CB896B92BECF2450501002111|+60
telegram=|_B344C5148614559100047B2800A02576FBFF447F0F64ACC70D7604ED121A202AB6041B95801D1BDF25A4578961E8FC50A9F7A8E711FBF7731FB030BC6EC652AFF94423D7B7F9D665B4F12EB252796CF6069E4731B8D701977B9B46CECAA56429188045C5C7E9BB7209E5C99306BCF352EBED03380C0B473351E0258306F5D20B369B6F4BFB493AE4F480FD151F7751B635153BA5CF5BC2E5885AE802AC66411D5842EAE0C0F956FD3F7BC21089B3420501002111|+152
telegram=|_B344C5148614559100047B2900A0259CCDE3A9A6A848BAF492CE06C053872EF9B132A7D156480A0BFE5EC571EBB79331119772B951282F974A0F57CFB0C015668024D70391813AC2030B046433DCB58C30E65976DE6191B44FDFA23AD5A2DBEB87013E5EE4C36EFDEE9A49320DCCF3D059858B2B26DB526D03F04C8A22A8CB72C0CCB78969AC6C55F99E565B4260956066E36E41F0A493959E3A1FEDA3030110689A8E3BB19A33061EA30CE66D1A390501002111|+243
telegram=|_B344C5148614559100047B2A00A02598DFA045A040E7C080174B920CBCCAD189D8F206F6B9549D01DEE01B1B37B0E6E161B7A031E61E57AB03E0513CD61527A346BBE26E51DA75DEF7291D5098A1319CA57A1B43222A6C9AD9C1503CB5FB9B5DE9706A3A696EBAEA68D3DC77639E7FF2E0C7779CCAA43AF1FCE56BF555BD2B3E1C10395351ED8959602E549469D824B0301C16832CE9A2E93236585CF6E430499722CCAF3C48205EDB04DC5207457D0501002111|+334
telegram=|_B344C5148614559100047B2B00A02582B6410AE4C0D2CF439F0B98FA2B8C6B001979E53BD6A438AAC331300FBE3302F87F447E63FA4064197396E086B97E92724DE6150A2A14862F834EB9BCCC45165547E31F7E399DC843CD2549595CD84FD9EFCCE6ED5263DE9E62998F9B0E66B11B1D835BEB980E3930C70640FBC4A859ED12A008576577F2640CD886BC423E7A99FC4036DF78D2042A950652777483BDEA386F80076C6C4DCA620DFB5A6925280501002111|+425
telegram=|_B344C5148614559100047B2C00A02502C133D2F8AD15408BBCC27497215247B3B6729BE61D3EFD24391A2FF6AB07B7B7F4FC6FD60B1C3CDFD89395A41B4C27EB385338581C24FE70CE0D1092AFECCD00C7D2A786A19A2CBC445B3A123CA76DFBE3BFA3FEF88580BFE1098A59FD88D5CE2422D13CFDB0E7509E615CBEB12E860DEB9CE7E6E234DC453FA9B49EE7EA0324A6AF0F30EAF009198B0969E4EA08D78FA690A4AE46DC27A8EB8CB2AD084B6C0501002111|+516
telegram=|_B344C5148614559100047B2D00A02533E9EDC017118851BBF65CFECDDDA0FFE8FFCA69DCDB0B6EA34018F544C8195AD5E5C6FBC01724DD1F381284BD20200512FCBD428F82FE96CBB9F8D24C66EA5F2A4C3B90B988A8F620F6E4929CD6A47E0F346984D64FEFC7BDB3256183F52C767A866FE9A69E97E6A10246F5F1414701EADEEFB5592CA1F6C6A28EE978B95B409126A82F38BBB4EC0736970A9BFE90947FE03F0592B2FE338EBC2E9052159AB70501002111|+606
```

I have the decryption key too and can share that in private, if necessary.

afflux, 2023-02-12:
so I guess we have 8-bit access number, 8-bit status, 16-bit TPL config, followed by 10 blocks ciphertext. IV is manufacturer + address + 8*access number. (sorry if this is common for other types, I'm not very familiar with (w)mbus). Decrypts fine with my key, both with '/' trailer and fill bytes at the end.
No idea about the remaining 5 bytes in the end though

afflux, 2023-02-14:
I'm treating 7b just like 7a for now, which seems to work fine for decryption - not sure if that is causing the rest of the data interpretation to fail:
```
Using driver : sensostar 00/00
000   : b3 length (179 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : c514 dll-mfct (EFE)
004   : 86145591 dll-id (91551486)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7b tpl-ci-field (EN 13757-3 Application Layer with Compact frame (short tplh))
011   : 23 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : a025 tpl-cfg 25a0 (synchronous AES_CBC_IV nb=10 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 47 dif (64 Bit Integer/Binary Instantaneous value storagenr=1)
018   : 39 vif (Volume flow 10⁻⁵ m³/h)
019 C?: 245A2B31EA222D1D
027   : 00 dif (No data Instantaneous value)
028   : 00 vif (Energy mWh)
029   : 74 dif (32 Bit Integer/Binary Value during error state storagenr=1)
030   : 23 vif (On time days)
031 C?: 090000DF
035   : 2C dif (8 digit BCD Minimum value)
036   : AC vif (Power 10¹ W)
037   : 1B combinable vif (RecordErrorCodeMeterToController)
038 C?: 00009D1C
042   : 00 dif (No data Instantaneous value)
043   : 00 vif (Energy mWh)
044   : 00 dif (No data Instantaneous value)
045   : 1C vif (Mass 10¹ kg)
046   : 00 dif (No data Instantaneous value)
047   : 00 vif (Energy mWh)
048   : AC dif (8 digit BCD Minimum value)
049   : 1B dife (subunit=0 tariff=1 storagenr=22)
050   : 00 vif (Energy mWh)
051 C?: 00311B00
055   : 00 dif (No data Instantaneous value)
056   : 77 vif (Actuality duration days)
057   : 1A dif (4 digit BCD Maximum value)
058   : 00 vif (Energy mWh)
059 C?: 0013
061   : 1A dif (4 digit BCD Maximum value)
062   : 00 vif (Energy mWh)
063 C?: 00D8
065   : 19 dif (2 digit BCD Maximum value)
066   : 00 vif (Energy mWh)
067 C?: 00
068   : D1 dif (8 Bit Integer/Binary Maximum value storagenr=1)
069   : 19 dife (subunit=0 tariff=1 storagenr=19)
070   : 00 vif (Energy mWh)
071 C?: 00
072   : C6 dif (48 Bit Integer/Binary Instantaneous value storagenr=1)
073   : 19 dife (subunit=0 tariff=1 storagenr=19)
074   : 00 vif (Energy mWh)
075 C?: 00BD190000BD
081   : 19 dif (2 digit BCD Maximum value)
082   : 00 vif (Energy mWh)
083 C?: 00
084   : BD dif (variable length Value during error state)
085   : 19 dife (subunit=0 tariff=1 storagenr=18)
086   : 00 vif (Energy mWh)
087   : 00 varlen=0
088   : BD dif (variable length Value during error state)
089   : 19 dife (subunit=0 tariff=1 storagenr=18)
090   : 00 vif (Energy mWh)
091   : 00 varlen=0
092   : BD dif (variable length Value during error state)
093   : 19 dife (subunit=0 tariff=1 storagenr=18)
094   : 00 vif (Energy mWh)
095   : 00 varlen=0
096   : BD dif (variable length Value during error state)
097   : 19 dife (subunit=0 tariff=1 storagenr=18)
098   : 00 vif (Energy mWh)
099   : 00 varlen=0
100   : BD dif (variable length Value during error state)
101   : 19 dife (subunit=0 tariff=1 storagenr=18)
102   : 00 vif (Energy mWh)
103   : 00 varlen=0
104   : BD dif (variable length Value during error state)
105   : 19 dife (subunit=0 tariff=1 storagenr=18)
106   : 00 vif (Energy mWh)
107   : 00 varlen=0
108   : BD dif (variable length Value during error state)
109   : 19 dife (subunit=0 tariff=1 storagenr=18)
110   : 00 vif (Energy mWh)
111   : 00 varlen=0
112   : BD dif (variable length Value during error state)
113   : 19 dife (subunit=0 tariff=1 storagenr=18)
114   : 00 vif (Energy mWh)
115   : 00 varlen=0
116   : BC dif (8 digit BCD Value during error state)
117   : 19 dife (subunit=0 tariff=1 storagenr=18)
118   : 00 vif (Energy mWh)
119 C?: 00761900
123   : 00 dif (No data Instantaneous value)
124   : 4E vif (Volume flow ext. l/s)
125   : 19 dif (2 digit BCD Maximum value)
126   : 00 vif (Energy mWh)
127 C?: 00
128   : 98 dif (Selection for Readout Maximum value)
129   : 18 dife (subunit=0 tariff=1 storagenr=16)
130   : 00 vif (Energy mWh)
131   : 00 dif (No data Instantaneous value)
132   : 09 vif (Energy 10¹ J)
133   : 18 dif (Selection for Readout Maximum value)
134   : 00 vif (Energy mWh)
135   : 00 dif (No data Instantaneous value)
136   : 50 vif (Mass g/h)
137   : 17 dif (64 Bit Integer/Binary Maximum value)
138   : 00 vif (Energy mWh)
139 C?: 005E1600009E1500
147   : 00 dif (No data Instantaneous value)
148   : 08 vif (Energy J)
149   : 15 dif (32 Bit Real Maximum value)
150   : 00 vif (Energy mWh)
151 C?: 002B1400
155   : 00 dif (No data Instantaneous value)
156   : 79 vif (Enhanced identification)
157   : 13 dif (24 Bit Integer/Binary Maximum value)
158   : 00 vif (Energy mWh)
159 C?: 002F2F
162   : 2F skip
163   : 2F skip
164   : 2F skip
165   : 2F skip
166   : 2F skip
167   : 2F skip
168   : 2F skip
169   : 2F skip
170   : 2F skip
171   : 2F skip
172   : 2F skip
173   : 2F skip
174   : 2F skip
175   : 05 dif (32 Bit Real Instantaneous value)
176   : 01 vif (Energy 10⁻² Wh)
177 C?: 0021
179   : 11 dif (8 Bit Integer/Binary Maximum value)

{
    "media":"heat",
    "meter":"sensostar",
    "name":"",
    "id":"91551486",
    "total_kwh":null,
    "timestamp":"2023-02-14T19:45:30Z"
}
```

afflux, 2023-02-14:
So it seems this frame being compact means the DIF and VIF fields are not actually contained :disappointed: 
There is supposed to be a separate frame (tpl-ci 0x6a) which contains the corresponding field descriptions, which is needed to parse this data. I'll try to capture some more and see if the meter ever transmits 0x6a.

weetmuts, 2023-02-15:
Yep, that is correct. Wmbusmeters already have support for collecting the format bytes from a long telegram and reusing the format for a compact telegram.  But I have never seens a 0x6a telegram with only the format bytes,
even though I knew they exist.

If you manage to collect them please post them here!

afflux, 2023-02-26:
I did not see any `0x6a` frame for this specific meter, but I did find some example data which I shared with you in private.

Also, I was able to correlate the data in my decrypted frame with actual values and found the following decoding. It is from an Engelmann *SensoStar U*, operating in compact mode.

measurement values are always 32-bit little endian. Heat in kWh, flow in liters.
```
2f2f 4739 7bf2 2725ef22   561d0000 6a2e0900 00  df2c ac1b0000 [historical] [2f2f2f2f... trailer]
^    ^    ^    ^ ^ ^ ^^   ^        ^        ^   ^    ^
|    |    |    | | | |m   |        |        |   |    |--last yearly value for total heat
|    |    |    | | | Yh   |        |        |-? |--yearly readout date (binary: yyyddddd yyyymmmm)
|    |    |    | | &1f=dd |        |
|    |    |    | | &e0=Yl |        |--current total flow
|    |    |    | &1f=HH   |--current total heat
|    |    |    | &e0=??
|    |    |    &3f=MM
|    |    |    &c0=??
|    |    |    |-- current readout datetime
|    |    |
|    |    |-- decryption crc?
|    |-- frame format identifier
|-- const decryption ok marker
```

To be more clear, this is how the date/time is parsed:
```python
def parse_date(buf):
    day = buf[0] & 0x1f
    month = buf[1] & 0xf
    year = 2000 + ((buf[1] >> 4) << 3) | (buf[0] >> 5)
    return datetime.date(year, month, day)

def parse_time(buf):
    minute = buf[0] & 0x3f
    hour = buf[1] & 0x1f
    return datetime.time(hour, minute)

def parse_datetime(buf):
    return datetime.datetime.combine(parse_date(buf[2:]), parse_time(buf))
```

historical consists of the 30 most recent values, recorded on the end of 15th and the end of the last day of each month. First is most recent

closed 2023-05-28 by github-actions[bot]