wmbusmeters/wmbusmeters #696: Kamstrup Multical 403

HAPSagan, 2022-11-24:
I'm trying to add my Kamstrup Multical 403 but can't find any driver. Can someone help me?

HAPSagan, 2022-11-24:
Forgot to mention that the the telegram says "driver: unknown!"

kroimon, 2022-11-25:
Please provide one or more example telegrams, in the best case with links to an analysis at https://wmbusmeters.org/.
Without this, no one can guess why your telegrams could not be parsed.

HAPSagan, 2022-11-25:
telegram=|3F442D2C7934347134048D2088A5CC33228EAD79EB470F0F_048C01005D810200752A010052850100D8670300B21B2C0E50000000200000004900000000000000|+181

key=E953179E876CFE2D1C2139D366DFF4E2

Auto driver : not found! Best driver : 00/00 Using driver : 00/00 000 : 3f
length (63 bytes) 001 : 44 dll-c (from meter SND_NR) 002 : 2d2c dll-mfct
(KAM) 004 : 79343471 dll-id (71343479) 008 : 34 dll-version 009 : 04
dll-type (Heat meter) 010 : 8d ell-ci-field (ELL: Extended Link Layer II (8
Byte)) 011 : 20 ell-cc (slow_resp sync) 012 : 88 ell-acc 013 : a5cc3322 sn
(AES_CTR) 017 : 21e1 payload crc (calculated 393e ERROR) 019 CE:
A87B6B49FC2C098433C751DB705A578328D15EFDFB3D1CDCC59A4FDE60C2066B1198418474505DC1FA809E8EAC
failed decryption. Wrong key? { "media":"heat", "meter":"unknown",
"name":"", "id":"71343479", "timestamp":"1970-01-01T00:00:00Z" } Using:
wmbusmeters: 1.9.0-129-g53a2d65 53a2d650e541d1a77a97573df0779c154895e5cc

weetmuts, 2022-11-25:
@HAPSagan actually the telegram you provided here is decrypted already...payload crc is ok.

https://wmbusmeters.org/analyze/3F442D2C7934347134048D2088A5CC33228EAD79EB470F0F048C01005D810200752A010052850100D8670300B21B2C0E50000000200000004900000000000000

But as you can see it is a compressed telegram, ie we need the longer telegram first to be able to decode this.
When wmbusmeters is running, it will stay silent until it receives the longer telegram, after that it can decode these compressed telegrams.

weetmuts, 2022-11-25:
See if you can collect a long telegram, then we can see which driver is the best for it!

closed 2023-03-05 by github-actions[bot]