wmbusmeters/wmbusmeters #697: Problem with Docker container

YOYONL89, 2022-11-24:
Hello all,

I have updated the wmbusmeters docker container, there by i have adjusted the conf file to the example format as is written in the github code.
Unfortunately now the docker wont start anymore.

Error in Portainer:
```
No such log level: info
Not a valid device im871a:c1
```

Conf:
```

loglevel=info
# You can use auto:t1 to find the device you have connected to your system.
# But do not use auto here since it will cause unnecessary and slow probing of the serial ports.
device=im871a:c1
logtelegrams=false
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day
logfile=/wmbusmeters_data/logs/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h xxx.xxx.xxx.xxx -t wmbusmeters/$METER_ID -m "$METER_JSON"
ignoreduplicates=true
```

Did something change along the way for the container?

weetmuts, 2022-11-24:
Can you check that you have not accidentally used a windows editor for the conf file? It will insert CRLF instead of just LF lines....

YOYONL89, 2022-11-24:
ahh crap, your right there where CRLF lines in it.
Must have reviewed the settings in wordpad, modified something in a moment of not paying attention and saved the file.
Sorry aka Solved!

closed 2022-11-24 by YOYONL89