wmbusmeters/wmbusmeters #699: HA Add-on :: MQTT discovery :: Topics

oandry72, 2022-11-25:
Can you please combine attributes of meters into one topic with meter names?

https://github.com/weetmuts/wmbusmeters/blob/53a2d650e541d1a77a97573df0779c154895e5cc/ha-addon/mqtt_discovery.sh#L60

like this
`topic="$MQTT_DISCOVERY_PREFIX/$component/wmbusmeters/${aryKV['id']}/$attribute/config"`

If we have more than 3-5 meters, it turns into a mess.
It would be better if all related attributes were located in meter branches.

kroimon, 2022-11-25:
According to [the Home Assistant documentation on MQTT discovery messages](https://www.home-assistant.io/integrations/mqtt/#discovery-messages), the topic format needs to be `<discovery_prefix>/<component>/[<node_id>/]<object_id>/config`, so an additional level doesn't seem to be supported.

oandry72, 2022-11-25:
But it works for the Z2M integration
[image]

kroimon, 2022-11-25:
As you can see from your screenshot, Z2M does not use an additional level in the topic hierarchy, but instead uses a separate node for each of its devices instead of grouping them under a common Z2M node.

So what you are asking for is using 
`topic="$MQTT_DISCOVERY_PREFIX/$component/${aryKV['id']}/$attribute/config"`
instead of
`topic="$MQTT_DISCOVERY_PREFIX/$component/wmbusmeters/${aryKV['id']}_$attribute/config"`

This will result in more nodes directly below "sensor", because you then get one node per wmbus device instead of only one called "wmbusmeters", are you sure that helps with readability?

oandry72, 2022-11-25:
Ahh, ok, got it now 

But what if we get rid of the wmbusmeters name from the topic?
like this
sensor\0089710\rssi_dbm\config
sensor\0089710\total_m3\config
etc.

closed 2023-04-05 by github-actions[bot]