wmbusmeters/wmbusmeters #702: Sagecom EF4EVO Gas meter

sgargel, 2022-11-27:
My gas and energy reseller company installed in my house this new " smart meter with remote communication and management system".

https://www.sagemcom.com/V02/fileadmin/user_upload/Energy/Electricity/Niv4/Meter_Italia/EF4EVO.pdf

As stated in the User Manual provided, it uses Wireless M-Bus 169 Mhz radio-communication and communication protocol DLMS-COSEM.

As I want to try to have a remote measurement read of my gas usage, I tryed to capture radio signals with `rtl_433 -f 169728051 -s 1600000 -S all -A` and after about 10 hours of nothing, I found some 'cu8' files saved on my disk.

I'm not sure how to "read" them and also I'm not sure if those signals can be in any way "decoded". 

Any help or comment will be very appreciated.

Thanks!

BIBOLV, 2022-11-27:
Please check https://github.com/weetmuts/wmbusmeters/issues/677

wilds, 2022-12-09:
I think this could be useful:
DLMS COSEM library https://github.com/Gurux/GuruxDLMS.c

closed 2023-03-18 by github-actions[bot]