wmbusmeters/wmbusmeters #703: Sensus iPERL warm(!) water?

marcelstoer, 2022-11-28:
wmBus noob here, a little lost.

wmbusmeters and the iM871A tell me that they found a Sensus/GWF warm water meter for which there be an unknown driver. Yet, what I see in the basement is a "regular" iPERL (cold) water meter at the pipe feed our house. Also, the reported ID is not what I think I am seeing on the meter.

```
pi@white-rose:~/wmbusmeters $ build/wmbusmeters /dev/ttyUSB0:im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 875103bf
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x00
                device: im871a[00103852]
                  rssi: -96 dBm
                driver: unknown!
```
I have the debug output at the very end.

I tried force-applying the 'iperl' driver onto this meter using a random key (as I don't have my own yet) but that doesn't seem the way to handle this use case.

```
pi@white-rose:~/wmbusmeters $ build/wmbusmeters /dev/ttyUSB0:im871a:t1,c1 MyWaterMeter iperl 875103bf 00112233445566778899AABBCCDDEEFF
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
(meter) MyWaterMeter: meter detection could not find driver for id: 875103bf mfct: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) MyWaterMeter: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 875103bf mfct: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
MyWaterMeter	875103bf	null m³	null m³/h	2022-11-28 07:46.42
MyWaterMeter	875103bf	null m³	null m³/h	2022-11-28 07:46.51
```

**Photos**
Ignore the yellow circle around the optical sensor hole.
[image]
[image]

**Debug output**
```
pi@white-rose:~/wmbusmeters $ build/wmbusmeters --debug /dev/ttyUSB0:im871a:t1,c1
(wmbusmeters) version: 1.9.0-148-g073aafd
(config) using device: /dev/ttyUSB0:im871a:t1,c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" im871a
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601F3390000C61B"
(im871a) checkIM871AFrame "A581100833001601F3390000C61B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 1bc6 expected 1bc6
(im871a) received full frame
(device info bytes) "33001601F3390000"
(im871a/im170a) info: type=im871a mode=collector firmware=16 hci=01 uid=000039f3
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000200B32552381000010001FF07023200000000001450"
(im871a) checkIM871AFrame "A5810616FF000200B32552381000010001FF07023200000000001450"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 5014 expected 5014
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000200B32552381000010001FF0702320000000000"
(serial) EVENT thread interrupted
(im871a/im170a) config: other link_mode=s2 id=00103852 media=00 version=01 c_field=00 auto_rssi=00
(im871a/im170a) are you there? yes 00103852 im871a firmware: 16
(main) opening /dev/ttyUSB0:im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810616FF000200B32552381000010001FF07023200000000001450"
(im871a) checkIM871AFrame "A5810616FF000200B32552381000010001FF07023200000000001450"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 5014 expected 5014
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00103852
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001601F3390000C61B"
(im871a) checkIM871AFrame "A581100833001601F3390000C61B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 1bc6 expected 1bc6
(im871a) received full frame
(im871a) device info completed
(im871a) device info: type=im871a mode=collector firmware=16 hci=01 uid=000039f3
(im871a) set config to set link mode 0a
(serial /dev/ttyUSB0) sent "A501030600020A300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received binary "A5C2031944E61EBF0351870006B2010654168A26737CE256319854BF6F2D1A92"
(im871a) checkIM871AFrame "A5C2031944E61EBF0351870006B2010654168A26737CE256319854BF6F2D1A92"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 45 (-100 dBm)
(im871a) got crc16 921a expected 921a
(im871a) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=1ee6 (GWF) A=875103bf VER=00 TYPE=06 (Warm Water (30°C-90°C) meter) (driver unknown!) DEV=im871a[00103852] RSSI=-100
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=b2
Received telegram from: 875103bf
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x00
                device: im871a[00103852]
                  rssi: -100 dBm
                driver: unknown!
(wmbus) 000   : 19 length (25 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : e61e dll-mfct (GWF)
(wmbus) 004   : bf035187 dll-id (875103bf)
(wmbus) 008   : 00 dll-version
(wmbus) 009   : 06 dll-type (Warm Water (30°C-90°C) meter)
(wmbus) 010   : b2 tpl-ci-field (Mfct specific)
(wmbus) 011 C?: 010654168A26737CE256319854BF6F mfct specific
(wmbus) telegram from 875103bf ignored by all configured meters!
```

marcelstoer, 2022-11-29:
SURPRISE! It's not a "warm water" meter but a _gas_ meter 😄 

I moved the Pi closer to the basement as the RSSI was so low anyway. I am now seeing two meters one of them being the iPERL:

```
pi@white-rose:~ $ wmbusmeters/build/wmbusmeters /dev/ttyUSB0:im871a:t1,c1
Started config im871a[00103852] on /dev/ttyUSB0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 20343320
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                device: im871a[00103852]
                  rssi: -83 dBm
                driver: iperl
Received telegram from: 875103bf
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x00
                device: im871a[00103852]
                  rssi: -91 dBm
                driver: unknown!
```
As the only other meter I see in the basement is the GWF gas meter (with a connected RF module) I conclude that it's the one reported as "type: Warm Water".

[image]

marcelstoer, 2022-12-01:
Superseded by #712 -> closing

closed 2022-12-01 by marcelstoer

alexboss, 2024-01-23:
> **Photos** Ignore the yellow circle around the optical sensor hole. [image] 

Dear @marcelstoer 

Are you sure it's an optical sensor there ? 

I thought it was initially, but can't find any information in Sensus documents and leaflets.

My plan was to cover it with a LED pulse sensor and then get metrics in HASSIO, but not so sure it's an optical sensor.

I wrote to Sensus, but didn't get any answer:

```
My municipality has just installed a sensus iPERL.
I would like to automatically track my water consumption.
Can you tell me what the hole above the meter screen is? Is this an LED for optical pulse reading?
In which case, what volume of water does one pulse correspond to?
I looked at the doc https://www.xylem.com/siteassets/brand/sensus/resources/brochure/sensus-iperl-high-performance-static-water-meter-brochure.pdf but nothing is listed on this subject .
Thanking you in advance.
Sincerely.
```

Thanks for any feedback provided about this little mysterious hole.

Best regards.

Alexandre

marcelstoer, 2024-01-23:
> Are you sure it's an optical sensor there ?

I'm not, it was just a guess.