wmbusmeters/wmbusmeters #706: Moving wmbusmeters to new Rasperry not working

kovu94, 2022-11-29:
Hi all of you,

at the moment I'll try to move my wmbusmeters installation from my old (testing) BananaPi M1 to a Raspberry Pi 3b because of the instability of the old BananaPi.
First I tried with DietPi image but I'm not able to get my CUL working. Now I'm on the actual official Rasbperry Image.

root@raspberrypi:/home/kovu94# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian

I installed all software packages and compiled wmbusmeters without errors. When I start it with the working config file from my BananaPi it works but it seems to constantly restarting (the log is flooding with found telegrams and also probing of several ports and also top shows the wmbusmetersd with increasing PIDs), but at least the access to the CUL works and telegrams are received and forwarded by MQTT to my HASS environment.

In the moment I'll add the CUL directly in the config file it's not working.

Working configuration in config-file: 
device=auto:t1

Not working: 
device=/dev/ttyUSB0:t1 or /dev/ttyUSB0:cul:t1

I also tried a symlink because I want to use two different CULs on the device, one for wmbus one for switching some plugs via pilight. But these are not installed yet.

Everytime I use another value then auto:t1 he constantly saying:
(config) number of meters: 1
No wmbus device detected, waiting for a device to be plugged in.

Tried with /dev/ttyACM0, /dev/ttyUSB-WMBUS (my symlink, see below) or /dev/ttyUSB0 and with several extensions like :t1 or :cul:t1

My symlink is build like this:
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", SYMLINK+="ttyUSB-CUL433", GROUP="dialout", MODE>
SUBSYSTEM=="tty", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="204b", SYMLINK+="ttyUSB-WMBUS", GROUP="dialout", MODE=>

What did I wrong? I have absolutely no idea what's wrong because it was working out of the box with the BananaPi.

kovu94, 2022-11-30:
Addition:
It's working without issues when I fire up the following command:

wmbusmeters  --verbose --debug /dev/ttyUSB-WMBUS cul:t1

(wmbusmeters) version: 1.9.0-151-g9a8cd21
(config) using device: /dev/ttyUSB-WMBUS
(config) using device: cul:t1
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect cul)
(serial /dev/ttyACM0) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary "6235333434413531313039373135373732373630374643413838433030314439303046303032433235423334333036303030323339324138343633423544303735383843443741423330303331303731303338414136454345334530383846414634374434444533454431384532334545364638393736463331423333333941433830443933444130424643303542343331414232453630334538333442344631334430443539424531363431394241364535393933384632313534303030395620312E36372043554C3836380D0A"
(cul) probe response "b5344A511097157727607FCA88C001D900F002C25B343060002392A8463B5D07588CD7AB30031071038AA6ECE3E088FAF47D4DE3ED18E23EE6F8976F31B3339AC80D93DA0BFC05B431AB2E603E834B4F13D0D59BE16419BA6E59938F21540009V 1.67 CUL868<0D><0A>"
(serialtty) closed /dev/ttyACM0 (detect cul)
(cul) are you there? yes
(serial) EVENT thread interrupted
(main) found specified device (cul:t1) that matches detected device (/dev/ttyACM0:cul[]:38400/0)
(main) opening cul:t1
Started config cul on /dev/ttyACM0 listening on t1
(cul) on /dev/ttyACM0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyACM0 will happen every 82800 seconds
(cul) set link mode t
(serial /dev/ttyACM0) sent "6272740D0A"
(serial) EVENT thread interrupted
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyACM0) sent "5832310D0A"
(wmbus) no alarm (expected activity) for cul
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) check if /dev/ttyAMA0 can be accessed
(serial) EVENT thread interrupted
(serial) tty /dev/ttyAMA0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 4 (detect cul)

But when I edit the config file like this:

device=/dev/ttyUSB-WMBUS cul:t1
or
device=/dev/ttyUSB-WMBUS:cul:t1
or
device=/dev/ttyUSB-WMBUS

this happens after starting it:

(wmbusmeters) version: 1.9.0-151-g9a8cd21
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: /dev/ttyUSB-WMBUS:cul:t1
(config) number of meters: 1
**No wmbus device detected, waiting for a device to be plugged in.**
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 725 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>722
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 726 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

Documentation says: 
Check the config file /etc/wmbusmeters.conf and edit the device. For example: /dev/ttyUSB1:amb8465:c1,t1 or im871a:c1,t1 or im871a[457200101056]:t1.
So I think  /dev/ttyUSB-WMBUS:cul:t1 is correct?

BIBOLV, 2022-11-30:
Yes, you can try that and also add `donotprobe=/dev/ttyAMA0` - as mentioned in documentation, serial interface causes grief on Pi.

kovu94, 2022-12-01:
Still not working - startup log with donotprobe=all:

root@raspberrypi:/home/kovu94# wmbusmeters --useconfig
(config) "device" "/dev/ttyUSB-WMBUS"
(config) "donotprobe" "all"
(config) "logtelegrams" "true"
(config) "format" "json"
(config) "meterfiles" "/var/log/wmbusmeters/meter_readings"
(config) "meterfilesaction" "overwrite"
(config) "meterfilesnaming" "name-id"
(config) "# meterfilestimestamp=day" ""
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log"

Sill saying no wmbus-device found:

(wmbusmeters) version: 1.9.0-151-g9a8cd21
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: /dev/ttyUSB-WMBUS
(config) number of meters: 1
**No wmbus device detected, waiting for a device to be plugged in.**
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 712 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>709
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 713 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

dtk33d, 2022-12-06:
Same here, also have raspi with bullseye.

BIBOLV, 2022-12-06:
@kovu94, can you post log with debug from service running with non working config?

closed 2023-03-18 by github-actions[bot]

kovu94, 2023-03-23:
Sorry for the extremely delayed answer. Due to a broken leg of my partner there was no time for playing around.
But last weekend I've got a huge update to the banapi which is working fine since then so I'm not in need to use the raspberry.