wmbusmeters/wmbusmeters #707: water meters reporting updated values only once per day

ddppddpp, 2022-11-30:
I'm using wmbusmeters (1.9.0-76-gb9f3446...) to collect usage data for a hot and a cold water meter using the mkradio4 and mkradio3 drivers.
While the reported values seem accurate, they are only updated once per 24h, usually following the same pattern (one meter being updated ~ 23:50 local time and the other in the early AMs), no matter what the actual reported quantity is.
Is this behaviour normal?
Here's a copy of my configuration

```
loglevel=normal
device=rtlwmbus
logtelegrams=true
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
metterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
meterfilesnaming=name
meterfilestimestamp=day
shell=<correct call to mosquitto_pub>
alarmshell=<correct call to mosquitto_pub>
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)

```

meter file

```
name=coldwater
driver=mkradio4
id=xxxxxxx
key=00112233445566778899AABBCCDDEEFF
```

BIBOLV, 2022-11-30:
I would say, yes, if you do not receive any other messages, then they are probably configured to report only once a day, if they are owned by some utility company, in order to preserve battery for longer life. If you own them then IMO you can contact seller and ask them to change configuration to send telegrams more frequently.

ddppddpp, 2022-12-01:
Thank you, limiting the reporting frequency sounds smart. Technically those are my meters, but I still need to call somebody in the utility company so that they can come over... not gonna happen :-)

ddppddpp, 2022-12-01:
> I would say, yes, if you do not receive any other messages, then they are probably configured to report only once a day, if they are owned by some utility company, in order to preserve battery for [...]

2nd thought, why do I keep getting telegrams every 3 mins or so.... Isn't the bulk amount of energy spent on radio transmission, not on registering a magnetic impulse by a rotating dial or whatever they use...

closed 2022-12-01 by ddppddpp

BIBOLV, 2022-12-01:
Wait, you mean that telegram are being sent, but actual consumtion figure changes once a day inside in telegram? That sounds strange. 
Can you provide few telegram to take a look.

ddppddpp, 2022-12-04:
```
telegram=|2F446850583660149572A2_06772D5A0A4018240004000B19190000000000000000000000000000000000000000000000|+1162
telegram=|2F446850004636247462A2_06BE2CED16401899011000082C2C2C3427221921101E171525032727302A332A2B313B3133|+1173
telegram=|2F446850583660149572A2_06772D5A0A4018240004000B19190000000000000000000000000000000000000000000000|+1197
telegram=|2F446850004636247462A2_06BE2CED16401899011000082C2C2C3427221921101E171525032727302A332A2B313B3133|+1205
telegram=|2F446850583660149572A2_06772D5A0A4018240004000B19190000000000000000000000000000000000000000000000|+1231
telegram=|2F446850583660149572A2_06772D5A0A4018240004000B19190000000000000000000000000000000000000000000000|+1266
telegram=|2F446850004636247462A2_06BE2CED16401899011000082C2C2C3427221921101E171525032727302A332A2B313B3133|+1276
telegram=|2F446850004636247462A2_06BE2CED16401899011000082C2C2C3427221921101E171525032727302A332A2B313B3133|+1309
telegram=|2F446850583660149572A2_06772D5A0A4018240004000B19190000000000000000000000000000000000000000000000|+1335
telegram=|2F446850004636247462A2_06BE2CED16401899011000082C2C2C3427221921101E171525032727302A332A2B313B3133|+1339
```

BIBOLV, 2022-12-04:
Yes, have not seen such behavior, but either way - this is meter side configuration. You can try to reach out to our utility company and ask to change config.

jsuchal, 2022-12-06:
This is the same behavior is see with my radio3 water meters. Probably nothing can be done here. I suspect this is because of personal data privacy issues.

Anyways, the resolution on my meter is 0.1m^3 so I wouldnt probably see much difference even if it reported current data every time.