wmbusmeters/wmbusmeters #709: Heat meter "Vario451" after upgrade to Wmbusmeters 0.2.24 reports 0 values.

Volcano0, 2022-12-01:
Heat meter "Vario451" after upgrade to Wmbusmeters 0.2.24 reports 0 values.

Data from MQTT Explorer.

{
  "media": "heat",
  "meter": "vario451",
  "name": "heat",
  "id": "12345657",
  "total_kwh": 0,
  "current_kwh": 0,
  "previous_kwh": 0,
  "timestamp": "2022-12-01T10:19:04Z",
  "device": "rtlwmbus[00000001]",
  "rssi_dbm": 127
}

After downgrade to 0.2.23 it shows correct values.

weetmuts, 2022-12-02:
Can you provide an example telegram?

weetmuts, 2022-12-02:
Or test the latest version on github?

Volcano0, 2022-12-09:
I'm sorry for delay.
I tried to reproduce the problem and I have installed version 0.2.24 again, but it was working back.
So I;m not able to give you more details.

BIBOLV, 2023-01-03:
Closing as issue is not possible to reproduce currently.

closed 2023-01-03 by BIBOLV